espressif/vscode-esp-idf-extension #572: Can't flash via UART

SERIDJ, 2021-11-10:
When I first created my project, VSCode (with esp-idf extension) asked me if I wanted to flash via JTAG or UART. I selected accidentally the JTAG option and now I am unable to flash the board because I don't have correctly set up JTAG.

I don't want to configure JTAG at this point on the project, so I would like to go back to UART flashing, but I can't find the configuration parameter on the SDK configuration editor.

If anyone could point me to the correct configuration file would be very much appreciated.

brianignacio5, 2021-11-10:
The Flash command will save your preferred type of flashing in the `idf.flashType` configuration setting. You can modify this setting directly in your User settings, workspace settings or in your project's `.vscode/settings.json`.

You have several options:

* Use `ESP-IDF: Flash (UART) your project` command to flash with UART directly.
* Set `idf.flashType` to UART.

SERIDJ, 2021-11-10:
@brianignacio5  thanks for ur help its works well with  : **Use ESP-IDF: Flash (UART)** command 

 i'm not familiar with vscode ,i work with Eclipse IDE that why  iam  i'm a little lost .... 

So to change  this param idf.flashType  , i should find configuration settings  ...

brianignacio5, 2021-11-11:
Sure, here is the [vscode settings](https://code.visualstudio.com/docs/getstarted/settings) documentation. Basically 3 places where you can define a setting: User Settings, Workspace Settings and in your project's `.vscode/settings.json`

PR #576 (2021-11-18): add flashing types to docs
Fix #572 

Add `idf.flashType` to basic use and readme documentation. Fix keyboard shortcut command reference
closed 2021-11-18 by brianignacio5