espressif/vscode-esp-idf-extension #574: Menuconfig dont open

baldhead69, 2021-11-16:
Hi,

I am trying to open menuconfig but it dont open
[image]
.

baldhead69, 2021-11-17:
@brianignacio5 ,

Some suggestion ?

What other way to open the menuconfig only with the visual studio code extension ?

In cmd i can't open menuconfig too.

brianignacio5, 2021-11-17:
Hi @baldhead69 Could you please share the required troubleshoot information as specified [here](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting)

Have you configured the extension as explained [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md) ?

baldhead69, 2021-11-17:
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/7556476/esp_idf_vsc_ext.log)

baldhead69, 2021-11-17:
[image]
[vscode-app-1637168186362.log](https://github.com/espressif/vscode-esp-idf-extension/files/7556508/vscode-app-1637168186362.log)
[image]

baldhead69, 2021-11-17:
I am using my old project setup in vsc.

There is some project example ?

baldhead69, 2021-11-17:
".vscode" folder inside project

[.vscode.zip](https://github.com/espressif/vscode-esp-idf-extension/files/7556628/default.vscode.zip)

baldhead69, 2021-11-17:
The first time i tried to open menuconfig after installing the expressif extension, menuconfig open normally.

Now, the menuconfig dont open more.

baldhead69, 2021-11-17:
Building an flashing are ok.

baldhead69, 2021-11-17:
Now the menuconfig apparently opened.

I removed the dot from the name of a folder from my components.

The strange thing is that even with the dotted folder the project was compiling.

But full clean still does not work:
[image]

baldhead69, 2021-11-17:
if i open visual studio code as an admin, the first time the full clean work, but if i build again and try to full clean again, the full clean work. 

if i open visual studio code as an admin, the first time the full clean work, but if i build again, open menuconfig, close menuconfig, and try to full clean again, the full clean dont work( EBUSY message again ).

brianignacio5, 2021-11-18:
The issue with the `SDK Configuration editor` is a build issue you are having with the `__component_get_requirements`. Here is the error:

```
CMake Error at C:/esp-idf/tools/cmake/component.cmake:238 (message):
Call Stack (most recent call first):
  C:/esp-idf/tools/cmake/build.cmake:491 (__component_get_requirements)
  C:/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:15 (project)

,stack:Error: CMake Error at C:/esp-idf/tools/cmake/component.cmake:238 (message):
Call Stack (most recent call first):
  C:/esp-idf/tools/cmake/build.cmake:491 (__component_get_requirements)
  C:/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:15 (project)

        at m.printError (c:sers�3893983\.vscodetensionspressif.esp-idf-extension-1.2.0\disttension.js:1:28725)
        at Socket.<anonymous> (c:sers�3893983\.vscodetensionspressif.esp-idf-extension-1.2.0\disttension.js:1:28154)
        at Socket.emit (events.js:315:20)
        at addChunk (internal/streams/readable.js:309:12)
        at readableAddChunk (internal/streams/readable.js:284:9)
        at Socket.Readable.push (internal/streams/readable.js:223:10)
        at Pipe.onStreamRead (internal/stream_base_commons.js:188:23)
```

Sometime the `ESP-IDF Full Clean` can't fully delete the build folder because another process is using some file with it. Maybe try to run `Dispose current SDK Configuration Editor server process` and try `ESP-IDF Full Clean` again. If that doesn't work try to delete the folder yourself from the file explorer.

closed 2021-11-19 by baldhead69