espressif/vscode-esp-idf-extension #579: IDF select port can not change port

edieruby, 2021-11-23:
- OS Name & Version:  windows10
- VSCode Version: 1.62.3
- ESP-IDF Version: 4.3.1

Port changed from COM6 to COM3, but it not works.

[image]
[image]
[image]
[image]
[image]

brianignacio5, 2021-11-24:
When you use the `ESP-IDF: Select port to use` command, the `idf.port` is updated. But the thing is, there is another setting called `idf.saveScope` which define where is the setting saved (User, Workspace or Workspace folder). Probably your project's .vscode/settings.json has a `idf.port` setting that overrides the global user settings (which is probably the one being updated). More about this in the [multiple projects documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/MULTI_PROJECTS.md).

In the GitHub master, we had updated so that `idf.port` is updated to the workspace folder only to avoid this issue.
Please try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/115914250.zip) to test it.

brianignacio5, 2021-12-02:
Closing this issue for lack of response, comment here if you still have additional issues.

closed 2021-12-02 by brianignacio5

moefear85, 2022-01-30:
I've recently begun facing this problem. idf.port in settings.json is set to /dev/ttyACM2. It is not set in the ".code-workspace" file. Yet still the extension is always trying to use /dev/ttyUSB1 when flashing, which leads to errors. I explicitly select the project I'm working on every time I update the port, and the extension indicates it changed successfully. but it still keeps using the wrong port. Infact, this issue also happens when I select the mcu type.