espressif/vscode-esp-idf-extension #580: I have ever install IDF with shell in ubuntu, but the vscode extension not identify it

Yecgaa1, 2021-11-23:
[image]
I have ever set  IDF_PATH=$HOME/esp/esp-idf
And I have ever try to set this
[image]
[image]
I can build but not flash with this extension. If with shell and export.sh, I can do it
I know the reason is that I dont set the path with export.sh
But how I can solve this problem with this extension automatically

Yecgaa1, 2021-11-23:
I know this topic so I have set IDF_PATH=$HOME/esp/esp-idf
[image]

brianignacio5, 2021-11-24:
HI @Yecgaa1 the documentation you cited is the part of the extension setup wizard, where we try to detect an existing setup in $HOME/esp/esp-idf and if found the setup wizard will show the `Use existing setup` option. The setup wizard appears when you click on menu View -> Command Palette -> search `ESP-IDF: Configure ESP-IDF extension`.

In the [settings documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md#use-of-environment-variables-in-esp-idf-settingsjson-and-tasksjson) we specify that we don't recognize in $HOME in the extension configuration settings. Just replace it with `${env:HOME}/esp/esp-idf`. 

If you are also manually configuring the extension, please review the [Setup documentation for json manual configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md#json-manual-configuration) or I'll recommend using the setup wizard for UI configuration by clicking menu View -> Command Palette -> search `ESP-IDF: Configure ESP-IDF extension`

closed 2021-11-24 by brianignacio5

Yecgaa1, 2021-11-26:
Yes，you are right, thanks. And I want to remind others that remember to set the "Python Bin Path" and "Tools Path"