espressif/vscode-esp-idf-extension #581: [Bug Report]:

brixxx, 2021-11-25:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Trying to configure the idf-extension the "Add OpenOCD dev rules"-button leads to: 

_EISDIR: illegal operation on a directory, copyfile '/home/r/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/contrib/60-openocd.rules' -> '/etc/udev/rules.d/'_

I manually copied the rules (required sudo), but they are not recognized by VSCode and so the configuration process is stuck, even after restarting VSCode and the config process:

_sudo cp /home/r/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/contrib/60-openocd.rules /etc/udev/rules.d/_

I also tried "sudo chmod 777 /etc/udev/rules.d", but it did not change anything.

**To Reproduce**
Try to configure the plugin.

**Expected behavior**
It should be possible to continue with the configuration of the plugin.

**Environment (please complete the following information):**

- OS Name & Version: [Kubuntu 20.04]
- VSCode Version: [1.62.3]
- ESP-IDF Version: [4.3.1]
- Python Version: [3.8]

brianignacio5, 2021-11-26:
Hi @brixxx we actually removed this command with #544 because of #541 and now we just show the suggestion of the copy command you shared.

This is because the Visual Studio Code extension doesn't really has admin privileges to copy anything to `/etc/udev/rules.d/` .

What do you mean the configuration is stuck there ? If the setup wizard is showing you this Copy dev rules button, the extension should be already configured. Please share your troubleshooting information if you are having any issue.

closed 2021-11-26 by brianignacio5

brixxx, 2021-11-26:
Hi,
sorry, my fault!
The configuration dialog is stuck at the dialog.
However, the extension is indeed already usable.

Thank you very much :-)