espressif/vscode-esp-idf-extension #583: [Feature Request]: Allow us to disable WSL detection by a configuration flag

filipproch, 2021-11-27:
**Is your feature request related to a problem? Please describe.**
At the moment you are detecting a WSL distro and forcing on us different behavior - using powershell.exe
There are ways to passthrough USB devices to a WSL distro already
see https://github.com/dorssel/usbipd-win
But this prohibits the use.

**Describe the solution you'd like**
Add a configuration option to disable WSL detection.

**Describe alternatives you've considered**
Unsetting the environment variable used for detection breaks VSCode so that's not an option.

**Additional context**
The USB device is for me directly available in WSL
[image]

brianignacio5, 2021-11-29:
Hi @filipproch thanks for your feedback. We made a PR recently (#578) to use the env variable WSL_DISTRO_NAME instead of the kernel version we were using before.

Could you please try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/119078917.zip) and see if this solution works for you ?

Otherwise we could add such setting to specify use of `powershell.exe` on flash task and monitor tasks.

filipproch, 2021-12-15:
@brianignacio5 that's actually the problem - because I am running a WSL and you can't unset the WSL_DISTRO_NAME environment variable

I am passing through my USB devices to the WSL distro and need to use deploy/monitor from terminal separately, because the extension has no way to turn off the "powershell.exe" behaviour

I just want it to totally forget I am using WSL and behave like its a normal linux distro

PR #590 (2021-12-15): add wsl enable flag
Fix #583 with `idf.wslEnable` configuration setting to use powershell for flash and monitor tasks. WSL_DISTRO_NAME should also be defined as environment variable (this is done by latest WSL 2 distributions)
closed 2021-12-15 by brianignacio5