espressif/vscode-esp-idf-extension #593: [Bug Report]: NVS partition editor empty

guivo, 2021-12-15:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [X ] Make sure you have searched for existing bugs and features request before you post an issue.
- [X] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ X] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
I have a nvs.csv file in the root directory of my project. Right clicking on the file and selecting on the editor field 

**To Reproduce**
Try to open the editor.

**Expected behavior**
I would have expected the field

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS Name & Version: Windows 10
- VSCode Version: 1.63.1
- ESP-IDF Version: 4.3.1
- Python Version: 3.8.7

brianignacio5, 2021-12-15:
As specified in [NVS Partition editor doc](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/NVS_PARTITION_EDITOR.md) first 2 lines should contain:

```csv
key,type,encoding,value     <-- column header (must be the first line)
namespace_name,namespace,,  <-- First entry must be of type "namespace"
```

guivo, 2021-12-15:
The file should satisfy the the constraints:

```
key,type,encoding,value
configuration,namespace,,
board,data,string,ledstrip
version,data,u32,1
cert,file,string,rsa_cert.pem
key,file,string,rsa_private.pem
server,file,string,google_roots.pem
```

It is actually properly seen by another PC, same VS code version, from which I got the code from git.

brianignacio5, 2021-12-16:
It might be fixed by #545 in master which adds some testing to this feature. This is not yet publish to marketplace but will be released soon.

guivo, 2021-12-16:
As additional comment, if a start adding keys and values the file become usable in the interface.

However, I still cannot generate the binary file.

To stress the scenario, in case it may help, I have 2 computer:
1. Originally created the nvs.csv file, I can update the generate, the binary and flash the partition
2. Checks out the file from git, it cannot edit the file or create the binary

brianignacio5, 2021-12-16:
Can you try with this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/127236607.zip) and see if it works ?

So the idea is that if you are getting the csv file from git is not working ? Maybe there is some OS specific characters in the csv ?

guivo, 2021-12-16:
You right!

I forced the end line to LF and the NSV partition UI and ability to generate the NSV binary were restored.

closed 2021-12-16 by guivo

mbastida123, 2023-02-02:
Hi, I'm facing the same issue. The moment I change branches or anything related to git the NVS partition editor shows a blank configuration.

I see the issue is closed but it isn't clear to me if the issue was fixed.

mbastida123, 2023-02-02:
> Can you try with this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/127236607.zip) and see if it works ?
> 
> So the idea is that if you are getting the csv file from git is not working ? Maybe there is some OS specific characters in the csv ?

It is exactly that:
[image]

Using notepad I change to the unix format:
[image]

And now it works. Will you be able to fix this?

brianignacio5, 2023-04-26:
@mbastida123 Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/656267342.zip) ?