espressif/vscode-esp-idf-extension #597: [Bug Report]: ESP-IDF 4.4 doesn't work

MarkusvonStaden, 2021-12-20:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).

When trying to build anything, I get the Error Messagt
`> Executing task: cmake --build . <

[0/1] Re-running CMake...
FAILED: build.ninja 
C:\Users\marku\.espressif\tools\cmake\3.16.4\bin\cmake.exe -SC:\Users\marku\TEST -BC:\Users\marku\TEST\build
CreateProcess failed: The system cannot find the file specified.
ninja: error: rebuilding 'build.ninja': subcommand failed`
When looking into the given Path, there is no 3.16.4 CMake, but 3.20.3

**To Reproduce**
Install ESP-IDF 4.4 (I know it is not a stable version, but I think it is still relevant)
Build project

**Expected behavior**
Compiling without any error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: Windows 10
- VSCode Version: 1.63.2
- ESP-IDF Version: 4.4
- Python Version: 3.8

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Output**
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19043 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Git\cmd;C:\Program Files\MATLAB\R2020b\runtime\win64;C:\Program Files\MATLAB\R2020b\bin;C:\Program Files (x86)\PDFtk Server\bin\;C:\Program Files\nodejs\;C:\Users\marku\anaconda3;C:\Users\marku\anaconda3\Library\mingw-w64\bin;C:\Users\marku\anaconda3\Library\usr\bin;C:\Users\marku\anaconda3\Library\bin;C:\Users\marku\anaconda3\Scripts;C:\Users\marku\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\marku\AppData\Local\Programs\Python\Python39\;C:\Users\marku\AppData\Local\Microsoft\WindowsApps;C:\Users\marku\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\marku\AppData\Roaming\npm 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.63.2 
Visual Studio Code language de 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.2.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\marku\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\marku\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin;C:\Users\marku\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\marku\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\marku\.espressif\tools\riscv32-esp-elf\esp-2021r2-8.4.0\riscv32-esp-elf\bin;C:\Users\marku\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\marku\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\marku\.espressif\tools\cmake\3.20.3\bin;C:\Users\marku\.espressif\tools\openocd-esp32\v0.10.0-esp32-20211111\openocd-esp32\bin;C:\Users\marku\.espressif\tools\ninja\1.10.2;C:\Users\marku\.espressif\tools\idf-exe\1.0.2;C:\Users\marku\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\marku\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\marku\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20211111/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\marku\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM3
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\marku\.espressif
Git Path (idf.gitPath) C:\Users\marku\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\marku\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\marku\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-8.4.0\xtensa-esp32s2-elf\bin: true
[... 80 lines cut ...]
Latest error at ENOENT: no such file or directory, open 'c:\Users\marku\TEST\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

Click the menu `View` and select `Output`. This extension uses several output channels: ESP-IDF, ESP-IDF Debug Adapter, Heap Trace, OpenOCD and SDK Configuration Editor.

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1 or View Menu -> Command Palette) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows): "C:\\Users\\marku\\esp\\esp-idf"
- `idf.pythonBinPath`: "C:\\Users\\marku\\.espressif\\python_env\\idf4.4_py3.8_env\\Scripts\\python.exe"
- `idf.openOcdConfigs` : [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "target/esp32.cfg"
    ]
- `idf.customExtraPaths`: "C:\\Users\\marku\\.espressif\\tools\\xtensa-esp32-elf\\esp-2021r2-8.4.0\\xtensa-esp32-elf\\bin;C:\\Users\\marku\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2021r2-8.4.0\\xtensa-esp32s2-elf\\bin;C:\\Users\\marku\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2021r2-8.4.0\\xtensa-esp32s3-elf\\bin;C:\\Users\\marku\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r2-8.4.0\\riscv32-esp-elf\\bin;C:\\Users\\marku\\.espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Users\\marku\\.espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Users\\marku\\.espressif\\tools\\cmake\\3.20.3\\bin;C:\\Users\\marku\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20211111\\openocd-esp32\\bin;C:\\Users\\marku\\.espressif\\tools\\ninja\\1.10.2;C:\\Users\\marku\\.espressif\\tools\\idf-exe\\1.0.2;C:\\Users\\marku\\.espressif\\tools\\ccache\\4.3\\ccache-4.3-windows-64;C:\\Users\\marku\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
- `idf.customExtraVars`: "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\marku\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20211111/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}"
- `idf.adapterTargetName`: "esp32"

brianignacio5, 2021-12-21:
Can you try to delete the build folder and try to build again? maybe there is some older CMake cache version which has conflict with current configuration.

MarkusvonStaden, 2021-12-21:
Already tried that before, that didn't help. Now I also deleted the .vscode folder and somehow that worked.

closed 2021-12-21 by MarkusvonStaden

brianignacio5, 2021-12-22:
Might have been a local settings (the project .vscode directory) overriding user settings which made conflicts to your build environment.

andresbarrer, 2023-11-02:
> Already tried that before, that didn't help. Now I also deleted the .vscode folder and somehow that worked. (At least I think that was, what solved it. I also uninstalled the IDF and installed it again as admin)

I just spent 2 hours trying to fix this, thank you. Deleting the .vscode folder for some reason worked.