espressif/vscode-esp-idf-extension #598: IDF extension installation does not end:

sergioaiello, 2021-12-26:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Even though IDF extension installation ("ESP-IDF setup" tab) ends correctly with the sentence "all settings have been configured. you can close this window", at project next re-opening VSC warns "Command failed: idf.py.exe -v error checking idf-exe version Error: command failed: idf.py.exe -v" in the "OUTPUT" tab and re-open the "ESP-IDF setup" tab.

**To Reproduce**
Opening a project with VSC.

**Expected behavior**
no "Command failed: idf.py.exe -v .... " warning in the "OUTPUT" tab
and
no "ESP-IDF setup" tab opening

**Screenshots**
N.A.

**Environment (please complete the following information):**
[Doctor.txt](https://github.com/espressif/vscode-esp-idf-extension/files/7776299/Doctor.txt)
> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Output** (the "OUTPUT" tab listing follows)
Command failed: idf.py.exe -v
error checking idf-exe version
Error: command failed: idf.py.exe -v

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process.

**Logs**
- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/7776423/esp_idf_vsc_ext.log)

**Extension Configuration settings**
Open Command Palette (F1 or View Menu -> Command Palette) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.
[idf_versions.txt](https://github.com/espressif/vscode-esp-idf-extension/files/7776424/idf_versions.txt)

brianignacio5, 2021-12-28:
This is an issue with `idf-exe` itself. There is a couple of things you can do:

1) Replace the idf.py.exe executable with the one in  [idf-exe v1.0.3](https://github.com/espressif/idf_py_exe_tool/releases/download/v1.0.3/idf-exe-v1.0.3.zip)
2) Try using esp-idf master which already has tool dependency updated for **idf-exe v1.0.3** as done in 1)
3) Disable show extension setup with `idf.showOnboardingOnInit: false`.

closed 2021-12-28 by brianignacio5

sergioaiello, 2021-12-29:
Hello
Thank you for the reply.
1st step:
replacing the idf.py.exe rapper from v1..0.2 to v1.0.3 (as per link here above) solved both warning in the "OUTPUT" tab and "ESP-IDF setup" tab (in VSC main section) does not show up anymore. I confirm the ESP_IDF release under test: 4.4 branch release
In the first step I did not change the "idf.showOnboardingOnInit" value from "true"
2nd step
changing the "idf.showOnboardingOnInit" value to "false" I did not observe any changes from step 1 here above
3rd step
replacing back the idf.py.exe rapper from v1..0.3 to v1.0.2 I got again the warning in the "OUTPUT" tab and the "ESP-IDF setup" tab (in the main section) kept on not showing up.
4th step
changing back the "idf.showOnboardingOnInit" value to "true" I did not observe any changes
Therefore with setup restored as in the initial conditions I have the following result:
1.
the "ESP-IDF setup" tab does not show up anymore (could you figure out the reason of this?)
2.
warning in the "OUTPUT" tab ("Command failed: idf.py.exe -v ......") showed up again

Regarding to point "1", looking at ESP-IDF VS Code Extension  ("Details" tab) I see "version 1.3.0" but in the Doctor.txt I reported 1.2.0. Maybe the "idf.py.exe" change caused a permanent VSC extension's update as well? Would this give an explanation of improvement where "idf.py.exe" file was restored to initial 1.0.2?

brianignacio5, 2022-01-19:
Hey @sergioaiello sorry for late reply.

In ESP-IDF there is a file called tools.json which specifies the required version of each tool as you can see [here](https://github.com/espressif/esp-idf/blob/master/tools/tools.json#L681). In the latest versions the idf.py.exe version should have been updated to use the 1.0.3 version.

In Visual Studio Code, idf.py menuconfig running from terminal have been reported to not work properly. You can either use an external terminal or use the extension `ESP-IDF: SDK Configuration editor` command.