espressif/vscode-esp-idf-extension #600: [Feature Request]: open-vsx support for use with Theia

nated0g, 2021-12-29:
I am trying to bundle an IDE with a Pi based gateway that manages a fleet of ESP-IDF based devices.  I plan to use Theia for this. I watched the presentation @brianignacio5 did as part of EclipseCon2021 and was very interested. I could not find any further information about integrating ESP-IDF and Theia.

Should it be possible to use this extension as is with Theia?  If so, support for open-vsx would be great to ease installation.

brianignacio5, 2022-01-04:
Hi @nated0g thank you for your feature request.

We have the extension vsix installer release for open-vsx in the [Github Releases](https://github.com/espressif/vscode-esp-idf-extension/releases) which is basically the extension without Microsoft C/C++ Dependency. We have a Theia + ESP-IDF docker image which has the IDE with the features shown in the EclipseCON 2021 but we haven't released yet, will plan to release it soon to our Github too.

At some point we also will release to the open-vsx registry also, when we sort out some details internally.

closed 2022-01-26 by github-actions[bot]

brianignacio5, 2022-02-21:
Open-VSX release [here](https://open-vsx.org/extension/espressif/esp-idf-extension)