espressif/vscode-esp-idf-extension #601: [Feature Request]: ESP32-C3-DevKitM-1 question.

Amtran7599, 2022-01-01:
I have built esp-idf environment,i could build when device target is esp32.However I couldn't build if I set device target to esp32c3.

I want to know  what else should I set for built and flash in esp32c3

Add any other context or screenshots about the feature request here.
This is wrong report when I set device to esp32c3.
[Error: non zero exit code 2 Usage: C:\Users\Amtran\esp\esp-idf\tools\idf.py set-target [OPTIONS] {esp32|esp32s2} Try 'C:\Users\Amtran\esp\esp-idf\tools\idf.py set-target --help' for help. Error: Invalid value for '{esp32|esp32s2}': 'esp32c3' is not one of 'esp32', 'esp32s2'. at ChildProcess.<anonymous> (c:\Users\Amtran\.vscode\extensions\espressif.esp-idf-extension-1.3.0\dist\extension.js:1:4394) at ChildProcess.emit (events.js:315:20) at ChildProcess.EventEmitter.emit (domain.js:467:12) at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)]

brianignacio5, 2022-01-04:
Hi @Amtran7599 , the issue is that you are using an ESP-IDF version which doesn't have ESP32-C3 support. 

Please update your ESP-IDF version to enable esp32c3 as possible IDF_TARGET.

brianignacio5, 2022-01-10:
Closing this issue for lack of activity. Please comment here if you still have any issues.

closed 2022-01-10 by brianignacio5