espressif/vscode-esp-idf-extension #602: restart Visual Studio Code and run this wizard again.

FanhuaCloud, 2022-01-01:
When I use the plugin to configure esp-idf, the plugin tells me this prompt

FanhuaCloud, 2022-01-01:
When I use the old version 1.2.0 to install, there is no such problem

gaupen1186, 2022-01-01:
> 当我使用插件配置esp-idf时，插件告诉我这个提示

我也遇到了

FanhuaCloud, 2022-01-01:
> > 当我使用插件配置esp-idf时，插件告诉我这个提示
> 
> 我也遇到了

可以先回退安装完再升级

js4iot, 2022-01-01:
I was having the same issues, and I followed these instructions. Now it works!

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#esp-idf-tools-installer

FanhuaCloud, 2022-01-01:
It seems that we need to wait for the development team to fix this problem...

spaceMan00, 2022-01-03:
I am having the same problem with the 1.3.0 release on Mac OS. Everything works, I can build, flash monitor etc. on VS Code. But it's just whenever I try to Configure ESP-IDF extension I get the "restart Visual Studio Code and run this wizard again.". 

Rolling back to 1.2.0 it works, and I don't get this prompt.

brianignacio5, 2022-01-04:
Could you share the extension log please, it seems there is a tool validation that is failing (something that needs to be on PATH but is not being found)

* Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
* Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

solhuebner, 2022-01-05:
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/7817698/esp_idf_vsc_ext.log)

brianignacio5, 2022-01-06:
Can you provide more information about you setup @spaceMan00 ?

Basically we show tis restart Visual Studio Code and run this wizard again if prerequisites are not met. For MacOS/Linux this means to find the git version and some python versions. With the doctor command output it might be helpful to see the issue. Most likely you just need to update `idf.gitPath` from `git` to `/usr/bin/git` since now git version validation needs full git path.

brianignacio5, 2022-01-06:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/137655771.zip) ? @solhuebner

solhuebner, 2022-01-06:
@brianignacio5 I uninstalled the extension in VS Code, removed the esp and .espressif in my user folder and removed the .vscode and build folder in my project to have a clean plate. And it seems fine with the version you provided. It is currently installing release/v4.4 (release branch) 👍

solhuebner, 2022-01-06:
Worked! I got the success screen :)
The only thing I saw in the console were a few "WARNING: You are using pip version 20.3.3; however, version 21.3.1 is available."
Setting device target for my project worked too.
It compiled fine (I really like the new memory overview at the end btw).

brianignacio5, 2022-01-06:
You could always remove this warning by Open an ESP-IDF Terminal and running `python -m pip install --upgrade pip` to update pip within the virtual env but it should not break anything.

We-Sky, 2022-01-07:
I met this problem too

brianignacio5, 2022-01-07:
Can you provide more information about your setup @We-Sky ? You can also try the [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/137655771.zip) ?

solhuebner, 2022-01-07:
@We-Sky  [image]

spaceMan00, 2022-01-07:
@brianignacio5 Find attached my Doctor command output as well as the extension log. Please let me know what the issue might be from my side. 
[IDF_ext_doctor.txt](https://github.com/espressif/vscode-esp-idf-extension/files/7828524/IDF_ext_doctor.txt)
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/7828528/esp_idf_vsc_ext.log)

brianignacio5, 2022-01-07:
For some reason you have some default values in your settings.json @spaceMan00 .

In the doctor output, notice that `idf.espIdfPath` and `idf.gitPath` are defined with `{env:IDF_PATH}` and `git` respectively. Maybe you are overriding these settings in a local `<your-project-dir>/.vscode/settings.json` ?

brianignacio5, 2022-01-10:
@FanhuaCloud have you tried the [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/137655771.zip) ?

FanhuaCloud, 2022-01-10:
> @FanhuaCloud have you tried the [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/137655771.zip) ?

yes the problem is solved when i use that vsix ,thank you!

brianignacio5, 2022-01-10:
Cool! Then I'll close this issue. If any of you have any further issue please comment here (if related to this) or open a new issue. 

Let us know how it goes @spaceMan00

closed 2022-01-10 by brianignacio5

fronders, 2022-01-20:
I had the same issue for WSL2. Fixed by changing but the `idf.gitPath` from `git` to `/usr/bin/git`

matthew798, 2022-01-29:
Hello,

I am having the same issue. It's worth noting that I have the IDF already installed (manually) and I just renamed the folder to esp.back. I was hoping to try it out through the plugin.

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.4.0-91-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /home/matthew/.local/bin:/home/matthew/.nvm/versions/node/v17.1.0/bin:/home/matthew/.cargo/bin:/home/matthew/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/matthew/Android/Sdk/platform-tools 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.63.2 
Visual Studio Code language en 
Visual Studio Code shell bash 
ESP-IDF Extension version 1.3.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /home/matthew/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) /home/matthew/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin:/home/matthew/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin:/home/matthew/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin:/home/matthew/.espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin:/home/matthew/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/matthew/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/matthew/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210902/openocd-esp32/bin
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/home/matthew/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210902/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /home/matthew/.espressif/python_env/idf4.4_py3.8_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) /home/matthew/.espressif
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to /home/matthew/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin: true
Access to /home/matthew/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin: true
Access to /home/matthew/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin: true
Access to /home/matthew/.espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin: true
Access to /home/matthew/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to /home/matthew/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to /home/matthew/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210902/openocd-esp32/bin: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.25.1
ESP-IDF version x.x
Python version 3.8.10
Python's pip version 21.3.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitstring version: 3.1.9
Brotli version: 1.0.9
certifi version: 2021.10.8
cffi version: 1.15.0
charset-normalizer version: 2.0.7
click version: 8.0.3
construct version: 2.10.54
contextlib2 version: 21.6.0
cryptography version: 35.0.0
ecdsa version: 0.17.0
Flask version: 0.12.5
Flask-Compress version: 1.10.1
Flask-SocketIO version: 2.9.6
future version: 0.18.2
gcovr version: 5.0
gdbgui version: 0.13.2.0
gevent version: 1.5.0
greenlet version: 1.1.2
idf-component-manager version: 0.3.2b0
idna version: 3.3
[... 17 lines cut ...]
requests version: 2.26.0
requests-toolbelt version: 0.9.1
schema version: 0.7.4
semantic-version version: 2.8.5
setuptools version: 58.5.3
six version: 1.16.0
tqdm version: 4.62.3
urllib3 version: 1.26.7
websocket-client version: 1.2.1
Werkzeug version: 0.16.1
wheel version: 0.37.0
xmlrunner version: 1.7.7
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Error
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Error
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Error
----------------------------------------------------------- Latest error -----------------------------------------------------------------
[... cut ...] thon /home/matthew/esp/esp-idf/tools/check_python_dependencies.py -r "/home/matthew/.vscode/extensions/espressif.esp-idf-extension-1.3.0/esp_debug_adapter/requirements.txt"
/home/matthew/.espressif/python_env/idf4.4_py3.8_env/bin/python: can't open file '/home/matthew/esp/esp-idf/tools/check_python_dependencies.py': [Errno 2] No such file or directory

--------------------------------------------------------------------------------------------------------------------------------------------

```

brianignacio5, 2022-01-29:
Make sure that idf.espIdfPath has a correct value (Current value is /home/matthew/esp/esp-idf) and change the idf.gitPath from git to /usr/bin/git in your settings.json like @fronders did :)

tigerwang202, 2022-01-30:
> 

It seams solve the problem.

xobs, 2022-02-15:
This issue still seems to be occurring -- I just installed `v1.3.0` from inside vscode, which appears to be the latest version that is released.

Will there be a new version released anytime soon? Or is the official suggestion to sideload the `.vsix` file since `v1.3.0` does not appear to work.

brianignacio5, 2022-02-15:
the thing is we are trying to fix as many important issues as possible before releasing the next version, stuff like Esp32 S3 debug which requires a bit more testing and multiple projects support issue.

Apologies for the troubles @xobs

ua100k, 2022-02-16:
Why don't you remove the v1.3.0 extension from VS code. It is causing a lot of wasted time with those who think if it is there it should work.  It seems you are getting a black eye if you keep it up there.

ademyankov, 2022-02-17:
I agree... this stuff is simply unusable... It is not an extension but a debugging challenge! I tried everything... Installing IDF manually and setting up all the env variables... installing 1.2 version... nothing works... it keeps finding new issues to give me a headache... 

[image]

Did anyone was able to make it work somehow?

ademyankov, 2022-02-17:
Thanks... Finally, I was able to install and configure everyhting and it works now. For some reason espressif extension (1.2) was not able to see the latest version of C/C++ extenstion. I had to down version it to v1.8.4.

TylerCode, 2022-02-22:
Had this same issue on VSCode 1.64 and 1.65-Insider

Using the vsix fixed the issue on both. Might be worth fixing from the extension explorer but wanted to put my version on the pile.

Paulo-Pecegueiro, 2022-03-23:
> I had the same issue for WSL2. Fixed by changing but the `idf.gitPath` from `git` to `/usr/bin/git`
Hi, Can you explain to me what exactly you did to solve this problem for WSL2?
First of all, I'm not a linux user, so I don't really have experience with this.
I have successfully installed WSL2.
I've already installed the requirements through Ubuntu's terminal and reloaded VS Code several times, but the IDF extension in VS Code is still notifying:
"First, install the [ESP-IDF Prerequisites for Linux]
restart Visual Studio Code and run this wizard again."

fronders, 2022-03-24:
@Paulo-Pecegueiro you simply need to open VSCode settings, specifically Extensions -> ESP-IDF. Then look for Idf: Git Path - there I had value of just `git` and changed it to `/usr/bin/git`

[image]

If you don't know you git path - you can launch WSL and execute `which git` to get your git path
[image]

GitProdEnv, 2022-06-01:
I can confirm it, that it is required to add 

```json
"idf.gitPath": "/usr/bin/git"
```

to `.vscode/settings.json`. Then you can proceed to the setup stage. Maybe you could show some sort of warning, this is hard to debug.

brianignacio5, 2022-06-02:
Seems that this settings also has to have a Windows and MacOS/Linux settings to avoid this issue in scenarios such as WSL which still picks the Windows settings here.

hzmacso, 2022-10-18:
Hi I have the same issue for macOS
1. I have installed ESP-IDF on macOS vis Terminal, and I am able to build projects via Terminal, but cannot configure the VSCode Extension  on macOS
2. I tried vsix installer and still get the same issue 

[image]

brianignacio5, 2022-10-18:
There is a bug fix in #821 related to this. Will fix soon !

Veldmonster, 2022-10-23:
I found on Windows 10 this happens if I have "IDF_TOOLS_PATH" environment variable defined as a result of an earlier installation. Rename the environment variable and try again.