espressif/vscode-esp-idf-extension #603: [Bug Report]: "Failed to init Debug Adapter!" at ESP32-S3-DevKit C-1 with FT232H

TakuyaAtMono, 2022-01-04:
**Pre Bug Report Checklist**

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**

"Failed to init Debug Adapter!" at ESP32-S3-DevKit C-1 with FT232H

**Environment (please complete the following information):**

- OS win32 x64 10.0.19043 
- Visual Studio Code version 1.63.2 
- ESP-IDF Extension version 1.3.0
- ESP-IDF Version: release/v4.4 (release branch)
- Python version 3.8.7

- Hardware
	- ESP32-S3-DevKit C-1
		- eFuse "STRAP_JTAG_SEL" is burned and GPIO3 connected to GND to select external jtag.
	- Adafruit FT232H( https://www.adafruit.com/product/2264 )

- Using example project "blink".

- Additional context
	- "ESP-IDF:build, Flash snd start a monitor on your device" is working collectly.
	- GDB debugging from the command line is working at same environment.
	- Debugging with Eclipse and Espressif IDF Plugin for Eclipse is working at same environment.

launch.json
```
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch"
    }
  ]
} 
```

and attached ESP-IDF_Doctor_command_output.txt
[ESP-IDF_Doctor_command_output.txt](https://github.com/espressif/vscode-esp-idf-extension/files/7808737/ESP-IDF_Doctor_command_output.txt)

**To Reproduce**

- install, make "blink" project from example, some configrations and test with commandline GDB. and press F5.

**Output**

ESP-IDF Debug Adapter
```
DEBUG_ADAPTER_STARTED
2022-01-05 00:27:59,483 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
DEBUG_ADAPTER_READY2CONNECT
2022-01-05 00:27:59,483 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
DEBUG_ADAPTER_STOPPED
2022-01-05 00:27:59,565 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Stopped] : ESP-IDF Debug Adapter
```

OpenOCD
```
Open On-Chip Debugger  v0.10.0-esp32-20211111 (2021-11-10-21:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

adapter speed: 20000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: no device found
❌ Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Target halted, PC=0x42017E26, debug_reason=00000001
Info : esp32s3.cpu1: Target halted, PC=0x42017E26, debug_reason=00000000
Info : Listening on port 3333 for gdb connections
```

**Extension Configuration settings**

Preferences: Open Settings (JSON)
```
{
    "idf.adapterTargetName": "esp32s3",
    "idf.gitPath": "C:\\Users\\kaminosono\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe",
    "idf.espIdfPathWin": "C:\\Users\\kaminosono\\esp\\esp-idf",
    "idf.pythonBinPathWin": "C:\\Users\\kaminosono\\.espressif\\python_env\\idf4.4_py3.8_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "C:\\Users\\kaminosono\\.espressif",
    "idf.customExtraPaths": "C:\\Users\\kaminosono\\.espressif\\tools\\xtensa-esp32-elf\\esp-2021r2-8.4.0\\xtensa-esp32-elf\\bin;C:\\Users\\kaminosono\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2021r2-8.4.0\\xtensa-esp32s2-elf\\bin;C:\\Users\\kaminosono\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2021r2-8.4.0\\xtensa-esp32s3-elf\\bin;C:\\Users\\kaminosono\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r2-8.4.0\\riscv32-esp-elf\\bin;C:\\Users\\kaminosono\\.espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Users\\kaminosono\\.espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Users\\kaminosono\\.espressif\\tools\\cmake\\3.20.3\\bin;C:\\Users\\kaminosono\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20211111\\openocd-esp32\\bin;C:\\Users\\kaminosono\\.espressif\\tools\\ninja\\1.10.2;C:\\Users\\kaminosono\\.espressif\\tools\\idf-exe\\1.0.3;C:\\Users\\kaminosono\\.espressif\\tools\\ccache\\4.3\\ccache-4.3-windows-64;C:\\Users\\kaminosono\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\kaminosono\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20211111/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "target/esp32s3.cfg"
    ],
    "idf.portWin": "COM3",
    "idf.flashType": "UART",
    "cmake.configureOnOpen": true,
    "extensions.ignoreRecommendations": true
}
```

**Screenshots**

- img1_work collectly with build flash and monitor.png
[image]
- img2_work collectly with command line GDB.png
[image]
- img3_work collectly with eclipse.png
[image]
- img4_NOT work with vscode extention.png
[image]

brianignacio5, 2022-01-05:
Could you change 

```
"idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "target/esp32s3.cfg"
    ],
```

to 

```
"idf.openOcdConfigs": [
        "board/esp32s3-ftdi.cfg
    ],
```

The issue seems to be starting openOCD from Visual Studio Code since openOCD config files don't match the ones you are using in the command line.

```
 Error: no device found
❌ Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
```

TakuyaAtMono, 2022-01-06:
Thank you for your prompt support!
I tried it but it did not solve the problem.

As for the vid 0403, pid 6010 error, openOCD is searching FT2232H and FT232H in order, and FT2232H (vid 0403, pid 6010) is not found, but FT232H (vid 0403, pid 6014) is found afterwards, and "Info : Listening on port 3333 for gdb connections", so I am guessing that it is not related to this problem.
In fact, the GDB debug log also outputs the error that vid 0403, pid 6010 is not found.

[image]

Setting 1
"interface/ftdi/esp32_devkitj_v1.cfg",
"target/esp32s3.cfg"
The log of OUTPUT>OpenOCD when press F5 is below
```
Open On-Chip Debugger  v0.10.0-esp32-20211111 (2021-11-10-21:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

adapter speed: 20000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Listening on port 3333 for gdb connections
```

Setting 2
"board/esp32s3-ftdi.cfg"
The log of OUTPUT>OpenOCD when press F5 is below
```
Open On-Chip Debugger  v0.10.0-esp32-20211111 (2021-11-10-21:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

adapter speed: 5000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Listening on port 3333 for gdb connections
```

What can I try next?

TakuyaAtMono, 2022-01-22:
Hi @brianignacio5 ,

After that, I still get the same error.
Is there anything we can try or check?
If it helps to change launch.json, as the error dialog suggests, it would be helpful to have some advice for that change.

brianignacio5, 2022-01-24:
Sorry for late reply @TakuyaAtMono 

Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/147436665.zip) and see if it is better ? We added some fixes in debug adapter in #617

closed 2022-01-29 by brianignacio5

TakuyaAtMono, 2022-01-29:
Thank you for the information.
I have not been able to try yet. I will check next week and report back.

TakuyaAtMono, 2022-01-31:
Hi @brianignacio5  ,

I was able to confirm that VScode debugging works using the vsix installer that you introduced.
Thank you very much for your help.
If there is any information that I should check and share on my end for you, please let me know.

homerliu1, 2022-10-28:
Same error , "Failed to init Debug Adapter!", happens for
ESP32-S3-DevKitC-1-N8R2
builtin usb-jteg. Any idea?

brianignacio5, 2022-10-28:
Could you set `logLevel:4` in your project's `.vscode/launch.json` and share openOCD and debug adapter output @homerliu1 ?

homerliu1, 2022-11-22:
In my case, I solved the problem by install python2.7. 
It seems ESP-IDF vs code extension is using Python 2.7.

brianignacio5, 2022-11-22:
This changes should be already be in Marketplace release. ESP-IDF vscode extension require Python 3  or higher. 

Could you share the troubleshooting information ? Maybe there was an issue in your setup configuration.

matvictor, 2022-11-23:
I've downloaded the universal installer (esp framework v4.4.2). Then went in vscode and installed the ESP-IDF extention using my existing installation. Wrote my  program and then tried to lauch the OCD server. But I allways get the failed to init debug adapter. 
Maybe I need to change something int the lauch.json in order to make it work.

brianignacio5, 2022-11-23:
Follow the [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) so we can understand what is happening in your setup.

In the launch.json change the `logLevel` to `4` and share the former and the openOCD output in the `ESP-IDF` Output channel.

matvictor, 2022-11-23:
[report.txt](https://github.com/espressif/vscode-esp-idf-extension/files/10073767/report.txt)
Here's my openOCD output :

[OpenOCD]
Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
[OpenOCD]
debug_level: 2

[OpenOCD]
Info : only one transport option; autoselect 'jtag'
[OpenOCD]
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: target chip id set to 9
[OpenOCD]
Warn : Transport "jtag" was already selected
[OpenOCD]
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
[OpenOCD]
❌ Error: esp_usb_j[OpenOCD]
tag: could not find or open device!

Assertion failed!

Program: C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin\openocd.exe
File: ../src/jtag/core.c, Line 338

Expression: jtag_trst == 0
[OpenOCD]
[Stopped] : OpenOCD Server
[Debug Adapter]
2022-11-23 09:33:08,427 - Debug Adapter (main) - DEBUG - START_TIME_2022_11_23_09-33-08
2022-11-23 09:33:08,428 - Debug Adapter (main) - DEBUG - Init errors:None
2022-11-23 09:33:08,428 - Debug Adapter (main) - INFO - Working directory: C:\Users\matvictorino\AppData\Local\Programs\Microsoft VS Code
[Debug Adapter]
2022-11-23 09:33:08,428 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 4,
    'developer_mode': 'none',
    'device_name': 'esp32s3',
    'elfpath': (   'c:\\Espressif\\frameworks\\esp-idf-v4.4.2\\projects\\Co2metre2022\\build\\main.elf',),
    'log_file': 'c:\\Espressif\\frameworks\\esp-idf-v4.4.2\\projects\\Co2metre2022\\debug.log',
    'log_mult_files': None,
    'log_no_debug_console': None,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 43474,
    'postmortem': False,
    'tmo_scale_factor': 1,
    'toolchain_prefix': 'xtensa-esp32s3-elf-'}
DEBUG_ADAPTER_STARTED
2022-11-23 09:33:08,428 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
2022-11-23 09:33:08,428 - Debug Adapter (main) - INFO - Starting. Cmd: c:\Users\matvictorino\.vscode\extensions\espressif.esp-idf-extension-1.5.1\esp_debug_adapter\debug_adapter_main.py -d 4 -e c:\Espressif\frameworks\esp-idf-v4.4.2\projects\Co2metre2022\build\main.elf -l c:\Espressif\frameworks\esp-idf-v4.4.2\projects\Co2metre2022\debug.log -p 43474 -dn esp32s3 -a 0x10000 -t xtensa-esp32s3-elf-

[Debug Adapter]
2022-11-23 09:33:08,431 - Debug Adapter (main) - INFO - Listening on port 43474
[Debug Adapter]
DEBUG_ADAPTER_READY2CONNECT
2022-11-23 09:33:08,431 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
[Debug Adapter]
[... 156 lines cut ...]

2022-11-23 09:33:18,546 - Debug Adapter (main) - DEBUG - Stopping target poller
[Debug Adapter]
2022-11-23 09:33:18,546 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Adapter is Stopping...\n", "source": {}}, "seq": 12}

2022-11-23 09:33:18,546 - Debug Adapter (main) - DEBUG - Closing the GDB process
[Debug Adapter]
2022-11-23 09:33:18,546 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Stopping target poller\n", "source": {}}, "seq": 13}

2022-11-23 09:33:18,546 - Debug Adapter (main) - DEBUG - Closing the Socket
[Debug Adapter]
2022-11-23 09:33:18,546 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Closing the GDB process\n", "source": {}}, "seq": 14}

2022-11-23 09:33:18,546 - Debug Adapter (main) - DEBUG - Stopping of the Writer thread
[Debug Adapter]
2022-11-23 09:33:18,546 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Closing the Socket\n", "source": {}}, "seq": 15}

[Debug Adapter]
2022-11-23 09:33:18,547 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread
2022-11-23 09:33:18,547 - Debug Adapter (main) - DEBUG - Adapter is Stopped
DEBUG_ADAPTER_STOPPED
2022-11-23 09:33:18,547 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Debug Adapter]
2022-11-23 09:33:18,547 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect
[Stopped] : ESP-IDF Debug Adapter

matvictor, 2022-11-23:
Theses are my "Developer tools" console output errors when I try to open an openOCD server.

[[object Object]]OpenOCD server failed to start because of Error: esp_usb_jtag: could not find or open device!

Error: OpenOCD server failed to start because of Error: esp_usb_jtag: could not find or open device!
	at Socket.<anonymous> (\dist\extension.js:1:101322)
	at Socket.emit (node:events:526:28)
	at addChunk (node:internal/streams/readable:315:12)
	at readableAddChunk (node:internal/streams/readable:289:9)
	at Readable.push (node:internal/streams/readable:228:10)
	at Pipe.onStreamRead (node:internal/stream_base_commons:190:23)

PR #617 (2022-01-29): update debug adapter to use generic chip classes
Support debug backend chip generic classes. Now for chips which has no explicit GDB/OOCD class in debug backend generic ones are used basing on provided target triple extracted from toolchain prefix. So it is mandatory to provide toolcahin prefix (--toolchain-prefix argument) for debugging new chips.

Fix #542 

Fix #603