espressif/vscode-esp-idf-extension #605: Vs Code ESP IDF build error (IEP-613)

emhatek, 2022-01-10:
Hi All,
i got a problem during workspace build project as below:
 Executing task: ninja  <

ninja: error: loading 'build.ninja': The system cannot find the file specified.

The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.

> Executing task: ninja  <

ninja: error: loading 'build.ninja': The system cannot find the file specified.

The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.

Anyone has experince like this, how to resolve?

emhatek, 2022-01-10:
Hi  @kolipakakondal,
i'm already install manual the esp-IDF tools also.
replace Ninja sofware by manually download from your link, however still error with same message.

> Hi @emhatek
> 
> Are you facing this issue with the Eclipse plugin or VSCode extension? If it's from VSCode we could move this to https://github.com/espressif/vscode-esp-idf-extension
> 
> > ninja: error: loading 'build.ninja': The system cannot find the file specified.
> > The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
> 
> This error could be because PowerShell is unable to find the ninja executable. Could you try to install them separately and see if that helps https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup-scratch.html#ninja-build https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32s2/api-guides/tools/idf-tools.html#ninja

brianignacio5, 2022-01-10:
@emhatek Could you please share the required troubleshooting information?

https://github.com/espressif/vscode-esp-idf-extension#troubleshooting

Did you follow the extension setup ?

emhatek, 2022-01-10:
@brianignacio5, yes i do.
however my windows version is Windows 11 Home 21H2. does any body try at this latest windows?

brianignacio5, 2022-01-10:
Could you please share the required [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) ?

brianignacio5, 2022-01-19:
Closing this issue for lack of response from the user. Please comment here if you still have any issue ?

closed 2022-01-19 by brianignacio5

bb250250, 2022-06-13:
I meet this question too ,Are you solve this question?

brianignacio5, 2022-06-14:
Could you share the troubleshooting information @bb250250 ?

emhatek, 2022-06-14:
> Could you share the troubleshooting information @bb250250 ?

Dear Brian, let me check the detail to my desktop. 
I'm forgot the detail, however i'm remember re try the manual install downloads from the file from espressif.

saqibshafin, 2022-08-06:
Never, ever, have a folder name which consists of spaces or any special characters, within your directory.

Worked for me, when I removed spaces.

brianignacio5, 2022-08-08:
I'm sorry @saqibshafin but as specified in the ESP-IDF and this extension documentation, spaces in paths are not supported yet in the extension and build system yet. It is going to be supported in the future but so far not yet.

saqibshafin, 2022-08-08:
Thank you so much for your reply, dear. I figured it after some setbacks.
But thanks a lot, for your response.

Take care.

nahuel-uranga-sierra, 2022-10-31:
C:\Users\nahue\Documents\VS Code\ESP32\Blink LED\blink>idf.py build
Executing action: all (aliases: build)
Running cmake in directory 'c:\users\nahue\documents\vs code\esp32\blink led\blink\build'
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 'c:\users\nahue\documents\vs code\esp32\blink led\blink'"...
-- ccache will be used for faster recompilation
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/nahue/Documents/VS Code/ESP32/Blink LED/blink/sdkconfig
Traceback (most recent call last):
  File "C:/Espressif/frameworks/esp-idf-v4.4.2/tools/kconfig_new/confgen.py", line 640, in <module>
    main()
  File "C:/Espressif/frameworks/esp-idf-v4.4.2/tools/kconfig_new/confgen.py", line 260, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 947, in __init__
    self._init(filename, warn, warn_to_stderr, encoding)
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 1085, in _init
    self._parse_block(None, self.top_node, self.top_node).next = None
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 2982, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 2970, in _parse_block
    raise KconfigError(
kconfiglib.KconfigError: C:/Users/nahue/Documents/VS Code/ESP32/Blink LED/blink/build/kconfigs_projbuild.in:6: 'C:/Users/nahue/Documents/VS' not found (in 'source "C:/Users/nahue/Documents/VS"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
CMake Error at C:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/kconfig.cmake:266 (message):
  Failed to run confgen.py
  (C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe;C:/Espressif/frameworks/esp-idf-v4.4.2/tools/kconfig_new/confgen.py;--kconfig;C:/Espressif/frameworks/esp-idf-v4.4.2/Kconfig;--sdkconfig-rename;C:/Espressif/frameworks/esp-idf-v4.4.2/sdkconfig.rename;--config;C:/Users/nahue/Documents/VS
  Code/ESP32/Blink LED/blink/sdkconfig;--defaults;C:/Users/nahue/Documents/VS
  Code/ESP32/Blink
  LED/blink/sdkconfig.defaults;--env-file;C:/Users/nahue/Documents/VS
  Code/ESP32/Blink LED/blink/build/config.env).  Error 1
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/build.cmake:525 (__kconfig_generate_config)
  C:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/nahue/Documents/VS Code/ESP32/Blink LED/blink/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

> Could you share the troubleshooting information @bb250250 ?

nahuel-uranga-sierra, 2022-10-31:
> Could you share the troubleshooting information @bb250250 ?

Hi @brianignacio5, 
I followed your steps and try to build the project from the command line.
This is what happened:

 C:\Users\nahue\Documents\VS Code\ESP32\Blink LED\blink>idf.py build
Executing action: all (aliases: build)
Running cmake in directory 'c:\users\nahue\documents\vs code\esp32\blink led\blink\build'
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 'c:\users\nahue\documents\vs code\esp32\blink led\blink'"...
-- ccache will be used for faster recompilation
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/nahue/Documents/VS Code/ESP32/Blink LED/blink/sdkconfig
Traceback (most recent call last):
  File "C:/Espressif/frameworks/esp-idf-v4.4.2/tools/kconfig_new/confgen.py", line 640, in <module>
    main()
  File "C:/Espressif/frameworks/esp-idf-v4.4.2/tools/kconfig_new/confgen.py", line 260, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 947, in __init__
    self._init(filename, warn, warn_to_stderr, encoding)
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 1085, in _init
    self._parse_block(None, self.top_node, self.top_node).next = None
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 2982, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 2970, in _parse_block
    raise KconfigError(
kconfiglib.KconfigError: C:/Users/nahue/Documents/VS Code/ESP32/Blink LED/blink/build/kconfigs_projbuild.in:6: 'C:/Users/nahue/Documents/VS' not found (in 'source "C:/Users/nahue/Documents/VS"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
CMake Error at C:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/kconfig.cmake:266 (message):
  Failed to run confgen.py
  (C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe;C:/Espressif/frameworks/esp-idf-v4.4.2/tools/kconfig_new/confgen.py;--kconfig;C:/Espressif/frameworks/esp-idf-v4.4.2/Kconfig;--sdkconfig-rename;C:/Espressif/frameworks/esp-idf-v4.4.2/sdkconfig.rename;--config;C:/Users/nahue/Documents/VS
  Code/ESP32/Blink LED/blink/sdkconfig;--defaults;C:/Users/nahue/Documents/VS
  Code/ESP32/Blink
  LED/blink/sdkconfig.defaults;--env-file;C:/Users/nahue/Documents/VS
  Code/ESP32/Blink LED/blink/build/config.env).  Error 1
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/build.cmake:525 (__kconfig_generate_config)
  C:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/nahue/Documents/VS Code/ESP32/Blink LED/blink/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

brianignacio5, 2022-10-31:
You have spaces in your project path (`C:\Users\nahue\Documents\VS Code\ESP32\Blink LED`) @nahuel-uranga-sierra 

Could you move your repository to a location in your pc without spaces? As discussed above, spaces are not supported yet.

nahuel-uranga-sierra, 2022-10-31:
> You have spaces in your project path (`C:\Users\nahue\Documents\VS Code\ESP32\Blink LED`) @nahuel-uranga-sierra
> 
> Could you move your repository to a location in your pc without spaces? As discussed above, spaces are not supported yet.

I'm really sorry, It works

ekow-d, 2023-11-16:
Have you or how did you solve this?

brianignacio5, 2023-11-17:
Delete the build subdirectory and try again ?

ekow-d, 2023-11-17:
CMake Error at CMakeLists.txt:7 (if):
  if given arguments:

    "STREQUAL" "linux"

  Unknown arguments specified
-- Configuring incomplete, errors occurred! ,

 **this is the error i get now after deleting the build subdirectory and trying again**

na3eddin, 2024-05-11:
@brianignacio5 hello i have the same problem i deleted the build sub and still don't work, can you help me?