espressif/vscode-esp-idf-extension #610: [Feature Request]: (more and user advice)! don't try to use the same environment for CLMI and VS

philippe44, 2022-01-16:


brianignacio5, 2022-01-19:
Why you can't use the same environment between Vscode and CLI ?

Actually, if your environment variables are set (IDF_PATH, IDF_TARGET, OPENOCD_SCRIPTS) and the required IDF Tools are in PATH (git, xtensa-esp32-elf, opened-esp32, etc.) you should be able to use the extension without even using the extension setup.

Could you provide more information about what is happening in your setup ? Like the build output, extension log, etc. as defined in the [troubleshooting section](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting)

philippe44, 2022-01-20:
I will try but I already did reinstall everything separately because I really had to have it working back quickly. Typically, my CLI build system (just installed using the install.bat) stopped working after I ran the .visx installer and told it I wanted to use an existing toolset (4.3.2) whose location I gave. The idf.py build/menuconfig would fail to just run because of an issue with (missing) windows-curses and the advice to use pip to install it was a no-go.

brianignacio5, 2022-01-24:
how did you try to use idf.py build/menuconfig ? Did you just open a terminal on vscode or did you use the `ESP-IDF: Open ESP-IDF Terminal` command ? The 2nd option is the one that is going to use the extension settings to override a terminal with environment variables IDF_PATH, etc.

philippe44, 2022-01-24:
I use a cmd box (as usual) that launches the idf_cmd_init.bat (truly the export.bat) to set ENV variables and work form there

brianignacio5, 2022-01-29:
So you are not using the extension settings in this cmd box. The extension does install other python packages to the python virtual environment so I'll add a comment in the docs.

closed 2022-01-29 by brianignacio5

philippe44, 2022-01-30:
Thanks for the explanations

PR #621 (2022-01-29): add tutorial links
Add tutorials links in README.md and mention that the setup wizards install more python packages than ESP-IDF

Fix #610