espressif/vscode-esp-idf-extension #611: [Bug Report]: SDK Configuration Editor filter stops working

boarchuz, 2022-01-18:
**Describe the bug**
In some circumstances, certain filter strings in the SDK Configuration Editor will cause the filter to stop working (ie. changing the filter further has no effect on the displayed configuration items; clearing the filter has no effect).

**To Reproduce**
https://github.com/boarchuz/idfcodeext_filterissue (see readme for instructions)

Note, especially, that there are multiple components in bootloader_components and these are **shared with the main project via the EXTRA_COMPONENT_DIRS directive in CMakeLists.txt**. This also causes the menus for these components to be duplicated, however this is consistent with menuconfig (it may be relevant to this issue though?).

**Environment**

- OS Name & Version: Ubuntu 21.10
- VSCode Version: 1.3.0
- ESP-IDF Version: master (v5.0-dev-1188-g3ba47f1628)
- Python Version: 3.9.7

**Output**

(none)

**Extension Configuration settings**

- `idf.espIdfPath`: /home/matt/espressif/esp-idf
- `idf.pythonBinPath`: /home/matt/espressif/python_env/idf5.0_py3.9_env/bin/python
- `idf.customExtraPaths`: /home/matt/espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin:/home/matt/espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin:/home/matt/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin:/home/matt/espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin:/home/matt/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/matt/espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/matt/espressif/tools/cmake/3.20.3/bin:/home/matt/espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/home/matt/espressif/tools/ninja/1.10.2
- `idf.customExtraVars`: {\"OPENOCD_SCRIPTS\":\"/home/matt/espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts\"}
- `idf.adapterTargetName`: esp32

brianignacio5, 2022-02-07:
Thanks for reporting this @boarchuz , it seems the issue is because menuconfig creates 2 entries for each component for some reason. This make the SDK Configuration Editor fails because each menu item should have a unique id for VueJS rendering purposes.

Will contact with the ESP-IDF team to discuss about it.

boarchuz, 2022-02-17:
Sorry @brianignacio5 , I only just saw your update.

It looks like that's true for the purposes of Kconfig (which means it's true for the minimal example I linked). Unfortunately in my actual project these are not Kconfig-only components so I do need to add them with EXTRA_COMPONENT_DIRS for headers, etc to be available in the main project too. Maybe I'm going about that the wrong way.

Regardless, it seems like this is not an issue with the extension, so I'll close this and create a new issue in the IDF repo or forum regarding shared components.

closed 2022-02-17 by boarchuz

brianignacio5, 2022-02-17:
Apparently `bootloader_components ` will duplicate these components. Maybe using another name for this project folder ?

boarchuz, 2022-02-17:
Yeah that seems much better!
I've created `common_components` alongside `bootloader_components` and `components`, moved shared components in there, tweaked CMakeLists, and everything seems happy now: it's building, there are no duplicated menus, and filtering works.
Thanks for the nudge.