espressif/vscode-esp-idf-extension #612: [Bug Report]: SDK Configuration Editor displays invalid choice options

boarchuz, 2022-01-18:
**Describe the bug**
All options for Kconfig `choice` elements are displayed in the SDK Configuration Editor, even if invalid due to unmet `depends on` conditions.

**To Reproduce**
Create a new project with ESP32 as the target, open SDK Configuration Editor and note the presence of invalid options, eg.

- Serial flasher config -> Flash SPI Mode -> OPI
- Serial flasher config -> Flash Sampling Mode -> DTR Mode
- ESP32-specific -> CPU Frequency -> 40 MHz

**Environment**

- OS Name & Version: Ubuntu 21.10
- VSCode Version: 1.3.0
- ESP-IDF Version: master (v5.0-dev-1188-g3ba47f1628)
- Python Version: 3.9.7

**Output**

SDK Configuration Editor, upon selecting ESP32-specific -> CPU Frequency -> 40 MHz:
```
{"version": 2, "set": { "ESP32_DEFAULT_CPU_FREQ_40": true }}

---------------------------ERROR--------------------------

Error: The following config symbol(s) were not visible so were not updated: ESP32_DEFAULT_CPU_FREQ_40

-----------------------END OF ERROR-----------------------
{"version": 2, "values": {}, "ranges": {}, "visible": {}, "error": ["The following config symbol(s) were not visible so were not updated: ESP32_DEFAULT_CPU_FREQ_40"]}
```

**Extension Configuration settings**

- `idf.espIdfPath`: /home/matt/espressif/esp-idf
- `idf.pythonBinPath`: /home/matt/espressif/python_env/idf5.0_py3.9_env/bin/python
- `idf.customExtraPaths`: /home/matt/espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin:/home/matt/espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin:/home/matt/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin:/home/matt/espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin:/home/matt/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/matt/espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/matt/espressif/tools/cmake/3.20.3/bin:/home/matt/espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/home/matt/espressif/tools/ninja/1.10.2
- `idf.customExtraVars`: {\"OPENOCD_SCRIPTS\":\"/home/matt/espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts\"}
- `idf.adapterTargetName`: esp32

brianignacio5, 2022-02-07:
Can you please try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/158732093.zip) ?

boarchuz, 2022-02-07:
Thanks @brianignacio5 
That fixes the immediate issue, but Kconfig changes elsewhere aren't being 'pushed' to the selection which could be confusing.

eg.
- Select 'Dynamic' in the dropdown for ESP32_WIFI_TX_BUFFER ("Type of WiFi TX buffers").
- Enable ESP32_SPIRAM_SUPPORT ("Support for external, SPI-connected RAM"). This invalidates the 'Dynamic' option above.
- Note that the GUI has not updated ESP32_WIFI_TX_BUFFER, where 'Dynamic' is still selected (out of sync with sdkconfig).

brianignacio5, 2022-02-08:
Thanks for checking this out @boarchuz ! I added some code that hopefully fix this in this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/159883231.zip) same PR

boarchuz, 2022-02-08:
That's perfect, thanks again @brianignacio5!

PR #633 (2022-02-09): hide not visible options in dropdown
Fix #612
closed 2022-02-09 by brianignacio5