espressif/vscode-esp-idf-extension #615: [Bug Report]: Cannot open monitor using WSL2 with project in windows folder

fronders, 2022-01-19:
**Describe the bug**
Cannot use monitor on WSL2 while the project is in windows folder, the powershell outputs the "No such file or directory" error. 
The build and flash executes successfully, but opening monitor fails.
These are the lines from terminal:
```
export WSLENV=IDF_PATH/p
powershell.exe -Command "python \\\\wsl$\\Ubuntu\\home\\fronders\\esp\\esp-idf\\tools\\idf_monitor.py -p COM6 /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board/build/techmaker-board.elf"

fronders > /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board > export WSLENV=IDF_PATH/p
fronders > /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board > powershell.exe -Command "python \\\\wsl$\\Ubuntu\\home\\fronders\\esp\\esp-idf\\tools\\idf_monitor.py -p COM6 /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board/build/techmaker-board.elf"
usage: idf_monitor - a serial output monitor for esp-idf [-h] [--port PORT] [--disable-address-decoding] [--baud BAUD] [--make MAKE] [--encrypted]
                                                         [--toolchain-prefix TOOLCHAIN_PREFIX] [--eol {CR,LF,CRLF}] [--print_filter PRINT_FILTER]
                                                         [--decode-coredumps {info,disable}] [--decode-panic {backtrace,disable}] [--target TARGET] [--ws WS]
                                                         elf_file
idf_monitor - a serial output monitor for esp-idf: error: argument elf_file: can't open '/mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board/build/techmaker-board.elf': [Err
no 2] No such file or directory: '/mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board/build/techmaker-board.elf'
```

However it successfully opens monitor for project in wsl home folder:
```
export WSLENV=IDF_PATH/p
powershell.exe -Command "python \\\\wsl$\\Ubuntu\\home\\fronders\\esp\\esp-idf\\tools\\idf_monitor.py -p COM6 /home/fronders/esp/template-app/build/template-app.elf"

fronders > ~/esp/template-app > export WSLENV=IDF_PATH/p
fronders > ~/esp/template-app > powershell.exe -Command "python \\\\wsl$\\Ubuntu\\home\\fronders\\esp\\esp-idf\\tools\\idf_monitor.py -p COM6 /home/fronders/esp/template-app/build/template-app.elf"
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM6 instead...
--- idf_monitor on \\.\COM6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7008
load:0x40078000,len:14308
```

**Environment (please complete the following information):**

- OS Name & Version: Windows 10 x64 21H1 build 19043.1466 using WSL2 Ubuntu 20 (OS linux x64 5.10.60.1-microsoft-standard-WSL2)
- VSCode Version: 1.63.2 
- ESP-IDF extension version 1.3.0
- ESP-IDF version 4.3.1
- Python version 3.8.10

**Output**
Build and Flash are successfull:
```
> Executing task: ninja  <

[1/8] Performing build step for 'bootloader'
ninja: no work to do.
[6/6] Generating binary image from built executable
esptool.py v3.1-dev
Merged 2 ELF sections
Generated /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board/build/techmaker-board.bin

> Executing task: /home/fronders/.espressif/python_env/idf4.3_py3.8_env/bin/python /home/fronders/esp/esp-idf/tools/idf_size.py /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board/build/techmaker-board.map <

Total sizes:
 DRAM .data size:   10768 bytes
 DRAM .bss  size:   37288 bytes
Used static DRAM:   48056 bytes ( 132680 available, 26.6% used)
Used static IRAM:   63330 bytes (  67742 available, 48.3% used)
      Flash code:  296903 bytes
    Flash rodata:   66336 bytes
Total image size:~ 437337 bytes (.bin may be padded larger)

> Executing task: powershell.exe -Command "python \\\\wsl$\\Ubuntu\\home\\fronders\\esp\\esp-idf\\components\\esptool_py\\esptool\\esptool.py -p COM6 -b 921600 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x20000 techmaker-board.bin 0x1000 bootloader\bootloader.bin 0x8000 partition_table\partition-table.bin 0xf000 ota_data_initial.bin" <

esptool.py v3.1-dev
Serial port COM6
Connecting.....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:71:bf:5a:27:dc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00020000 to 0x0008afff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000f000 to 0x00010fff...
Compressed 437728 bytes to 266652...
Wrote 437728 bytes (266652 compressed) at 0x00020000 in 4.1 seconds (effective 850.0 kbit/s)...
Hash of data verified.
Compressed 18768 bytes to 12533...
Wrote 18768 bytes (12533 compressed) at 0x00001000 in 0.4 seconds (effective 368.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 156...
Wrote 3072 bytes (156 compressed) at 0x00008000 in 0.1 seconds (effective 438.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0000f000 in 0.1 seconds (effective 682.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
**Extension Configuration settings**
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /home/fronders/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) /home/fronders/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin:/home/fronders/.espressif/tools/xtensa-esp32s2-elf/esp-2021r1-8.4.0/xtensa-esp32s2-elf/bin:/home/fronders/.espressif/tools/xtensa-esp32s3-elf/esp-2021r1-8.4.0/xtensa-esp32s3-elf/bin:/home/fronders/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin:/home/fronders/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/fronders/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/fronders/.espressif/tools/cmake/3.16.4/bin:/home/fronders/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin:/home/fronders/.espressif/tools/ninja/1.10.2
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/home/fronders/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /home/fronders/.espressif/python_env/idf4.3_py3.8_env/bin/python
Serial port (idf.port) COM6
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) /home/fronders/.espressif
Git Path (idf.gitPath) /usr/bin/git

brianignacio5, 2022-01-24:
Can you test Does the file `/mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board/build/techmaker-board.elf` reachable in WSL 2 ? Seems it can't find it.

fronders, 2022-01-24:
Yes it is reachable, see logs below (copied elf path from powershell call)

```
export WSLENV=IDF_PATH/p
powershell.exe -Command "python \\\\wsl$\\Ubuntu-20.04\\home\\fronders\\esp\\esp-idf\\tools\\idf_monitor.py -p COM3 /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board/build/techmaker-board.elf"

fronders > /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board > export WSLENV=IDF_PATH/p
fronders > /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board > powershell.exe -Command "python \\\\wsl$\\Ubuntu-20.04\\home\\fronders\\esp\\esp-idf\\tools\\idf_monitor.py -p COM3 /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board/build/techmaker-board.elf"
usage: idf_monitor - a serial output monitor for esp-idf
       [-h] [--port PORT] [--disable-address-decoding]
       [--baud BAUD] [--make MAKE] [--encrypted]
       [--toolchain-prefix TOOLCHAIN_PREFIX] [--eol {CR,LF,CRLF}]
       [--print_filter PRINT_FILTER]
       [--decode-coredumps {info,disable}]
       [--decode-panic {backtrace,disable}] [--target TARGET]    
       [--ws WS]
       elf_file
idf_monitor - a serial output monitor for esp-idf: error: argument 
elf_file: can't open '/mnt/c/Users/alexa/Repo/techmaker-board/techm
aker-board/build/techmaker-board.elf': [Errno 2] No such file or di
rectory: '/mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board/b
uild/techmaker-board.elf'

fronders > /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board > hexdump -n 32 /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board/build/techmaker-board.elf
0000000 457f 464c 0101 0001 0000 0000 0000 0000
0000010 0002 005e 0001 0000 15ec 4008 0034 0000
0000020
```

However I was able to run the monitor the following way

```
fronders > /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board > powershell.exe -Command "pwd"                                                                                             

Path
----
C:\Users\alexa\Repo\techmaker-board\techmaker-board

fronders > /mnt/c/Users/alexa/Repo/techmaker-board/techmaker-board > powershell.exe -Command "python \\\\wsl$\\Ubuntu-20.04\\home\\fronders\\esp\\esp-idf\\tools\\idf_monitor.py -p COM3 build/techmaker-board.elf"
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- idf_monitor on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1780
load:0x40078000,len:13992
xtensa-esp32-elf-addr2line -pfiaC -e build/techmaker-board.elf 0x40078000: [WinError 2] The system cannot find the file specified
load:0x40080400,len:3648
xtensa-esp32-elf-addr2line -pfiaC -e build/techmaker-board.elf 0x40080400: [WinError 2] The system cannot find the file specified
entry 0x400805f0
xtensa-esp32-elf-addr2line -pfiaC -e build/techmaker-board.elf 0x400805f0: [WinError 2] The system cannot find the file specified
I (169) psram: This chip is ESP32-D0WD
I (169) spiram: Found 32MBit SPI RAM device
I (169) spiram: SPI RAM mode: flash 80m sram 40m
```

brianignacio5, 2022-01-30:
@fronders Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/153265613.zip) ?

fronders, 2022-01-30:
@brianignacio5 yeah that works now, thanks! 

It also prompted which interface to use: JTAG, UART or DFU, but I can't find where it stored the value in project's .vscode/settings.json, did it change the global value in User settings? I think this value better be set per project and maybe added to bottom toolbar for quick selection (next to target, COM port etc)?

brianignacio5, 2022-02-07:
Thanks for your suggestion @fronders !

brianignacio5, 2022-02-09:
I added the Flash type in the status bar and make so flash type is saved in workspace folder in this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/159984843.zip)

closed 2022-02-10 by brianignacio5

fronders, 2022-02-10:
@brianignacio5 I tested the vsix you provided, and yes it stores the value in workspace folder. 
But now there's a problem - it prompts for flash method and folder to save that every time I click flash :/ Should flash method be a separate menu on toolbar (like selecting port) and the flash button be like previously - only executing action?

brianignacio5, 2022-02-17:
I added these fixes in #641 @fronders  Please try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/166865530.zip) from master and see if it is okay

PR #625 (2022-02-10): use relative elf in wsl monitor
Use relative path of elf file for WSL 2 monitor task.

Fix using DFU select device on UART flash type.

Fix #615