espressif/vscode-esp-idf-extension #618: [Bug Report]: cannot open source file "sys/reent.h"

eos1d3, 2022-01-23:
**Describe the bug**
Using IDF 4.4RC1 blink example, an error occurred with the message:
```
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (/Users/xxx/esp/blink/main/blink_example_main.c).C/C++(1696)
cannot open source file "sys/reent.h" (dependency of "freertos/FreeRTOS.h")C/C++(1696)
```

in the `main.c` file, only` #include "freertos/FreeRTOS.h"` is highlighted with error while the other include files are found and no error is indicated.

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "esp_log.h"
#include "led_strip.h"
#include "sdkconfig.h"
```

**To Reproduce**
Copy IDF 4.4RC1 Blink example, change c_cpp_properties.json to the following and build.

```
{
    "configurations": [
        {
            "name": "Mac",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${workspaceFolder}/**"
            ],
            "defines": [],
            "macFrameworkPath": [
                "/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks"
            ],
            "compilerPath": "/usr/bin/clang",
            "cStandard": "c17",
            "cppStandard": "c++98",
            "intelliSenseMode": "macos-clang-x64"
        }
    ],
    "version": 4
}
```

**Expected behavior**
No error or warning is shown. But actually the program works with LED flashing. Just VSCode cannot fine the include file.

**Environment (please complete the following information):**

- MacOS Big Sur, 11.6.2
- VSCode Version: 1.63.2
- ESP-IDF Version: 4.4RC1
- Python Version: 3.9.10

**Extension Configuration settings**
vscode/settings.json:

{
    "idf.adapterTargetName": "esp32s2",
    "idf.port": "/dev/cu.usbserial-14C0",
}

brianignacio5, 2022-01-24:
Maybe use the compile_commands.json file as specified in [C_CPP_PROPERTIES configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) document ?

eos1d3, 2022-01-25:
Thank a lot for helping the issue.

I tried what you told in `compile_commands.json`,  it works now. But I find missing Tag Parser in IntelliSenseEngine .
So I add `"C_Cpp.intelliSenseEngine": "Tag Parser"` in settings.json, it also works.

brianignacio5, 2022-01-25:
The `"C_Cpp.intelliSenseEngine": "Tag Parser"` is added by the extension in settings.json when you create a project from examples, using the project wizard or using the add vscode configuration files command.

Closing this issue, glad it works.

closed 2022-01-25 by brianignacio5

eos1d3, 2022-01-25:
The missing Tag Parser is caused by openning folder in a copy of IDF Blink example directly.

Can the extension check this for better compatibility? Manually adding it is not an issue but we need to know this limitation exists.