espressif/vscode-esp-idf-extension #623: New VS Code Extension 1.3.0 seemed to break jtag programming. (Regresseion test before releasing!!!) (OCD-466)

pedrocon, 2022-01-27:
### Development Kit

ESP32-DevKitC

### Module or chip used

esp32-wroom

### Debug Adapter

esp-prog

### OpenOCD version

v0.10.0-esp32-20210401

### Operating System

windows

### Using an IDE ?

vs code

### OpenOCD command line

esp32-ethernet-kit

### JTAG Clock Speed

20 MHz

### ESP-IDF version

4.3.1

### Problem Description

I didnt upgrade the espressif vs code extension. It automatically updated (to 1.3.0 ) and I wasn't able to program with JTAG. I'm pissed off. I reverted to 1.2.0 and it worked right away.

### Expected behavior

It would be nice if jtag programming worked after updating the extension to 1.3.0 from 1.2.0.

gerekon, 2022-01-27:
@brianignacio5 PTAL

brianignacio5, 2022-01-28:
Could you provide more information about your setup?

**NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Output**

Click the menu `View` and select `Output`. This extension uses several output channels: ESP-IDF, ESP-IDF Debug Adapter, Heap Trace, OpenOCD and SDK Configuration Editor.

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

who-has-seen-the-wind, 2022-01-29:
Similar problem here, i.e cannot flash with JTAG within VSCode
using F1 "ESP-IDF: Flash (with JTag)"

Importantly, I can flash with JTAG within VSCode if I run
     openocd -f board/esp32-wrover-kit-3.3v.cfg -c "program_esp build/blink.bin 0x10000 verify exit"
which demonstrates that the hardware is ok.

Also, if I hit F5 in VSCode to debug that will work.
So, overall, it's as if the ESP-IDF VSCode extension if not delivering the flash commands to OpenOCD.

Here's my setup details:
Brand new clean install, unmodifed,  of ESP-IDF V4.4, and of VScode V1.63.2 on Windows 10. 
Debug Adapter: ESP-PROG
Module: ESP32-DEVKITV1 (ESP-WROOM-32)
OpenOCD version: v0.11.0-esp32-20211220 

View Output Results are as follows:

*** OpenOCD
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
adapter speed: 20000 kHz
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection

*** ESP-IDF 
 nothing
 
*** ESP-IDF Debug Adapter
  nothing
  
*** GDB Heap Trace
 nothing
 
 
 *** settings.json
```
 {
  "C_Cpp.intelliSenseEngine": "Tag Parser",
  "idf.adapterTargetName": "esp32",
  "idf.openOcdConfigs": [
    "interface/ftdi/esp32_devkitj_v1.cfg",
    "target/esp32.cfg"
  ]
}
```

*** launch.json
```
{
  "version": "0.2.0",
  "configurations": [
    { 
      "type": "espidf",
      "name": "Launch",
      "request": "launch"
    }
  ]
}
```

who-has-seen-the-wind, 2022-02-01:
Today I tried reverting the ESP-IDF VSCode Extension from v1.3 to v1.2
and then "ESP-IDF Flash (with JTag)" worked. 
So that seems to confirm the original poster's bug report.

It would be helpful if other readers using v1.3  
would comment here  as to whether using F1 "ESP-IDF Flash (with JTag)"
succeeds or fails (hangs).

Thanks.

brianignacio5, 2022-02-07:
Could you provide your setup information (as described in the New Issue template) to see what is happening ? Specially the extension log could be useful to see if there is an error !

Related to JTAG flashing, I made a PR with some fixes. Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/153265613.zip)

who-has-seen-the-wind, 2022-02-07:
Thanks for your reply.

The setup information (as described in the New Issue template) is attached to my first post in this thread,
and today as requested here's the extension log file: 
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/8012606/esp_idf_vsc_ext.log)
which doesn't contain much. Here is the Output window
[2022_01_06_openocd_debugLevel3.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8012540/2022_01_06_openocd_debugLevel3.txt)

I installed the VISX file that you provided, then used F1 "ESP-IDF: Configure ESP-IDF Extension" 
then full clean, then build (successfully), then F1 "ESP-IDF: Flash (with JTag)", which does not flash the ESP32.
Here is its Output window
[2022_01_06_openocd_debugLevel3_withVISXinstall..txt](https://github.com/espressif/vscode-esp-idf-extension/files/8012578/2022_01_06_openocd_debugLevel3_withVISXinstall.txt)
 
I also reverted again to v1.2 to double check that it succeeds, and it does.
Here is its Output Window:
[2022_01_06_openocd_debugLevel3_v1.2.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8012645/2022_01_06_openocd_debugLevel3_v1.2.txt)

On your system, with ESP-IDF v4.4 and VSCode Extension v1.3.0, does F1 "ESP-IDF: Flash (with JTag)" succeed?

who-has-seen-the-wind, 2022-02-07:
Here is the "ESP-IDF: Doctor Command" report
[2022_02_07_doctor.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8017439/2022_02_07_doctor.txt)

who-has-seen-the-wind, 2022-02-09:
Here is another confirmation that  reverting the ESP-IDF VSCode Extension from v1.3 to v1.2 
solves these problems
[https://www.esp32.com/viewtopic.php?f=40&t=25995](url)

meltdown03, 2022-02-12:
I did a little debugging and it gets stuck at: https://github.com/espressif/vscode-esp-idf-extension/blob/7b763614871012e9184dd4a76b42d49aad23066b/src/flash/jtagCmd.ts#L50
or maybe it's here:
https://github.com/espressif/vscode-esp-idf-extension/blob/7b763614871012e9184dd4a76b42d49aad23066b/src/taskManager.ts#L89

These are the errors: 

``` javascript
rejected promise not handled within 1 second: TypeError: Cannot read property 'execution' of undefined

Exception has occurred: TypeError: Cannot read property 'execution' of undefined
t s.executeTask (/opt/visual-studio-code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:101:6382)
	at Object.executeTask (/opt/visual-studio-code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:107:50297)
	at Function.<anonymous> (/home/neal/git/vscode-esp-idf-extension/dist/extension.js:84950:88)
	at Generator.next (<anonymous>)
	at /home/neal/git/vscode-esp-idf-extension/dist/extension.js:84909:71
	at new Promise (<anonymous>)
	at /home/neal/git/vscode-esp-idf-extension/dist/extension.js:84905:12
	at /home/neal/git/vscode-esp-idf-extension/dist/extension.js:84949:53
	at new Promise (<anonymous>)
	at Function.<anonymous> (/home/neal/git/vscode-esp-idf-extension/dist/extension.js:84949:20)
	at Generator.next (<anonymous>)
	at /home/neal/git/vscode-esp-idf-extension/dist/extension.js:84909:71
	at new Promise (<anonymous>)
	at /home/neal/git/vscode-esp-idf-extension/dist/extension.js:84905:12
	at Function.runTasks (/home/neal/git/vscode-esp-idf-extension/dist/extension.js:84948:16)
	at /home/neal/git/vscode-esp-idf-extension/dist/extension.js:78142:76
	at Generator.next (<anonymous>)
	at fulfilled (/home/neal/git/vscode-esp-idf-extension/dist/extension.js:78108:58)
	at processTicksAndRejections (internal/process/task_queues.js:93:5)```

who-has-seen-the-wind, 2022-02-12:
Thanks for your reply. 

Yes, that is consistent with what I've seen, i.e. it just hangs.

So that's about four users reporting this error, and zero users reporting success. 

Again, to brianignacio5, does this problem occur on your system?
i.e. With ESP-IDF v4.4 and VSCode Extension v1.3.0, does F1 "ESP-IDF: Flash (with JTag)" fail?

To any other users,  with ESP-IDF v4.4 and VSCode Extension v1.3.0,  does F1 "ESP-IDF: Flash (with JTag)" succeed or fail?

FanhuaCloud, 2022-02-14:
[新建文本文档.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8061714/default.txt)
I had the same problem, when I upload using openocd, it just says "Building Project: Flashing project into device..." and waits forever

meltdown03, 2022-02-16:
If you comment these 4 lines and build the extension from source, JTAG should work:
https://github.com/espressif/vscode-esp-idf-extension/blob/2bf03cf3f39c2bb047f505b84d485f3a9404d2be/src/flash/jtagCmd.ts#L49-L50
https://github.com/espressif/vscode-esp-idf-extension/blob/2bf03cf3f39c2bb047f505b84d485f3a9404d2be/src/flash/jtagCmd.ts#L54-L55

FanhuaCloud, 2022-02-16:
[vscode-app-1644984619379.log](https://github.com/espressif/vscode-esp-idf-extension/files/8076671/vscode-app-1644984619379.log)
Looks like something went wrong,and openocd show "Info : dropped 'tcl' connection"

meltdown03, 2022-02-16:
> [vscode-app-1644984619379.log](https://github.com/espressif/vscode-esp-idf-extension/files/8076671/vscode-app-1644984619379.log) Looks like something went wrong,and openocd show "Info : dropped 'tcl' connection"

[esp-idf-extension.zip](https://github.com/espressif/vscode-esp-idf-extension/files/8076810/esp-idf-extension.zip)
try this

FanhuaCloud, 2022-02-16:
> [esp-idf-extension.zip](https://github.com/espressif/vscode-esp-idf-extension/files/8076810/esp-idf-extension.zip) try this

Thanks, this version works fine

brianignacio5, 2022-02-16:
This should be already fixed in GitHub master no ?

FanhuaCloud, 2022-02-16:
> This should be already fixed in GitHub master no ?

The version I downloaded from the automated build still doesn't work

meltdown03, 2022-02-16:
It does not work. That debug traceback I posted is from the master branch.

brianignacio5, 2022-02-16:
I found the issue, posting a fix now

who-has-seen-the-wind, 2022-02-16:
Thanks @meltdown03 !
The version that you posted yesterday is able to flash with JTAG.

meltdown03, 2022-02-17:
> You're right @meltdown03 hopefully this PR #641 will fix the issue without losing those custom tasks possibility

Yes, it works great, and it doesn't keep asking which method to Flash with which was very annoying!
Here is the a zip for the other ppl if they are interested:
[esp-idf-extension.zip](https://github.com/espressif/vscode-esp-idf-extension/files/8084617/esp-idf-extension.zip)

PR #641 (2022-02-17): use workspace folder as readParameter scope
Add workspace folder to idfConfiguration readParameter since workspace folder settings are not being read if not explicitly specified.

Fix #626 

Fix #623
closed 2022-02-17 by brianignacio5