espressif/vscode-esp-idf-extension #624: No README.md files showing for the examples

meltdown03, 2022-01-29:
IDF - Master branch
OS - Archlinux
Extension - 1.3.0

When I open the Project Examples, no Readme files are displayed for any examples. I get a notification that `No README.md available for this project`. I verified the files are there.

I did some debugging with VS Code Extension Host and found this Exception when I click a Example.
It occurs at Line 998 of `utils.ts` (`marked.SetOptions`)

``` typescript
Exception has occurred: TypeError: Cannot read property 'setOptions' of undefined
t Module.markdownToWebviewHtml (/home/neal/git/vscode-esp-idf-extension/dist/extension.js:84191:52)
	at ExamplesPlanel.<anonymous> (/home/neal/git/vscode-esp-idf-extension/dist/extension.js:73174:108)
	at Generator.next (<anonymous>)
	at fulfilled (/home/neal/git/vscode-esp-idf-extension/dist/extension.js:73098:58)
```

I see the the contents of the readme in `markdownToWebviewHtml.content` as seen here: 
[image]

brianignacio5, 2022-01-29:
Thanks for reporting this @meltdown03 will take a look

brianignacio5, 2022-02-08:
Could you please try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/159034891.zip)

meltdown03, 2022-02-08:
It works good now. All the readme files show nicely.

PR #634 (2022-02-09): update marked an debug adapter deps to latest
Fix #624 

Update dependencies to latest version

marked
@vscode/debugadapter
@vscode/debugadapter-testsupport
@vscode/debugprotocol
@vscode/extension-telemetry
closed 2022-02-09 by brianignacio5