espressif/vscode-esp-idf-extension #626: [Bug Report]: Workspace Folder settings precedence ignored (.vscode/settings.json)

darthcloud, 2022-01-30:
**Describe the bug**
Workspace setting (in .code-workspace files) takes precedence over user setting fine.

However Workspace Folder settings (.vscode/settings.json) do not, Value I see reported from Doctor is the one from .code-workspace file. Actually even if no setting is set in either User and workspace level the workspace folder setting is still ignored and default value used instead.

This issue was recently mention in #287 at the end. You mention you could not reproduce in OSX but I do reproduce easily on OSX, Linux and Windows

My use cases is that I have multiple ESP projects I work on with different IDF version, using the workspace.code-workspace work fine to setup *ONE* of those project. But then I got to open two instance of vscode. If .vscode/settings.json worked I could have both project within the same workspace as described in https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/MULTI_PROJECTS.md#example

**To Reproduce**
I made a very simplified project workspace you can load to easily reproduce the issue:
https://github.com/darthcloud/vscodetest

workspace.code-workspace:
{
	"folders": [
		{
			"path": "."
		}
	],
	"settings": {
		"idf.gitPath": "roger"
	}
}

.vscode\settings.json:
{
    "idf.gitPath": "bob"
}

Load project in vs code: File->Open Workspace from File...
press F1 and type ESP-IDF: Doctor

**Expected behavior**
ESP-IDF: Doctor command should display line:
Git Path (idf.gitPath) bob

It show the git setting from workspace.code-workspace instead.

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS Name & Version: Windows 10 Pro Build 19041.vb_release.191206-1406
- VSCode Version:
Version: 1.63.2 (user setup)
Commit: 899d46d82c4c95423fb7e10e68eba52050e30ba3
Date: 2021-12-15T09:40:02.816Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.19041

- ESP-IDF Version: 4.4
- Python Version: Python 3.10.2

**Logs**
[doctor_linux.txt](https://github.com/espressif/vscode-esp-idf-extension/files/7966654/doctor_linux.txt)
[doctor_osx.txt](https://github.com/espressif/vscode-esp-idf-extension/files/7966656/doctor_osx.txt)
[doctor_win10.txt](https://github.com/espressif/vscode-esp-idf-extension/files/7966658/doctor_win10.txt)

solhuebner, 2022-01-31:
I do have issues with the COM port selection as it only takes the global one even if I change the scope. I can even see the settings file entries but they are ignored. Did not happen with the last version. I am using Windows 10 with the fixed version of the latest release.

darthcloud, 2022-02-16:
not stale, still got the issue

brianignacio5, 2022-02-16:
Can you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/165231474.zip)

I made some changes with regards to configuration settings reading in #641

darthcloud, 2022-02-16:
@brianignacio5 Look like it work, Workspace Folder settings are used.

I can now build 2 projects using different idf version within the same workspace!

Thank you very much for fixing this!!

solhuebner, 2022-02-16:
The shared version fixes the COM port selection issue but when I click the serial port icon I get the following message:
"t is not a function" / I reinstalled the extension and deleted the extension folder and .espressif plus esp folder but get the same error message. But it seems to work anyways. I attached the doctor log:

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/8080048/esp_idf_vsc_ext.log)

darthcloud, 2022-02-17:
I also see the error "t is not a function" when clicking on serial port icon.

Not sure if this related to this issue or the nightly extension but I can't get jtag debug to work:
I get error Couldn't find a debug adapter descriptor for debug type 'espidf' (extension might have failed

In bottom right I see IDF extension error:
Invalid OpenOCD script path or access is denied for file user.

But the Doctor output show a valid path for OPENOCD_SCRIPTS and openocd server launch fine.

[doctor_win_jtag.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8084707/doctor_win_jtag.txt)

[image]

closed 2022-02-17 by brianignacio5

brianignacio5, 2022-02-17:
the t is not a function should be fixed with commit a0b9e9dc570f9c2111b51a77494ef83b8641115b in that branch. Please try the current master @solhuebner with this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/166865530.zip)

Seems there is issue in the extension activation (that's why `espidf` debug type is not found). Could you try the former vsix installer and share the Extension log @darthcloud ?

darthcloud, 2022-02-17:
@brianignacio5 I installed the new one and same problem with jtag here the extension folder log:
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/8088379/esp_idf_vsc_ext.log)
[doctor_win_jtag2.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8088390/doctor_win_jtag2.txt)

 "t is not a function" is fixed btw

brianignacio5, 2022-02-17:
Same issue with `espidf` adapter not found ? Can you use other extension commands ?

Based on the doctor output, debug adapter and extension python packages are not installed in your virtual env. Can you try using the `Install ESP-IDF Python Packages` extension command ?

Do you have read permission for OPENOCD_SCRIPTS directory ?

darthcloud, 2022-02-17:
@brianignacio5 everything work, clean, build, flash, start ocd, only missing part is starting debug with launch.

OpenOCD start fine and see the 2 esp32 core so pretty sure access to OPENOCD_SCRIPTS folder.

Will try "Install ESP-IDF Python Packages" later tonight

darthcloud, 2022-02-17:
@brianignacio5 I fixed the python requirement for both the extension and adapter:
[doctor_win_jtag3.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8092881/doctor_win_jtag3.txt)

But I still get the same error:
{"user":true,"message":"Invalid OpenOCD script path or access is denied for the user","stack":"Error: Invalid OpenOCD script path or access is denied for the user\n\tat f.<anonymous> (\\dist\\extension.js:8:507292)\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:8:506639\n\tat new Promise (<anonymous>)\n\tat h (\\dist\\extension.js:8:506384)\n\tat \\dist\\extension.js:8:506952\n\tat new Promise (<anonymous>)\n\tat f.start (\\dist\\extension.js:8:506933)\n\tat Object.<anonymous> (\\dist\\extension.js:8:648021)\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:8:632946\n\tat new Promise (<anonymous>)\n\tat Oe (\\dist\\extension.js:8:632691)\n\tat Object.createDebugAdapterDescriptor (\\dist\\extension.js:8:646968)\n\tat \\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:98:37904\n\tat \\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:71:29253\n\tat new Promise (<anonymous>)\n\tat v (\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:71:29223)\n\tat R.getAdapterDescriptor (\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:98:37897)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2022-02-17T23:28:40.122Z"}

Is it possible to have a build that display the actual path it's trying to access in the log? and also which user it;s using?

My user is the owner of the full path and got access.

Should I open a new bug for this?

darthcloud, 2022-02-18:
@brianignacio5 opening my project folder alone resolved the access problem somehow. Reopening my multi-project workspace after that, the launch error was also gone.

But then I get another error from the debug adapter:
[debug_adapter_log.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8093046/debug_adapter_log.txt)

{"message":"Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:1:7563\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2022-02-18T00:02:02.999Z"}

and a empty box error message:
[image]

brianignacio5, 2022-02-18:
To see more information increase the logLevel in both the debug adapter and openOCD.

For debug adapter set `logLevel: 4` in launch.json and for openOCD set `idf.openOcdDebugLevel: 4` in your settings.json.

Based on the debug adapter output is waiting for target state. Maybe increase the Scale factor for gdb timeout with `tmoScaleFactor` in launch.json ?

PR #641 (2022-02-17): use workspace folder as readParameter scope
Add workspace folder to idfConfiguration readParameter since workspace folder settings are not being read if not explicitly specified.

Fix #626 

Fix #623