espressif/vscode-esp-idf-extension #629: [Feature Request]: No such file or directory: 'C:\\Users\\he\\esp\\esp-idf\\requirements.txt'

FanhuaCloud, 2022-02-04:
This problem arises when I use the plugin to install the idf of the master branch,Then I checked the latest esp-idf and found that the file was renamed to requirements.core.txt, I think we should be compatible with this change in the next version

Traceback (most recent call last):
  File "C:\Users\he\esp\esp-idf\tools\check_python_dependencies.py", line 34, in <module>
    with open(req_path) as f:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\he\\esp\\esp-idf\\requirements.txt'
Command failed: "C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe" "C:\Users\he\esp\esp-idf\tools\check_python_dependencies.py" -r "C:\Users\he\esp\esp-idf\requirements.txt"
Traceback (most recent call last):
  File "C:\Users\he\esp\esp-idf\tools\check_python_dependencies.py", line 34, in <module>
    with open(req_path) as f:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\he\\esp\\esp-idf\\requirements.txt'

remyhx, 2022-02-06:
> 

I just copied the requirements.core.txt to requirements.txt and it worked.

FanhuaCloud, 2022-02-06:
> > 
> 
> I just copied the requirements.core.txt to requirements.txt and it worked.

Yes,so it's a feature,should be compatible

remyhx, 2022-02-06:
Totally agree! Sloppy...

Also migrating projects is difficult. Eating my shoes now.  Flashing by vscode seems buggy (wants to open openocd server), in cmd flashing is possible.

brianignacio5, 2022-02-07:
This should be fixed with #620 but yet to be released on marketplace.

Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/149617099.zip) ?

FanhuaCloud, 2022-02-07:
> 
Thanks, it is available after installation.

closed 2022-02-07 by FanhuaCloud