espressif/vscode-esp-idf-extension #635: Setting "idf.enableCCache" to true does not work

meltdown03, 2022-02-08:
I tried setting `"idf.enableCCache": true` in the project `settings.json` file and globally.  When building, I watch the CMake configuration info at the beginning, I do not see the line `-- ccache will be used for faster recompilation`. 

I have to add this line in the `CMakeLists.txt` file
`set(CCACHE_ENABLE "1")`
in order to get it to work.

brianignacio5, 2022-02-08:
Oh I see. We were using a compiler flag `-DCCACHE_ENABLE=1` in the build compilation step. 

I'll take a look thanks :)

meltdown03, 2022-02-09:
It may actually be working now. Last night it wasn't, but today it is. It seemed hit-or-miss before though, idk.

meltdown03, 2022-02-09:
Another question, is it supposed to ask the flash method every time now? UART, DFU, or JTAG? Even when I set it in the `settings.json` to UART, it still asks.

meltdown03, 2022-02-09:
Ok I figured out when it does not work. So after it has run the cmake command and built (ninja) at least once, if I change something in the `sdkconfig` and rebuild. Then it only runs the `ninja` build so it does not use the `-CCACHE_ENABLE=1` compiler line

meltdown03, 2022-02-09:
**Initial Build:**
[image]

**Re-Running CMake:**
[image]

**Not Re-Running CMake for some reason:**
[image]

meltdown03, 2022-02-09:
I think I found how to make it glitch.

1. set `idf.enableCCache` to true and build project
2. Close VS Code or close the project (Ctrl-K, F)
3. Reopen VS Code and/or the project
4. Open "SDK Configuration editor" (not necessary to make any changes)
5. Build again. It will Re-Run CMake config, but no ccache anymore

If you watch the `./build/CMakeCache.txt` file when you open the SDK Configuration editor (step 4), you will see the `CCACHE_ENABLE:UNINTIALIZED=1` change to `CCACHE_ENABLE:UNINITIALIZED=0`

meltdown03, 2022-02-09:
> Hopefully this vsix installer
> <https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/160791289.zip>
> should fix the issue @meltdown03 <https://github.com/meltdown03> !

Ok, it works now when I perform those steps. If I find any other situations
where this happens, I will let you know. At least I know if it does happen,
a full clean will force it to work.

closed 2022-02-10 by brianignacio5

meltdown03, 2022-02-13:
I another time when CCache gets disabled, When you set the target to a different board, It forces a reconfigure and does not have the CCache arg when it does this. (As long as you don't run the SDK Config editor before building, if you do run it or do a manual full clean, it will work.)

PR #637 (2022-02-10): add ccache enable logic to sdk config editor
Fix #635 by adding ccache enable logic to SDK Configuration Editor server process.