espressif/vscode-esp-idf-extension #638: [Bug Report]: F5 to debug, error with "Failed to init Debug Adapter"

acebearxp, 2022-02-09:
Press F5 to launch debug , a message box with the error "Failed to init Debug Adapter".
But if I manually execute "idf.py gdbgui" , everything works fine.

OS: Windows11 21H2(22000.469)
VSCode: 1.64.1
vscode-esp-idf-extension: v1.3.0
esp-idf: release-4.4 [8153bfe4]

There're many different development environments in my laptop, stm32, esp32, dotnet, java... So I'm not starting vscode as most other people doing. Instead, I start a command shell window, executing "**$HOME/.espressif/idf_cmd_init.bat esp-idf-esp-idf-3bca1dcff0a1a9a66f79892b92e8174c**" first, as pic below
[image]

After that change to a workspace, then execute cmd "**code .**" to start the vscode  for my project.

esp-tools chain is installed into `C:\Users\acebear\.espressif`
esp-idf is installed into `C:\Users\acebear\source\esp-idf`
before executing idf_cmd_init.bat, only 2 esp related env-variables defined
```
IDF_PATH=C:\Users\acebear\source\esp-idf
IDF_TOOLS_PATH=C:\Users\acebear\.espressif
```

building, flash and any other functions are working fine, except "F5 Debug", when I launch debug, there's always a message box with the error "Failed to init Debug Adapter".
Any suggestion can I have a try?

the Debug console only 2 lines:
```
Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
```
the output for "ESP-IDF Debug Adapter":
```
DEBUG_ADAPTER_STARTED
2022-02-09 22:59:48,751 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
DEBUG_ADAPTER_READY2CONNECT
2022-02-09 22:59:48,757 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
DEBUG_ADAPTER_STOPPED
2022-02-09 22:59:48,799 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Stopped] : ESP-IDF Debug Adapter
```

the output for "OpenOCD":
```
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Target halted, PC=0x420072B8, debug_reason=00000001
Info : esp32s3.cpu1: Target halted, PC=0x420189A6, debug_reason=00000000
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
```

if I manually execute `idf.py gdbgui`, the web gdb debugger looks good
[image]

launch.json
```
{
[... 60 lines cut ...]
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to ${env:PATH}: false
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version undefined
ESP-IDF version undefined
Python version undefined
Python's pip version undefined
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: "/usr/bin/git" --version
ϵͳ�Ҳ���ָ����·����

--------------------------------------------------------------------------------------------------------------------------------------------

acebearxp, 2022-02-10:
There are some errors in "esp_idf_vsc_exg.log" as shown below

But all these errors related to ESP32/ESP32-S2, since the only chip I have is ESP32-S3 , I did not install tools for the other 2 chips

```
{"message":"Command failed: xtensa-esp32s2-elf-gcc --version\n'xtensa-esp32s2-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���\r\n���������ļ���\r\n","stack":"Error: Command failed: xtensa-esp32s2-elf-gcc --version\n'xtensa-esp32s2-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���\r\n���������ļ���\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Socket.<anonymous> (internal/child_process.js:439:11)\n\tat Socket.emit (events.js:315:20)\n\tat Pipe.<anonymous> (net.js:673:12)","level":"error","timestamp":"2022-02-10T06:17:22.577Z"}
{"message":"Error checking xtensa-esp32s2-elf version","stack":"Error: Command failed: xtensa-esp32s2-elf-gcc --version\n'xtensa-esp32s2-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���\r\n���������ļ���\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Socket.<anonymous> (internal/child_process.js:439:11)\n\tat Socket.emit (events.js:315:20)\n\tat Pipe.<anonymous> (net.js:673:12)","level":"error","timestamp":"2022-02-10T06:17:22.579Z"}
{"message":"Command failed: xtensa-esp32-elf-gcc --version\n'xtensa-esp32-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���\r\n���������ļ���\r\n","stack":"Error: Command failed: xtensa-esp32-elf-gcc --version\n'xtensa-esp32-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���\r\n���������ļ���\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Socket.<anonymous> (internal/child_process.js:439:11)\n\tat Socket.emit (events.js:315:20)\n\tat Pipe.<anonymous> (net.js:673:12)","level":"error","timestamp":"2022-02-10T06:17:22.581Z"}
...
```

acebearxp, 2022-02-10:
I just tried the [Use Microsoft C/C++ extension to debug](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md#use-microsoft-cc-extension-to-debug)
This works great!

brianignacio5, 2022-02-16:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/164922661.zip) ?

We made some fixes to the extension in GitHub master but haven't yet to publish to marketplace

brianignacio5, 2022-02-17:
Thanks for your feedback @acebearxp closing this issue then. 

Please let me know if you have an additional issue, will be releasing this new version to marketplace soon!

closed 2022-02-17 by brianignacio5

Ashutosh-017, 2023-09-28:
> I just tried the [Use Microsoft C/C++ extension to debug](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md#use-microsoft-cc-extension-to-debug) This works great!
Please tell me how to change this configuration setting 
[image]
[image]