espressif/vscode-esp-idf-extension #640: [Feature Request]: Remove all files and folders when you remove extension

fantasyland101, 2022-02-11:
**Is your feature request related to a problem? Please describe.**
When I remove the extention some maps and folders persist such as;
C:\Users\myuser\\.espressif

 

**A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]**
Programs and extentions (in my opinion) should be fully removed when uninstalled, if not the user says otherwise, as to not clutter the hard drive. 

**Describe the solution you'd like**
Remove all files and folders on uninstall 

**Describe alternatives you've considered**
Add a button to save settings as a .json file (if .espressif is used to store settings idk what it is used for )

**Additional context**
Thank you very mutch.

brianignacio5, 2022-02-16:
The thing is that `C:\Users\myuser\.espressif` could be installed not only by this extension but with other tools like you directly using ESP-IDF from terminal, the ESP-IDF Windows Installer or our Eclipse plugin. The extension uninstall mechanism is handle completely by Visual Studio Code and we don't control the uninstall lifecycle, only when extension activates/deactivates.

brianignacio5, 2022-02-17:
No worries, we could provide a way to delete this folder from the extension but it wouldn't be part of the uninstall event but a command UI wizard to install and remove ESP-IDF and tools.

We will provide something like this later, stay tuned!

closed 2022-03-11 by github-actions[bot]

freedaun, 2023-03-21:
Well, apparently like many users, my install of the VSCODE ESP-IDF extension just broke immediately. It complained that I had not set up a workspace folder and then, because it thought *I* had made the mistake, it gave up on me completely. F1 just results in the dreaded "espIdf.setup.start" not found, forever. 

Yes I am taking this personally. And no, I do not want everything dumped in my profile folder, and follow the Linux dinosaur. This is Windows, we put dev folders in the root of a drive (stop spamming me) and we don't muck around a day just to get something installed. 

Why don't you make sure you have a workspace before doing the work, and running the installer? 

So then I had to uninstall, rinse and repeat. Except this thing does not really uninstall. It is merely deactivated and the GB installed just sits there, coming back instantly when installed again. So I can't solve the problem (that's why the extension has only a 2.5 stars) by uninstalling.  So how then? 

This is the classic try-once-then-f*cked-forever situation developers love to drop users in, and call it a day. They never use the product they make. 

There is also a "ESP-IDF: Pick a workspace" command. But the command results isn't known. 

Espressif should provide:
- a full uninstall and 
- the option to rerun the installer, besides
- fixing the problem of why I got here in the first place

brianignacio5, 2023-03-22:
@freedaun What is the issue you are facing? You provided no [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) for us to understand the issue you are having ?

 Have you search for existing issues to see if the issue is fixed ?
 
 As we stated earlier, any vscode extension control 2 lifecycles activate and deactivate. Uninstall is not something we can control from the extension itself. We could delete `esp-idf` and `.espressif` folders, but not the extension itself in `$HOME/.vscode/extensions`.

Gaai, 2023-04-30:
So maybe it's a good idea to tell users the steps needed to cleanly remove all possible traces of the extension. So they can do a fresh install? I arrived here looking for that. I suppose others will too. 
Both on windows and mac my idf extension installs were broken right from the start. And I'd like to try again, but keep running in to leftover brokenness.

Gaai, 2023-04-30:
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md here's a start.

Gaai, 2023-04-30:
So I:
- uninstalled the extension again, 
- removed the extension folder from the $HOME/.vscode/extensions/
- removed all esp and esp-idf folders from $HOME,
- removed the $HOME/.espressif folder,
- made sure nothing esp or idf related was left in my $PATH (double checked .bash_profile, .zshrc, .zprofile for possible remnants. 
- quit vscode,
- rebooted, 

Thought to check the YouTube video you guys made for intallation of the extension. Which seems pretty old and different from how it goes now. 
Opened the other install page doc for reference. 

- installed esp-idf extension again and boom, it's right there without a second pauze... 
- after view>command>configure esp idf, it tells me:
`'Welcome.
Make sure that [ESP-IDF Prerequisites for MacOS](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/macos-setup.html)
are installed before choosing the setup mode.`
However following this link leeds me I think to the manual installation process, with cloning git and everything... which I don't think is what should be happening. 
...
Moving on

Gaai, 2023-04-30:
`
brew install cmake ninja dfu-util
Running `brew update --auto-update`...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/cask).
==> New Casks
ddpm

You have 1 outdated formula installed.

Warning: Treating cmake as a formula. For the cask, use homebrew/cask/cmake
Warning: cmake 3.26.3 is already installed and up-to-date.
To reinstall 3.26.3, run:
  brew reinstall cmake
Warning: ninja 1.11.1 is already installed and up-to-date.
To reinstall 1.11.1, run:
  brew reinstall ninja
Warning: dfu-util 0.11 is already installed and up-to-date.
To reinstall 0.11, run:
  brew reinstall dfu-util

python3 --version
Python 3.11.3

....

Submodule path 'components/nghttp/nghttp2/third-party/mruby': checked out '7c91efc1ffda769a5f1a872c646c82b00698f1b8'
Submodule path 'components/nghttp/nghttp2/third-party/neverbleed': checked out 'b967ca054f48a36f82d8fcdd32e54ec5144f2751'
Submodule path 'components/protobuf-c/protobuf-c': checked out 'dac1a65feac4ad72f612aab99f487056fbcf5c1a'
Submodule path 'components/spiffs/spiffs': checked out 'f5e26c4e933189593a71c6b82cda381a7b21e41c'
Submodule path 'components/tinyusb/tinyusb': checked out '28f89e13473d40637574bcbfe4142633b39899fd'
Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out 'c3d3a69021cfec3236ca2c0b63be4048ec6643a4'

/bin/sh: xtensa-esp32-elf-gcc: command not found
Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: xtensa-esp32-elf-gcc: command not found
Error checking xtensa-esp32-elf version
Error: Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: xtensa-esp32-elf-gcc: command not found

/bin/sh: xtensa-esp32s2-elf-gcc: command not found
Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: xtensa-esp32s2-elf-gcc: command not found
Error checking xtensa-esp32s2-elf version
Error: Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: xtensa-esp32s2-elf-gcc: command not found

/bin/sh: openocd: command not found
Command failed: openocd --version
/bin/sh: openocd: command not found
Error checking openocd-esp32 version
Error: Command failed: openocd --version
/bin/sh: openocd: command not found

/bin/sh: esp32s2ulp-elf-as: command not found
Command failed: esp32s2ulp-elf-as --version
/bin/sh: esp32s2ulp-elf-as: command not found
Error checking esp32s2ulp-elf version
Error: Command failed: esp32s2ulp-elf-as --version
/bin/sh: esp32s2ulp-elf-as: command not found

/bin/sh: esp32ulp-elf-as: command not found
Command failed: esp32ulp-elf-as --version
/bin/sh: esp32ulp-elf-as: command not found
Error checking esp32ulp-elf version
Error: Command failed: esp32ulp-elf-as --version
/bin/sh: esp32ulp-elf-as: command not found

`

oioiooi

Gaai, 2023-05-01:
So I found some other remnants:
check:
${workspaceFolder}/.vscode/settings.json
$HOME/Library/Application Support/Code/User/settings.json

So again:
-uninstalled the extension again,
-removed the extension folder from the $HOME/.vscode/extensions/
-removed all esp and esp-idf folders from $HOME,
-removed the $HOME/.espressif folder,
-removed settings from: $HOME/Library/Application Support/Code/User/settings.json
-made sure nothing esp or idf related was left in my $PATH (double checked .bash_profile, .zshrc, .zprofile for possible remnants).
-quit vscode,

Then for installing, to be sure nothing was wrong with the prerequisites I did the following:
-install prerequisites:
->> download latest cmake dmg https://cmake.org/download/
-install command line developer tools (git), ninja and dfu-util:
 xcode-select --install
 brew reinstall ninja dfu-util
-install rosetta to be sure
 /usr/sbin/softwareupdate --install-rosetta --agree-to-license
-check python version
 python3 --version
-check git version
 git --version
-upgrade if desired:
->>download from https://git-scm.com/download/mac

For extension:
-install extension
-view>command>configure esp idf
-advanced
-find esp on system (then it gave me the option to CLEAN previous install settings)
the for real with options:
-download from git (download server)
-version 5.0 (idf version)
-Users/me/esp (Esp idf container directory)
-Users/me/.espressif (Tools directory)
-user/bin/python3 (Python version)

Succes!! No I'm praying the rest of setup and examples will run as supposed too...

Gaai, 2023-05-01:
Still no luck loading the example sketches. Just hangs for a good while and then disappears.