espressif/vscode-esp-idf-extension #642: [Bug Report]: Cannot install Espressif IDF extension  v1.3.0 on Visual Studio Code. Keeps repeating

ua100k, 2022-02-11:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Following the instructions - Install Extension. Installed.
ESP-IDF: Configure ESP-IED Extension. This gives the screen "restart Visual Studio Code and run this wizard again." Regardless of restarting or reinstalling. Cannot get past this.

**To Reproduce**
1. Install Visual Studio Code. 
2. Install Espressif IDE extension in Visual Studio Code.
3. In Visual Studio Code, View->Command Palette->ESP-IDF: Configure ESP-IDF extension  

**Expected behavior**
According to  your install, this will provide a install the espressif programming environment .

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: Windows 10. 
- VSCode Version: [e.g. 1.35.1]  Haven't here - cannot get past the initial errors.
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration in your clipboard which you can copy here.

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.16299 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\MentorGraphics\9.5PADS\SDD_HOME\common\win32\bin;C:\MentorGraphics\9.5PADS\SDD_HOME\common\win32\lib;C:\MentorGraphics\9.5PADS\MGC_HOME.ixn\bin;C:\MentorGraphics\9.5PADS\MGC_HOME.ixn\lib;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Bitvise SSH Client;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Cadence\SPB_16.6\tools\pcb\bin;C:\Cadence\SPB_16.6\tools\bin;C:\Cadence\SPB_16.6\tools\libutil\bin;C:\Cadence\SPB_16.6\tools\fet\bin;C:\Cadence\SPB_16.6\tools\specctra\bin;C:\Cadence\SPB_16.6\tools\pspice;C:\Cadence\SPB_16.6\tools\capture;C:\Cadence\SPB_16.6\openaccess\bin\win32\opt;C:\Program Files\MATLAB\R2019a\bin;C:\Program Files\Zerynth\ztc\windows64;C:\Users\MerrillAudio12082016\AppData\Local\Programs\Python\Python37\Scripts\;C:\Users\MerrillAudio12082016\AppData\Local\Programs\Python\Python37\;C:\Users\MerrillAudio12082016\AppData\Roaming\Python\Python37\Scripts;C:\Users\MerrillAudio12082016\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.64.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.3.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\MerrillAudio12082016\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) c:\esp32_dev\.espressif\python_env\idf4.0_py3.7_env\Scripts;C:\Users\MerrillAudio12082016\AppData\Local\Programs\Python\Python39;c:\esp32_dev\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin;c:\esp32_dev\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin;c:\esp32_dev\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;c:\esp32_dev\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;c:\esp32_dev\.espressif\tools\cmake\3.13.4\bin;c:\esp32_dev\.espressif\tools\openocd-esp32\v0.10.0-esp32-20191114\openocd-esp32\bin;c:\esp32_dev\.espressif\tools\ninja\1.9.0;c:\esp32_dev\.espressif\tools\idf-exe\1.0.1;c:\esp32_dev\.espressif\tools\ccache\3.7
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"c:\\esp32_dev\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20191114/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) c:\esp32_dev\.espressif\python_env\idf4.0_py3.7_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\esp32_dev\.espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to c:\esp32_dev\.espressif\python_env\idf4.0_py3.7_env\Scripts: false
Access to C:\Users\MerrillAudio12082016\AppData\Local\Programs\Python\Python39: false
Access to c:\esp32_dev\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin: false
Access to c:\esp32_dev\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin: false
Access to c:\esp32_dev\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: false
Access to c:\esp32_dev\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: false
Access to c:\esp32_dev\.espressif\tools\cmake\3.13.4\bin: false
Access to c:\esp32_dev\.espressif\tools\openocd-esp32\v0.10.0-esp32-20191114\openocd-esp32\bin: false
Access to c:\esp32_dev\.espressif\tools\ninja\1.9.0: false
Access to c:\esp32_dev\.espressif\tools\idf-exe\1.0.1: false
Access to c:\esp32_dev\.espressif\tools\ccache\3.7: false
[... 24 lines cut ...]
> **NOTE:** Use `idf.openOcdDebugLevel` configuration setting to 3 or more to show debug logging in OpenOCD server output.

> **NOTE:** Use `logLevel` in your <project-directory>/.vscode/launch.json to 3 or more to show more debug adapter output.

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process output.

**Logs**
Please share the log file which can be obtained from:

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1 or View Menu -> Command Palette) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.

solhuebner, 2022-02-11:
Please try https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/137655771.zip and have a look at #602

griffin175, 2022-02-11:
I had the same issue and was able to make it work by installing v1.2.0 and updating to v1.3.0 after setting up.

acebearxp, 2022-02-12:
I've struggled with this for a couple of days and found a way finally.

A. Install `esp32 tools chain` and `esp-idf`. [Standard Setup of Toolchain](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/windows-setup.html)
```
You can define 2 environments IDF_TOOLS_PATH and IDF_PATH to point to your folders
IDF_TOOLS_PATH is for the tools chain folder, usually is C:/Users/USERNAME /.espressif
IDF_PATH is for the esp-idf framework, eg. C:/Users/USERNAME/source/esp-idf
```

B. open the 'settings.json' of VSCode [F1 -> Preferences : Open Settings(JSON) ], add following 5 entries:
```JSON
{
    "idf.pythonBinPathWin": "${env:IDF_TOOLS_PATH}/python_env/idf4.4_py3.8_env/Scripts/python.exe",
    "idf.gitPath": "${env:IDF_TOOLS_PATH}/tools/idf-git/2.30.1/cmd/git.exe",
    "idf.toolsPathWin": "${env:IDF_TOOLS_PATH}",
    "idf.espIdfPathWin": "${env:IDF_PATH}",
    "idf.customExtraPaths": "${env:IDF_TOOLS_PATH}/tools/xtensa-esp32s3-elf/esp-2021r2-patch2-8.4.0/xtensa-esp32s3-elf/bin;${env:IDF_TOOLS_PATH}/tools/riscv32-esp-elf/esp-2021r2-patch2-8.4.0/riscv32-esp-elf/bin;${env:IDF_TOOLS_PATH}/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin;${env:IDF_TOOLS_PATH}/tools/cmake/3.20.3/bin;${env:IDF_TOOLS_PATH}/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin;${env:IDF_TOOLS_PATH}/tools/ninja/1.10.2/;${env:IDF_TOOLS_PATH}/tools/idf-exe/1.0.3/;${env:IDF_TOOLS_PATH}/tools/ccache/4.3/ccache-4.3-windows-64;${env:IDF_TOOLS_PATH}/tools/dfu-util/0.9/dfu-util-0.9-win64;${env:IDF_PATH}/tools",
}
```
the `idf.customExtraPaths` depends on which tools chain of chip(esp32 esp32s2 esp32s3 esp32c3), you can retrieve it by open a esp-idf  shell command
[image]

C. install vscode esp extension, and config [F1->ESP-IDF: Configure ESP-IDF extension], choose the "use existing setup "
[image]

PS. if "use existing setup" is not shown, checking `C:\Users\USERNAME\.vscode\extensions\espressif.esp-idf-extension-1.3.0\esp_idf_vsc_ext.log`, there will give your some hits to find what's wrong

sbg97, 2022-02-12:
@acebearxp That worked for me. Thanks!

brianignacio5, 2022-02-14:
This is a duplicate of #602

closed 2022-02-14 by brianignacio5