espressif/vscode-esp-idf-extension #643: [Feature Request]: Refresh for EFuse Explorer

eos1d3, 2022-02-11:
**Is your feature request related to a problem? Please describe.**
The EFuse Explorer works. But whenever the development board is changed, or UART port is changed, I can't get data from new device. The only way is to restart VS Code.

**Describe the solution you'd like**
Add a refresh button to it, like what RainMaker tab does there.

brianignacio5, 2022-02-16:
Thanks for reporting this @eos1d3 will take a look

brianignacio5, 2022-02-16:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/166107379.zip)

closed 2022-02-21 by brianignacio5

eos1d3, 2022-02-21:
@brianignacio5 Sorry for my late reply. The vsix installer works very well. Great thank for the improvement!

PR #650 (2022-02-21): add refresh and clear button in efuse explorer
Fix #643 

Add refresh and clear eFuse summary results.