espressif/vscode-esp-idf-extension #646: [Bug Report]: Only wants to flash to COM3

Ostpopcorn, 2022-02-15:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
No matter what COM port I select it always selects COM3 when trying to flash the esp32 device, even if there is not COM3 in device manager.

**To Reproduce**

Well, it was working, both serial and JTAG. Don't know what happend.
Tried to reinsall the extension but the problem persists.

**Expected behavior**
Flash on COM9 and not COM3

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: Win10
- VSCode Version: v1.3.0
- ESP-IDF Version: v4.4
- Python Version: v3.8.7

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration in your clipboard which you can copy here.

**Logs**
Please share the log file which can be obtained from:

```
{"message":"Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\adied_000\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:1:7304\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2022-02-15T10:08:10.855Z"}
{"user":true,"message":"Failed to flash because of some unusual error","stack":"Error: Task ESP-IDF Flash exited with code 1\n\tat Function.<anonymous> (c:\\Users\\adied_000\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:1:32100)\n\tat Generator.next (<anonymous>)\n\tat c:\\Users\\adied_000\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:1:31158\n\tat new Promise (<anonymous>)\n\tat o (c:\\Users\\adied_000\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:1:30903)\n\tat c:\\Users\\adied_000\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:1:31918\n\tat l.fire (c:\\Users\\adied_000\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:64:1712)\n\tat s.$onDidEndTaskProcess (c:\\Users\\adied_000\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:97:54304)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2022-02-15T10:08:47.043Z"}
```

**Extension Configuration settings**
```

{
  "C_Cpp.intelliSenseEngine": "Tag Parser",
  "idf.portWin": "COM5",
  "idf.adapterTargetName": "esp32",
  "idf.openOcdConfigs": [
    "interface/ftdi/esp32_devkitj_v1.cfg",
    "target/esp32.cfg"
  ]
}
```

brianignacio5, 2022-02-16:
There are 3 places settings are saved based on `idf.saveScope`:

* User settings with `Preferences: Open settings (JSON)` command, 
* Workspace settings with `Preferences: Open workspace settings (JSON)` command
* Workspace folder setting in your `<project-dir>/.vscode/settings.json`

Can you see if settings are being overwritten from workspace folder -> workspace -> user settings ?

Ostpopcorn, 2022-02-16:
User settings with Preferences: Open settings (JSON) command,
```
...
"idf.saveScope": 1,
...
"idf.portWin": "COM1",
...
```

Workspace settings with Preferences: Open workspace settings (JSON) command
The whole file:
```
{
	"folders": [
		{
			"path": "."
		}
	],
	"settings": {
		"files.associations": {
			"chrono": "cpp",
			"gpio_cxx.hpp": "c",
			"complex": "cpp"
		}
	}
}
```
Workspace folder setting in your <project-dir>/.vscode/settings.json
The whole file:
```
{
  "C_Cpp.intelliSenseEngine": "Tag Parser",
  "idf.portWin": "COM5",
  "idf.adapterTargetName": "esp32",
  "idf.openOcdConfigs": [
    "interface/ftdi/esp32_devkitj_v1.cfg",
    "target/esp32.cfg"
  ]
}
```

Looks like it is not overwritten from here.

brianignacio5, 2022-02-17:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/166865530.zip) ?

Ostpopcorn, 2022-02-18:
Well, that worked! Thanks!
Now there is also a button to pick flash method which I dont think was ther before.

closed 2022-02-18 by Ostpopcorn