espressif/vscode-esp-idf-extension #647: [Bug Report]: Unable to run the extention

ademyankov, 2022-02-15:
The extension is stuck at the welcome screen.

[image]

I tried re-installing the extension many times, also tried deleting the folder at _.vscode\extensions\espressif.esp-idf-extension-1.3.0_, nothing helps.

There is no _esp_idf_vsc_ext.log_ file, I guess since the configuration phase was not finished.

VS Code:

[image]

Python:

[image]

Have anyone encountered a problem like this?

thegoodhen, 2022-02-15:
I had the same issue, downgrading to v. 1.2. seems to be a workaround.

brianignacio5, 2022-02-16:
Or fixed in #602  in GitHub master This is a duplicate.

closed 2022-02-16 by brianignacio5