espressif/vscode-esp-idf-extension #648: Suddenly unable to run the extension...

thegoodhen, 2022-02-15:
Hello,
my idf-py extension suddenly stopped working.

I know for a fact that my system PATH variable got corrupted recently and I had to restore it from a backup, but on a fresh Windows install on another computer, I don't really see many changes to it from the default state and to me it seems like most of the changes to PATH are done temporarily as we load the extension, right?

I have tried:

- Uninstalling it and reinstalling it again
- Uninstalling vscode and reinstalling it again
- Deleting all esp-idf related files from my system before reinstalling
- Disabling all other extensions except for C/C++, which is a dependency.
- Switching to different workspaces
- Downgrading to version 1.2.0.

Nothing has worked for me.

Problem description: When I turn on vscode, I get errors in the OUTPUT tab:

```
xtensa-esp32s2-elf-gcc nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.
Command failed: xtensa-esp32s2-elf-gcc --version
xtensa-esp32s2-elf-gcc nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.
Error checking xtensa-esp32s2-elf version
Error: Command failed: xtensa-esp32s2-elf-gcc --version
xtensa-esp32s2-elf-gcc nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.

esp32s2ulp-elf-as nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.
Command failed: esp32s2ulp-elf-as --version
esp32s2ulp-elf-as nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.
Error checking esp32s2ulp-elf version
Error: Command failed: esp32s2ulp-elf-as --version
esp32s2ulp-elf-as nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.

riscv32-esp-elf-gcc nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.
Command failed: riscv32-esp-elf-gcc --version
riscv32-esp-elf-gcc nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.
Error checking riscv32-esp-elf version
Error: Command failed: riscv32-esp-elf-gcc --version
riscv32-esp-elf-gcc nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.

openocd nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.
Command failed: openocd --version
openocd nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.
Error checking openocd-esp32 version
Error: Command failed: openocd --version
openocd nen� n�zvem vnit�n�ho ani vn�j�ho p��kazu,
spustiteln�ho programu nebo d�vkov�ho souboru.
(...)
```

These translate to "xxx is not recognized as an internal command operable program or batch file".

Full error log is appended as a file.

The output of the ESP IDF DOCTOR is:

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 6.1.7601 
[... 137 lines cut ...]
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------

```

Opening ESP-IDF terminal and running $env:path

shows that the path variable has not been correctly modified.

Thank you for your response!

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/8072720/esp_idf_vsc_ext.log)

thegoodhen, 2022-02-15:
Tried creating a new user account with a fresh install of vscode and the plugin. Did not work.

brianignacio5, 2022-02-16:
Settings seems to be ok but probably something is happening with your powershell execution, for some reason variables can't be found. What is the build output for example ?

thegoodhen, 2022-02-16:
> Settings seems to be ok but probably something is happening with your powershell execution, for some reason variables can't be found. What is the build output for example ?

Build output? I cannot even start the build. If I type 

`idf.py build`

I get

`The term 'idf.py' is not recognized as the name of a cmdlet, function, script file, or operable program.`

If I click the build button, it says "Something went wrong when trying to build the project." There is seemingly no error output.

I tried manually adding the necessary stuff to system PATH as a workaround:

C:\Users\thegoodhen\esp3\esp-idf\tools;C:\Users\thegoodhen\.espressif\tools;C:\Users\thegoodhen\.espressif\tools\idf-python;

 even though I know that on a fresh installation, the system PATH does not look like that. It's still failing, this time because it's running idf.py in a wrong python version... I can probably try to fix that by globally setting the python version to the one that comes with esp idf, but that's an ugly hack that I am not even sure if will work...

With the adjusted path, I was once able to get some terminal output when clicking the "build" button. It complained
about CMAKE not being able to find compilers, but I have been unable to reproduce it since.

I am stuck!

brianignacio5, 2022-02-17:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/166865530.zip) ? 

When you try the build command, there should be related terminal opened in your visual studio code session.

thegoodhen, 2022-02-17:
> Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/166865530.zip) ?
> 
> When you try the build command, there should be related terminal opened in your visual studio code session.

Hello! Tried it, does not install unfortunately. :/

```
17.2.2022 13:38:13 - Microsoft VSIX Installer
17.2.2022 13:38:13 - -------------------------------------------
17.2.2022 13:38:13 - vsixinstaller.exe version:
17.2.2022 13:38:13 - 15.9.3043
17.2.2022 13:38:13 - -------------------------------------------
17.2.2022 13:38:13 - Command line parameters:
17.2.2022 13:38:13 - C:\Program Files (x86)\Microsoft Visual Studio\Installer\resources\app\ServiceHub\Services\Microsoft.VisualStudio.Setup.Service\VSIXInstaller.exe,K:\esp-idf-extension.vsix
17.2.2022 13:38:13 - -------------------------------------------
17.2.2022 13:38:13 - Microsoft VSIX Installer
17.2.2022 13:38:13 - -------------------------------------------
17.2.2022 13:38:14 - Initializing Install...
17.2.2022 13:38:14 - Podrobnosti rozšíření...
17.2.2022 13:38:14 - 	Identifier         : esp-idf-extension
17.2.2022 13:38:14 - 	Name               : Espressif IDF
17.2.2022 13:38:14 - 	Author             : espressif
17.2.2022 13:38:14 - 	Version            : 1.3.0
17.2.2022 13:38:14 - 	Description        : Develop and debug applications for Espressif ESP32, ESP32-S2 chips with ESP-IDF
17.2.2022 13:38:14 - 	Locale             : en-US
17.2.2022 13:38:14 - 	MoreInfoURL        : 
17.2.2022 13:38:14 - 	InstalledByMSI     : False
17.2.2022 13:38:14 - 	SupportedFrameworkVersionRange : [0.0,2147483647.2147483647]
17.2.2022 13:38:14 - 
17.2.2022 13:38:14 - 	SignatureState     : Unsigned
17.2.2022 13:38:14 - 	Podporované produkty : 
17.2.2022 13:38:14 - 		Microsoft.VisualStudio.Code
17.2.2022 13:38:14 - 			Version : 
17.2.2022 13:38:14 - 
17.2.2022 13:38:14 - 	Odkazy             : 
17.2.2022 13:38:14 - Detaily podpisu...
17.2.2022 13:38:14 - 	Extension is not signed.
17.2.2022 13:38:14 - 
17.2.2022 13:38:14 - Vyhledávání podporovaných produktů...
17.2.2022 13:38:14 - Nalezen nainstalovaný produkt - Microsoft Visual Basic Express 2010
17.2.2022 13:38:14 - Nalezen nainstalovaný produkt - Microsoft Visual Web Developer Express 2010
17.2.2022 13:38:14 - Nalezen nainstalovaný produkt - Microsoft Visual Studio Express 2013 for Windows Desktop
17.2.2022 13:38:14 - Nalezen nainstalovaný produkt - Globální umístění
17.2.2022 13:38:14 - Nalezen nainstalovaný produkt - Visual Studio Community 2017
17.2.2022 13:38:14 - VSIXInstaller.NoApplicableSKUsException: Toto rozšíření nelze instalovat do žádného momentálně nainstalovaného produktu.
   v VSIXInstaller.ExtensionService.GetInstallableData(String vsixPath, String extensionPackParentName, Boolean isRepairSupported, IStateData stateData, IEnumerable`1& skuData)
   v VSIXInstaller.ExtensionPackService.IsExtensionPack(IStateData stateData, Boolean isRepairSupported)
   v VSIXInstaller.ExtensionPackService.ExpandExtensionPackToInstall(IStateData stateData, Boolean isRepairSupported)
   v VSIXInstaller.App.Initialize(Boolean isRepairSupported)
   v VSIXInstaller.App.Initialize()
   v System.Threading.Tasks.Task`1.InnerInvoke()
   v System.Threading.Tasks.Task.Execute()
--- Konec trasování zásobníku z předchozího místa, ze kterého byla vyvolána výjimka ---
   v Microsoft.VisualStudio.Telemetry.WindowsErrorReporting.WatsonReport.GetClrWatsonExceptionInfo(Exception exceptionObject)

```

The build button does not open any terminal window that I could see. I went throught the individual terminals and I see no build-related output...

brianignacio5, 2022-02-17:
This vsix installer is for Visual Studio Code. Are you trying to use Visual Studio ?

thegoodhen, 2022-02-17:
Aha, sorry. I just double clicked it and it said this. I uninstalled the extension and reinstalled it from the file you provided, but the behavior is the same...

Also, after I restored my user and system $PATH to default, the behavior changed. Now when I click the build button, it still says "Something went wrong while trying to build the project".

$env:path inside ESP-IDF terminal now yields:

PS L:\work\Prozeta\mbedTLS_test\https_mbedtls> $env:path
C:\Users\thegoodhen\esp\esp-idf\components\esptool_py\esptool;C:\Users\thegoodhen\esp\esp-idf\components\espcoredump;C:\Users\
thegoodhen\esp\esp-idf\components\partition_table;.;C:\Users\thegoodhen\esp\esp-idf\tools;C:\Users\thegoodhen\.espressif\tools
\idf-git\2.30.1\cmd;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\User
s\thegoodhen\AppData\Local\Programs\Microsoft VS Code\bin
PS L:\work\Prozeta\mbedTLS_test\https_mbedtls>

Trying to run idf.py says
Program 'idf.py' failed to execute: Pro tuto operaci není ke zvolenému souboru přidružena žádná aplikace

(Which translates to "no operation is assigned to the selected file). 

Idk, man. Frankly, I think that It's time for me to buy a new computer. I haven't reinstalled this Win7 machine for close to  14 years and it's a miracle that anything works, really.

closed 2022-03-11 by github-actions[bot]