espressif/vscode-esp-idf-extension #657: [Bug Report]: Debugging with ESP-PROG  not working.

lgLindstrom, 2022-02-28:
I cant get the ESP-PROG JTAG debugging to work. 
Either there is a bug, or I am missing some steps. There is a lot of instructions out on internet and a short period I suceded to debugging working but the instruction messed up so many other functions in the IDE I decided to go back to a basic extension setup. 

This is my steps to reproduce the problem.

1. Create new project by using ESP-IDF -> new project.
[image]

2. Select and create blink project
[image]

3. Result can be seen below.
[image]

4. VSCode settings file
[image]

5.  Compile the project
[image]

6. Upload and run the project
[image]

8.  Starting debugging by pressing Launch button
[image]

9.  A number of errors is displayed. Shown in pictures below.
[image]
[image]
[image]

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22000 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Python39\Scripts\;C:\Python39\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Program Files\dotnet\;C:\ProgramData\flutter\bin\;C:\ProgramData\flutter\bin\dart;C:\Program Files (x86)\Bitvise SSH Client;C:\Users\lglin\esp\esp-idf;C:\Users\lglin\.espressif;C:\Users\lglin\esp\esp-idf\tools;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Pico Technology\PicoScope6\;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files\Git\cmd;C:\Users\lglin\AppData\Local\Microsoft\WindowsApps;C:\Users\lglin\AppData\Roaming\npm;C:\Users\lglin\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\lglin\AppData\Local\Programs\Microsoft VS Code Insiders\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.64.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.3.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\lglin\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\lglin\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin;C:\Users\lglin\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\lglin\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\lglin\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin;C:\Users\lglin\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\lglin\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\lglin\.espressif\tools\cmake\3.16.4\bin;C:\Users\lglin\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin;C:\Users\lglin\.espressif\tools\ninja\1.10.2;C:\Users\lglin\.espressif\tools\idf-exe\1.0.1;C:\Users\lglin\.espressif\tools\ccache\3.7;C:\Users\lglin\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\lglin\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\lglin\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM13
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitjv1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\lglin\.espressif
Git Path (idf.gitPath) C:\Users\lglin\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\lglin\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\lglin\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\lglin\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\Users\lglin\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin: true
Access to C:\Users\lglin\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\lglin\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\lglin\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\Users\lglin\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin: true
Access to C:\Users\lglin\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\lglin\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\lglin\.espressif\tools\ccache\3.7: true
[... 88 lines cut ...]
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------

brianignacio5, 2022-03-01:
Could you share the openOCD output?

Also make sure to use these settings to increase the log level:

Use `idf.openOcdDebugLevel` configuration setting to 3 or more to show debug logging in OpenOCD server output.

Use `logLevel` in your <project-directory>/.vscode/launch.json to 3 or more to show more debug adapter output.

lgLindstrom, 2022-03-01:
Hi,, thanks for quick response.

This is a my settings file.

```
{
    "C_Cpp.intelliSenseEngine": "Tag Parser",
    "idf.adapterTargetName": "esp32",
    "idf.customExtraPaths": "C:\\Users\\lglin\\.espressif\\tools\\xtensa-esp32-elf\\esp-2021r1-8.4.0\\xtensa-esp32-elf\\bin;C:\\Users\\lglin\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2021r1-8.4.0\\xtensa-esp32s2-elf\\bin;C:\\Users\\lglin\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2021r1-8.4.0\\xtensa-esp32s3-elf\\bin;C:\\Users\\lglin\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r1-8.4.0\\riscv32-esp-elf\\bin;C:\\Users\\lglin\\.espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Users\\lglin\\.espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Users\\lglin\\.espressif\\tools\\cmake\\3.16.4\\bin;C:\\Users\\lglin\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20210401\\openocd-esp32\\bin;C:\\Users\\lglin\\.espressif\\tools\\ninja\\1.10.2;C:\\Users\\lglin\\.espressif\\tools\\idf-exe\\1.0.1;C:\\Users\\lglin\\.espressif\\tools\\ccache\\3.7;C:\\Users\\lglin\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\lglin\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.espIdfPathWin": "C:\\Users\\lglin\\esp\\esp-idf",
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitjv1.cfg",
        "target/esp32.cfg"
    ],
    "idf.portWin": "COM13",
    "idf.pythonBinPathWin": "C:\\Users\\lglin\\.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "C:\\Users\\lglin\\.espressif",
    "idf.openOcdDebugLevel": 4, 
}
```

This is my launch file:
```
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      "logLevel": 4
    }
  ]
}
```

This is from output window:

```
2022-03-01 10:50:14,860 - Debug Adapter (main) - DEBUG - START_TIME_2022_03_01_10-50-14
2022-03-01 10:50:14,860 - Debug Adapter (main) - DEBUG - Init errors:None
2022-03-01 10:50:14,861 - Debug Adapter (main) - INFO - Working directory: C:\Users\lglin\AppData\Local\Programs\Microsoft VS Code
2022-03-01 10:50:14,861 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 4,
    'developer_mode': 'none',
    'device_name': 'esp32',
    'elfpath': (   'c:\\Users\\lglin\\OneDrive\\experimentalProjects\\jtag\\build\\jtag.elf',),
    'log_file': 'c:\\Users\\lglin\\OneDrive\\experimentalProjects\\jtag\\debug.log',
    'log_mult_files': None,
    'log_no_debug_console': None,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 43474,
    'postmortem': False,
    'tmo_scale_factor': 1,
    'toolchain_prefix': 'xtensa-esp32-elf-'}
DEBUG_ADAPTER_STARTED
2022-03-01 10:50:14,861 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
2022-03-01 10:50:14,862 - Debug Adapter (main) - INFO - Starting. Cmd: c:\Users\lglin\.vscode\extensions\espressif.esp-idf-extension-1.3.0\esp_debug_adapter\debug_adapter_main.py -d 4 -e c:\Users\lglin\OneDrive\experimentalProjects\jtag\build\jtag.elf -l c:\Users\lglin\OneDrive\experimentalProjects\jtag\debug.log -p 43474 -dn esp32 -a 0x10000 -t xtensa-esp32-elf-

2022-03-01 10:50:14,868 - Debug Adapter (main) - INFO - Listening on port 43474 (host: DESKTOP-LG,local ip: 192.168.68.223)
DEBUG_ADAPTER_READY2CONNECT
[... 137 lines cut ...]
Listening on port 43474 (host: DESKTOP-LG,local ip: 192.168.68.223)
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
Got connection
Got json: {
    "arguments": {
        "adapterID": "espidf",
        "clientID": "vscode",
        "clientName": "Visual Studio Code",
        "columnsStartAt1": true,
        "linesStartAt1": true,
        "locale": "sv",
        "pathFormat": "path",
        "supportsInvalidatedEvent": true,
        "supportsMemoryReferences": true,
        "supportsProgressReporting": true,
        "supportsRunInTerminalRequest": true,
        "supportsVariablePaging": true,
        "supportsVariableType": true
    },
    "command": "initialize",
    "seq": 1,
    "type": "request"
}
Created gdb object of type: GdbEsp32
```

brianignacio5, 2022-03-01:
There should more detail in a `OpenOCD` output (This is the one from debug adapter) but there is one error in the log 

```
DEBUG - RESULT: error {'msg': '127.0.0.1:3333: Det gick inte att g366ra en anslutning eftersom '
        'm345ldatorn aktivt nekade det.'}
```

Which should point of something happening in the openOCD. You could try to increase the gdb timeout factor by setting  `tmoScaleFactor` to a number in the launch.json

lgLindstrom, 2022-03-01:
-"There should more detail in a OpenOCD .."

I don't understand ???

This is a view before I start debugging. 

[image]

Then after started debugging I get a timeout,,, it takes 4-5 seconds.

[image]

In the "terminal" window nothing have changed. 
I have presented "output" and "debug console" content for you.

Are you suggesting that a new shell is supposed to open ?

brianignacio5, 2022-03-01:
Next to the Debug Console there is a tab called `Output`. This will show output from many parts of vscode and extensions. There should be a dropdown to choose the output source in which there is one called `OpenOCD`

lgLindstrom, 2022-03-01:
Thank

This is the output from openocd  

```
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
User : 13 2 options.c:57 configuration_output_handler(): debug_level: 4
User : 14 2 options.c:57 configuration_output_handler(): 
Debug: 15 2 options.c:181 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 16 2 options.c:182 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 17 3 options.c:183 add_default_dirs(): exepath=C:/Users/lglin/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin
Debug: 18 3 options.c:184 add_default_dirs(): bin2data=../share/openocd
Debug: 19 3 configuration.c:42 add_script_search_dir(): adding C:\Users\lglin\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts
Debug: 20 4 configuration.c:42 add_script_search_dir(): adding C:\Users\lglin\AppData\Roaming/OpenOCD
Debug: 21 4 configuration.c:42 add_script_search_dir(): adding C:/Users/lglin/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin/../share/openocd/site
Debug: 22 4 configuration.c:42 add_script_search_dir(): adding C:/Users/lglin/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin/../share/openocd/scripts
User : 23 6 command.c:695 co❌ mmand_run_line(): embedded:startup.tcl:26: Error: Can't find interface/ftdi/esp32_devkitjv1.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
[Stopped] : OpenOCD Server
```

brianignacio5, 2022-03-01:
Change `idf.openOcdConfigs` in your settings.json to

```
"idf.openOCDConfigs": ["interface/ftdi/esp32_devkitj_v1.cfg", "target/esp32.cfg"]
```

lgLindstrom, 2022-03-01:
Possible one step further,, still not working.
Adding openocd output. 

I can still see this error:

```
2022-03-01 12:47:34,751 - Gdb - DEBUG - RESULT: error {'msg': '127.0.0.1:3333: Det gick inte att g366ra en anslutning eftersom '
        'm345ldatorn aktivt nekade det.'}
```
Translated the message is something like:
Unable to make a connection because ?? computer denied it.

```
2022-03-01 12:47:24,276 - Debug Adapter (main) - DEBUG - START_TIME_2022_03_01_12-47-24
2022-03-01 12:47:24,277 - Debug Adapter (main) - DEBUG - Init errors:None
2022-03-01 12:47:24,277 - Debug Adapter (main) - INFO - Working directory: C:\Users\lglin\AppData\Local\Programs\Microsoft VS Code
2022-03-01 12:47:24,277 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 4,
    'developer_mode': 'none',
    'device_name': 'esp32',
    'elfpath': (   'c:\\Users\\lglin\\OneDrive\\experimentalProjects\\jtag\\build\\jtag.elf',),
    'log_file': 'c:\\Users\\lglin\\OneDrive\\experimentalProjects\\jtag\\debug.log',
    'log_mult_files': None,
    'log_no_debug_console': None,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 43474,
    'postmortem': False,
    'tmo_scale_factor': 1,
    'toolchain_prefix': 'xtensa-esp32-elf-'}
DEBUG_ADAPTER_STARTED
2022-03-01 12:47:24,277 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
2022-03-01 12:47:24,277 - Debug Adapter (main) - INFO - Starting. Cmd: c:\Users\lglin\.vscode\extensions\espressif.esp-idf-extension-1.3.0\esp_debug_adapter\debug_adapter_main.py -d 4 -e c:\Users\lglin\OneDrive\experimentalProjects\jtag\build\jtag.elf -l c:\Users\lglin\OneDrive\experimentalProjects\jtag\debug.log -p 43474 -dn esp32 -a 0x10000 -t xtensa-esp32-elf-

2022-03-01 12:47:24,283 - Debug Adapter (main) - INFO - Listening on port 43474 (host: DESKTOP-LG,local ip: 192.168.68.223)
DEBUG_ADAPTER_READY2CONNECT
2022-03-01 12:47:24,284 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
2022-03-01 12:47:24,330 - Debug Adapter (main) - DEBUG - Got connection
2022-03-01 12:47:24,331 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug adapter -> Extension: DEBUG_ADAPTER_STARTED\n", "source": {}}, "seq": 0}

2022-03-01 12:47:24,331 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Starting. Cmd: c:\\Users\\lglin\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\esp_debug_adapter\\debug_adapter_main.py -d 4 -e c:\\Users\\lglin\\OneDrive\\experimentalProjects\\jtag\\build\\jtag.elf -l c:\\Users\\lglin\\OneDrive\\experimentalProjects\\jtag\\debug.log -p 43474 -dn esp32 -a 0x10000 -t xtensa-esp32-elf-\n\n", "source": {}}, "seq": 1}

2022-03-01 12:47:24,332 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Listening on port 43474 (host: DESKTOP-LG,local ip: 192.168.68.223)\n", "source": {}}, "seq": 2}

2022-03-01 12:47:24,332 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT\n", "source": {}}, "seq": 3}

2022-03-01 12:47:24,332 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Got connection\n", "source": {}}, "seq": 4}

2022-03-01 12:47:24,663 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'Content-Length: 407\\r\\n'<<

2022-03-01 12:47:24,663 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'\\r\\n'<<

2022-03-01 12:47:24,663 - Debug Adapter(Command Processor) - DEBUG - Got json: {
    "arguments": {
        "adapterID": "espidf",
        "clientID": "vscode",
        "clientName": "Visual Studio Code",
        "columnsStartAt1": true,
        "linesStartAt1": true,
        "locale": "sv",
        "pathFormat": "path",
        "supportsInvalidatedEvent": true,
        "supportsMemoryReferences": true,
[... 72 lines cut ...]
2022-03-01 12:47:34,997 - Debug Adapter(Command Processor) - DEBUG - Got json: {
    "arguments": {
        "restart": false
    },
    "command": "disconnect",
    "seq": 2,
    "type": "request"
}
2022-03-01 12:47:34,997 - Debug Adapter (main) - DEBUG - Adapter is Stopping...
2022-03-01 12:47:34,997 - Debug Adapter (main) - DEBUG - Stopping target poller
2022-03-01 12:47:34,997 - Debug Adapter (main) - DEBUG - Closing the GDB process
2022-03-01 12:47:34,997 - Debug Adapter (main) - DEBUG - Closing the Socket
2022-03-01 12:47:34,997 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Got json: {\n    \"arguments\": {\n        \"restart\": false\n    },\n    \"command\": \"disconnect\",\n    \"seq\": 2,\n    \"type\": \"request\"\n}\n", "source": {}}, "seq": 9}

2022-03-01 12:47:34,998 - Debug Adapter (main) - DEBUG - Stopping of the Writer thread
2022-03-01 12:47:34,998 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug Adapter stopped\n", "source": {}}, "seq": 10}

2022-03-01 12:47:34,998 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread
2022-03-01 12:47:34,998 - Debug Adapter (main) - DEBUG - Adapter is Stopped
DEBUG_ADAPTER_STOPPED
2022-03-01 12:47:34,998 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
2022-03-01 12:47:34,998 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect
[Stopped] : ESP-IDF Debug Adapter

```

brianignacio5, 2022-03-01:
and the new openOCD output ? The problem seems to be there.

lgLindstrom, 2022-03-01:
File to long to publish in window.
Adding as attachemen
[openocd.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8161588/openocd.txt)
t

brianignacio5, 2022-03-01:
need to lower a bit the openOCd log level to see what is happening :(

lgLindstrom, 2022-03-01:
Please dont do this to me ......
Suddenly it is working. 
Applications stops on the first line.  I can set and remove breakpoint. Step between them.
I have experienced this before. Working for a short time, suddenly ,,, not working.

I have added the output files.
[esp_idf_output.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8162011/esp_idf_output.txt)
[openocd_output.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8162012/openocd_output.txt)

brianignacio5, 2022-03-02:
Can you lower the openOCD log level ? Is too much information to see the error 

You say it works and then it doesn't? Could you provide some steps to reproduce this ?

brianignacio5, 2022-03-03:
Too much logging can affect the openOCD performance because many timing operations.

Thanks, closing this issue .

closed 2022-03-03 by brianignacio5

lgLindstrom, 2022-03-03:
Just one more question.

Will the change get into the vscode extension?

brianignacio5, 2022-03-07:
Which change do you mean ? You can change these values with the `ESP-IDF: Select OpenOCD Board Configuration` command already

lgLindstrom, 2022-03-07:
That's true, but I used the default blink template when I create the project and apparently it produced something that did not work.
But maybe I miss something.

Anyhow, thanks again.