espressif/vscode-esp-idf-extension #658: Cannot flash via JTAG

raffaeler, 2022-03-01:
**Pre Bug Report Checklist**
- [X] Make sure you have searched for existing bugs and features request before you post an issue.
- [X] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [X] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
I am working with a ESP32-WROVER-KIT which is configured with the JTAG interface as for the docs.
I can successfully flash the firmware using the command line:
```
openocd -d3 -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover-kit-3.3v.cfg
```

Instead, flashing from VSCode fails.

**To Reproduce**
I have tried several `settings.json` configurations. This one looks like the most reasonable:
```
{
    "idf.adapterTargetName": "esp32",
    "idf.openOcdDebugLevel": 3,
    "idf.openOcdConfigs": [
        "board/esp32-wrover-kit-3.3v.cfg"
    ],
    "cmake.configureOnOpen": false,
}
```

From process explorer, I can see the command line launched from VSCode is the following:
```
openocd -d3 -f board/esp32-wrover-kit-3.3v.cfg
```

Anyway, the flashing process starts but it hangs.

The complete (level 3) log is the following:
```
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 5 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 5 options.c:63 configuration_output_handler(): 
Debug: 15 5 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 16 6 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 17 6 options.c:246 add_default_dirs(): exepath=C:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
Debug: 18 6 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 19 6 configuration.c:42 add_script_search_dir(): adding C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts
Debug: 20 7 configuration.c:42 add_script_search_dir(): adding C:/Users/raffaeler/AppData/Roaming/OpenOCD
Debug: 21 7 configuration.c:42 add_script_search_dir(): adding C:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/site
Debug: 22 8 configuration.c:42 add_script_search_dir(): adding C:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/scripts
Debug: 23 8 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/board/esp32-wrover-kit-3.3v.cfg
Debug: 24 11 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/interface/ftdi/esp32_devkitj_v1.cfg
Debug: 25 11 command.c:146 script_debug(): command - adapter driver ftdi
Debug: 27 11 command.c:146 script_debug(): command - ftdi_vid_pid 0x0403 0x6010 0x0403 0x6014
Debug: 29 12 command.c:146 script_debug(): command - ftdi_channel 0
Debug: 31 12 command.c:146 script_debug(): command - ftdi_layout_init 0x0008 0xf00b
Debug: 33 12 command.c:146 script_debug(): command - ftdi_layout_signal LED -data 0x1000
Debug: 35 12 command.c:146 script_debug(): command - ftdi_layout_signal LED2 -data 0x2000
Debug: 37 12 command.c:146 script_debug(): command - ftdi_layout_signal LED3 -data 0x4000
Debug: 39 13 command.c:146 script_debug(): command - ftdi_layout_signal LED4 -data 0x8000
Debug: 41 13 command.c:146 script_debug(): command - reset_config none
Debug: 43 13 command.c:146 script_debug(): command - adapter speed 20000
Debug: 45 13 core.c:1822 jtag_config_khz(): handle jtag khz
Debug: 46 13 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 47 14 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 48 14 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/target/esp32.cfg
Debug: 49 14 command.c:146 script_debug(): command - transport select jtag
Debug: 50 15 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/target/esp_common.cfg
Debug: 51 16 command.c:146 script_debug(): command - add_help_text program_esp write an image to flash, address is only required for binary images. verify, reset, exit, compress, restore_clock are optional
Debug: 53 16 command.c:1115 help_add_command(): added 'program_esp' help text
[... 347 lines cut ...]
    "level": "error",
    "timestamp": "2022-03-01T10:17:48.520Z"
}
{
    "message": "Cannot access filePath: ${env:MDF_PATH}",
    "stack": "Error: ENOENT: no such file or directory, access '${env:MDF_PATH}'\n\tat Object.accessSync (fs.js:220:3)\n\tat Object.e.accessSync (electron/js2c/asar_bundle.js:5:7989)\n\tat I (c:\\Users\\raffaeler\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:1:4561)\n\tat c:\\Users\\raffaeler\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:8:247757\n\tat Generator.next (<anonymous>)\n\tat c:\\Users\\raffaeler\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:8:242209\n\tat new Promise (<anonymous>)\n\tat l (c:\\Users\\raffaeler\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:8:241954)\n\tat c:\\Users\\raffaeler\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:8:247384\n\tat c:\\Users\\raffaeler\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:8:248464\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\raffaeler\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:8:245882)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)",
    "level": "error",
    "timestamp": "2022-03-01T10:17:48.521Z"
}
{
    "message": "Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n",
    "stack": "Error: Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\raffaeler\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:1:7304\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)",
    "level": "error",
    "timestamp": "2022-03-01T10:17:50.035Z"
}
{
    "message": "Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n",
    "stack": "Error: Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\raffaeler\\.vscode\\extensions\\espressif.esp-idf-extension-1.3.0\\dist\\extension.js:1:7304\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)",
    "level": "error",
    "timestamp": "2022-03-01T10:20:58.969Z"
}
```

**Additional context**
VSCode was launched from the Espressif command prompt as "code ." to inherit the environment variables.

raffaeler, 2022-03-01:
**ESP-IDF Extension for Visual Studio Code report**
```
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 C:\Espressif\python_env\idf4.4_py3.8_env 
System environment variable PATH 
 C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch2-8.4.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Espressif\tools\cmake\3.20.3\bin;C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Espressif\frameworks\esp-idf-v4.4\tools;C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\;C:\Espressif\tools\idf-git\2.34.2\cmd\;C:\Espressif;C:\Program Files\Eclipse Foundation\jdk-8.0.302.8-hotspot\bin;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\android\android-sdk\platform-tools;C:\android\android-sdk\emulator;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Microsoft SDKs\Azure\CLI2\wbin;C:\WINDOWS;C:\WINDOWS\system32;C:\ProgramData\chocolatey\bin;C:\Program Files\Broadcom\Broadcom 802.11 Network Adapter;C:\Program Files\Java\jdk1.8.0_102\bin;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\EPSON\UPOSDriverForFiscal\OPOS.NET;C:\Program Files (x86)\Lexmark\PrinterDriverConfigurationUtility\;C:\Program Files\NVIDIA Corporation\Nsight Compute 2019.5.0\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\CMake\bin;C:\Program Files (x86)\Yarn\bin\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Pandoc\;C:\Program Files\MiKTeX\miktex\bin\x64\;C:\Program Files\OpenJDK\openjdk-8u292-b10\bin;C:\Program Files\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Git\cmd;C:\Program Files\nodejs\;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files (x86)\Bitvise SSH Client;C:\Users\raffaeler\.cargo\bin;C:\Users\raffaeler\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\raffaeler\AppData\Local\Programs\Python\Python39\;C:\Program Files (x86)\SSH Communications Security\SSH Secure Shell;C:\Program Files (x86)\Linaro\gcc-linaro-arm-linux-gnueabihf-4.9-2014.09\bin;C:\Program Files\Docker Toolbox;C:\Program Files\Microsoft VS Code\bin;C:\Users\raffaeler\AppData\Local\Programs\Fiddler;C:\Users\raffaeler\AppData\Local\GitHubDesktop\bin;H:\util;C:\Users\raffaeler\AppData\Roaming\nvm;C:\Program Files\nodejs;C:\Users\raffaeler\.dotnet\tools;C:\Users\raffaeler\AppData\Local\Markdown Monster;C:\Users\raffaeler\.dotnet\tools;C:\Users\raffaeler\AppData\Local\Microsoft\WindowsApps;C:\Users\raffaeler\AppData\Local\Microsoft\WindowsApps;C:\intelFPGA_lite\20.1\modelsim_ase\win32aloem;C:\Users\raffaeler\AppData\Roaming\npm;C:\Program Files\OpenSSL-Win64\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.64.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.3.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/frameworks/esp-idf-v4.4/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch2-8.4.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Espressif\tools\cmake\3.20.3\bin;C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
[... cut ...]
---------------------------------------------------- Visual Studio Code c_cpp_properties.json ----------------------------------------------
{
    "configurations": [
      {
        "name": "ESP-IDF",
        "cStandard": "c11",
        "cppStandard": "c++17",
        "compileCommands": "${workspaceFolder}/build/compile_commands.json"
      }
    ],
    "version": 4
  } 
--------------------------------------------------------------------------------------------------------------------------------------------
```

raffaeler, 2022-03-01:
** SDKConfig file ** 
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
# end of Bootloader config

[... 1285 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```

brianignacio5, 2022-03-01:
Could you try the current extension in GitHub master with this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/171864844.zip) there is a couple of fixes related to JTAG.

Will soon be pushed to marketplace!

raffaeler, 2022-03-01:
@brianignacio5 I just installed version 1.4.0 from the marketplace.
Thanks

closed 2022-03-01 by raffaeler