espressif/vscode-esp-idf-extension #659: Cannot monitor when the board is configured in JTAG mode

raffaeler, 2022-03-01:
## Environment
All the configuration files and versions are the same of this other thread: https://github.com/espressif/vscode-esp-idf-extension/issues/658
The only exception is that I now updated the VSCode extension to `version 1.4.0`.

## Issue description
Since I previously programmed the board using the serial port, I know the sample output data on the serial port (Monitor). I tested both the `peripheral/spi_master/lcd` and the `peripheral/get-started/blink` samples.

When I switch the board to JTAG, the serial port used to Monitor/logging purposes does not exists anymore as it now used from JTAG.
The problem is that now I cannot see the Monitor output anymore, even if I select the other serial port (`COM11` in my case, which is mapped to the FTDI driver named `Interface 1`) .

The output of the monitor is listed below.
How can I Monitor the board that is in JTAG mode?

## Monitor output
```
> C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe C:\Espressif\frameworks\esp-idf-v4.4\tools\idf.py -p COM11 monitor
Executing action: monitor
Running idf_monitor in directory h:\dev.hw\esp-samples\lcd
Executing "C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4/tools/idf_monitor.py -p COM11 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 h:\dev.hw\esp-samples\lcd\build\spi_master.elf -m 'C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'C:\Espressif\frameworks\esp-idf-v4.4\tools\idf.py' '-p' 'COM11'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM11 instead...
--- idf_monitor on \\.\COM11 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
```

raffaeler, 2022-03-01:
It looks like a documentation issue: restoring the serial jumpers of the wrover kit **in addition** to the JTAG ones, made the serial monitor work.
The documentation just talk about enabling either the serial jumpers **or** the JTAG ones.

If someone confirms, this issue can be closed. Thanks.

brianignacio5, 2022-03-15:
I think this is not an extension issue but the understanding of JTAG/UART structure.

PTAL at the [JTAG documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/index.html#how-it-works) where the monitoring/logging is specified through UART (although you could add logging to JTAG as shown in the [app trace to host example](https://github.com/espressif/esp-idf/tree/abea9e4c02bb17e86298aec4e299780399e4789f/examples/system/app_trace_to_host) )

closed 2022-03-15 by brianignacio5