espressif/vscode-esp-idf-extension #660: [Bug Report]: v1.4.0 no longer shows build terminal window

rtheil-growlink, 2022-03-02:
With the update to 1.4.0 I can't see the progress of a build in the terminal window because it's hidden. If I want to see the build terminal window, I have to put an error in code and build in order to see that terminal window. Once that terminal window is visible, it is no longer focused when a build starts. We, as developers, rely on the progress window of compilation, even when successful. I need to see the flash progress, and on chips that require the setup button to be pressed I need to see when it's waiting for me to press that button. Why remove it? Why not make it an option?

Tried downgrading to 1.3.0 and can't get it back.

taamo-code, 2022-03-03:
因为这个原因，我目前也选择使用v1.3.0

brianignacio5, 2022-03-03:
You just need to set `idf.notificationSilentMode: true` in your settings.json

rtheil-growlink, 2022-03-03:
That works. I think that option is a good addition. It'll be nice to not have the error notifications stick around until dismissed.

closed 2022-03-03 by rtheil-growlink