espressif/vscode-esp-idf-extension #661: [Feature Request]:  Bring back display of COM port in status bar

rtheil-growlink, 2022-03-02:
v1.2.0 of the extension showed the selected COM port  in the tool bar at the bottom of VS code. v1.3.0 removed that, and now there's no way to see which is selected. I am currently developing for 4 different controllers using the ESP32 and I have no idea which COM Port is selected until I try to flash or I choose the COM port again. Please bring the display of the selected COM Port as that was a very useful feature.

taamo-code, 2022-03-03:
I agree with the proposal！Because it's important

brianignacio5, 2022-03-03:
The feature was never removed. Could you show an screenshot of your status bar? Maybe because of other extension icons is hidden.

rtheil-growlink, 2022-03-03:
For reference:
[image]

I'll try out that vsix and see if it fixes the issue.