espressif/vscode-esp-idf-extension #663: [Bug Report]: QEMU Doesn't Launch: "Cannot read property 'fsPath' of undefined"

stevenoonan, 2022-03-04:
**Describe the bug**
The QEMU feature does not launch.

I click the [ESP-IDF QEMU] link on the activity bar. It creates merged_qemu.bin, and then prompts me to select Start QEMU. Immediately a status box appears "Cannot read property 'fsPath' of undefined".

**To Reproduce**
I am following the QEMU steps here: https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/QEMU.md

I created the Hello World example using the ESP-IDF: Show Examples Projects in VSCode v1.65 on Windows 11. The I opened the container using VSCode's dev container support. It opens and builds the container.

(Side note: The first thing that VSCode complains about the "ESP-IDF" configuration in c_cpp_properties.json has a bad "compilerPath" because it is still referencing the windows path. I can fix this manually but it would be nice if this was taken care of automatically somehow. Obviously if I go back to developing without the container in Windows the path is again invalid.)

When I run "Build your project" it fails because I had already built the project in windows. So I remove the windows built 'build' folder. It then builds successfully.

I then click the [ESP-IDF QEMU] link on the activity bar. It creates merged_qemu.bin, and then prompts me to select Start QEMU. Immediately a status box appears "Cannot read property 'fsPath' of undefined".

I have figured out how to manually get QEMU running  and launch a gdb session through VS Code so I know all the tools and files are there correctly.

**Expected behavior**
QEMU would launch, something like: 
`qemu-system-xtensa -nographic -gdb tcp:3333 -S -machine esp32 -drive file=build/merged_qemu.bin,if=mtd,format=raw`

`gdb build/merged_qemu.elf` would also run with the following start up commands: 
```
target remote localhost:3333
monitor system_reset
tb app_main
c
```
**Environment (please complete the following information):**

- OS Name & Version: Windows 11 21H1 19043.1556, WSL2 Docker
- VSCode Version: 1.65.0
- ESP-IDF Version: 1.4.0

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.10.60.1-microsoft-standard-WSL2 
System environment variable IDF_PYTHON_ENV_PATH 
 /opt/esp/python_env/idf4.3_py3.6_env 
System environment variable PATH 
 /vscode/vscode-server/bin/linux-x64/b5205cc8eb4fbaa726835538cd82372cc0222d43/bin/remote-cli:/opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin:/opt/esp/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin:/opt/esp/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/opt/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/opt/esp/tools/cmake/3.16.4/bin:/opt/esp/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/opt/esp/python_env/idf4.3_py3.6_env/bin:/opt/esp/idf/tools:/opt/qemu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.65.0 
Visual Studio Code language en 
Visual Studio Code shell /bin/bash 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /opt/esp/idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/opt/esp/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /opt/esp/python_env/idf4.3_py3.6_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) /opt/esp
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.17.1
[... 86 lines cut ...]
 
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: /opt/esp/python_env/idf4.3_py3.6_env/bin/python /opt/esp/idf/tools/check_python_dependencies.py -r "/root/.vscode-server/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/requirements.txt"

--------------------------------------------------------------------------------------------------------------------------------------------

```

**Logs**
The relevant logged error in /root/.vscode-server/extensions/espressif.esp-idf-extension-1.4.0/esp_idf_vsc_ext.log:
`{"user":true,"message":"Cannot read property 'fsPath' of undefined","stack":"TypeError: Cannot read property 'fsPath' of undefined\n\tat u.<anonymous> (/dist/extension.js:8:594999)\n\tat Generator.next (<anonymous>)\n\tat /dist/extension.js:8:593214\n\tat new Promise (<anonymous>)\n\tat l (/dist/extension.js:8:592959)\n\tat u.start (/dist/extension.js:8:594820)\n\tat u.<anonymous> (/dist/extension.js:8:593804)\n\tat Generator.next (<anonymous>)\n\tat s (/dist/extension.js:8:593016)\n\tat runMicrotasks (<anonymous>)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2022-03-04T20:08:01.927Z"}`

**Extension Configuration settings**
Entirety of settings.json is:
```
{
    "idf.gitPath": "/usr/bin/git",
    "idf.espIdfPathWin": "C:\\Users\\snoonan\\esp\\esp-idf",
    "idf.pythonBinPathWin": "C:\\Users\\snoonan\\.espressif\\python_env\\idf4.4_py3.8_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "C:\\Users\\snoonan\\.espressif",
    "idf.customExtraPaths": "C:\\Users\\snoonan\\.espressif\\tools\\xtensa-esp32-elf\\esp-2021r2-patch2-8.4.0\\xtensa-esp32-elf\\bin;C:\\Users\\snoonan\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2021r2-patch2-8.4.0\\xtensa-esp32s2-elf\\bin;C:\\Users\\snoonan\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2021r2-patch2-8.4.0\\xtensa-esp32s3-elf\\bin;C:\\Users\\snoonan\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r2-patch2-8.4.0\\riscv32-esp-elf\\bin;C:\\Users\\snoonan\\.espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Users\\snoonan\\.espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Users\\snoonan\\.espressif\\tools\\cmake\\3.20.3\\bin;C:\\Users\\snoonan\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20211220\\openocd-esp32\\bin;C:\\Users\\snoonan\\.espressif\\tools\\ninja\\1.10.2;C:\\Users\\snoonan\\.espressif\\tools\\idf-exe\\1.0.3;C:\\Users\\snoonan\\.espressif\\tools\\ccache\\4.3\\ccache-4.3-windows-64;C:\\Users\\snoonan\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"/opt/esp/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts\"}",
    "idf.ninjaArgs": ["-j 8"]
}
```

broose-goose, 2022-03-05:
@stevenoonan I ran across this same issue as I was installing ESP-IDF and trying to run QEMU.

I got further by following the instructions here: https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/QEMU.md

Apparently QEMU integration doesn't run on the host, but runs in a docker container.

broose-goose, 2022-03-05:
Update: I followed those steps, build the project in the docker container, and still had this issue... so there's that

brianignacio5, 2022-03-07:
Could you try [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/179466428.zip)

stevenoonan, 2022-03-07:
@brianignacio5 The new installer fixes the issue. I was able to start QEMU, debug it and see output on the IDF Monitor. Thanks!

Mair, 2022-05-08:
@brianignacio5 
Can we please re-open this issue? its still fails and I cant install the extension from the link. 
error log

```8/05/2022 5:43:49 PM - Found installed product - Global Location
8/05/2022 5:43:49 PM - Found installed product - Visual Studio Community 2022 Current
8/05/2022 5:43:49 PM - VSIXInstaller.NoApplicableSKUsException: One or more extensions are for Visual Studio Code. Try installing them in Visual Studio Code.
   at VSIXInstaller.ExtensionService.GetInstallableDataImpl(IInstallableExtension extension, String extensionPackParentName, Boolean isRepairSupported, IStateData stateData, IEnumerable`1& skuData)
   at VSIXInstaller.ExtensionService.GetInstallableData(String vsixPath, String extensionPackParentName, Boolean isRepairSupported, IStateData stateData, IEnumerable`1& skuData)
   at VSIXInstaller.ExtensionPackService.IsExtensionPack(IStateData stateData, Boolean isRepairSupported)
   at VSIXInstaller.ExtensionPackService.ExpandExtensionPackToInstall(IStateData stateData, Boolean isRepairSupported)
   at VSIXInstaller.App.Initialize(Boolean isRepairSupported)
   at VSIXInstaller.App.Initialize()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.VisualStudio.Telemetry.WindowsErrorReporting.WatsonReport.GetClrWatsonExceptionInfo(Exception exceptionObject)
```

brianignacio5, 2022-05-09:
Hi @Mair Could you try [vsix installer link](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/222158785.zip) ?

Maybe the link is unavailable since we closed the PR

Mair, 2022-05-10:
Hi @brianignacio5,

Trying to install it I get 
[image]

Mair, 2022-05-10:
The log...

```
11/05/2022 9:11:26 AM - Microsoft VSIX Installer
11/05/2022 9:11:26 AM - -------------------------------------------
11/05/2022 9:11:26 AM - vsixinstaller.exe version:
11/05/2022 9:11:26 AM - 17.0.5226-preview5
11/05/2022 9:11:26 AM - -------------------------------------------
11/05/2022 9:11:26 AM - Command line parameters:
11/05/2022 9:11:26 AM - C:\Program Files (x86)\Microsoft Visual Studio\Installer\resources\app\ServiceHub\Services\Microsoft.VisualStudio.Setup.Service\VSIXInstaller.exe,C:\Users\mairs\AppData\Local\Temp\Rar$DIa23916.44144\esp-idf-extension.vsix
11/05/2022 9:11:26 AM - -------------------------------------------
11/05/2022 9:11:26 AM - Microsoft VSIX Installer
11/05/2022 9:11:26 AM - -------------------------------------------
11/05/2022 9:11:27 AM - Skipping product Microsoft.VisualStudio.Product.BuildTools (dd647c74) since it does not support extensions
11/05/2022 9:11:27 AM - Initializing Install...
11/05/2022 9:11:27 AM - Extension Details...
11/05/2022 9:11:27 AM - 	Identifier         : esp-idf-extension
11/05/2022 9:11:27 AM - 	Name               : Espressif IDF
11/05/2022 9:11:27 AM - 	Author             : espressif
11/05/2022 9:11:27 AM - 	Version            : 1.4.0
11/05/2022 9:11:27 AM - 	Description        : Develop and debug applications for Espressif ESP32, ESP32-S2 chips with ESP-IDF
11/05/2022 9:11:27 AM - 	Locale             : en-US
11/05/2022 9:11:27 AM - 	MoreInfoURL        : 
11/05/2022 9:11:27 AM - 	InstalledByMSI     : False
11/05/2022 9:11:27 AM - 	SupportedFrameworkVersionRange : [0.0,2147483647.2147483647]
11/05/2022 9:11:27 AM - 
11/05/2022 9:11:27 AM - 	SignatureState     : Unsigned
11/05/2022 9:11:27 AM - 	Supported Products : 
11/05/2022 9:11:27 AM - 		Microsoft.VisualStudio.Code
11/05/2022 9:11:27 AM - 			Version : 
11/05/2022 9:11:27 AM - 			ProductArchitecture : x86
11/05/2022 9:11:27 AM - 
11/05/2022 9:11:27 AM - 	References         : 
11/05/2022 9:11:27 AM - Signature Details...
11/05/2022 9:11:27 AM - 	Extension is not signed.
11/05/2022 9:11:27 AM - 
11/05/2022 9:11:27 AM - Searching for applicable products...
11/05/2022 9:11:27 AM - Found installed product - Global Location
11/05/2022 9:11:27 AM - Found installed product - Visual Studio Community 2022 Current
11/05/2022 9:11:27 AM - VSIXInstaller.NoApplicableSKUsException: One or more extensions are for Visual Studio Code. Try installing them in Visual Studio Code.
   at VSIXInstaller.ExtensionService.GetInstallableDataImpl(IInstallableExtension extension, String extensionPackParentName, Boolean isRepairSupported, IStateData stateData, IEnumerable`1& skuData)
   at VSIXInstaller.ExtensionService.GetInstallableData(String vsixPath, String extensionPackParentName, Boolean isRepairSupported, IStateData stateData, IEnumerable`1& skuData)
   at VSIXInstaller.ExtensionPackService.IsExtensionPack(IStateData stateData, Boolean isRepairSupported)
   at VSIXInstaller.ExtensionPackService.ExpandExtensionPackToInstall(IStateData stateData, Boolean isRepairSupported)
   at VSIXInstaller.App.Initialize(Boolean isRepairSupported)
   at VSIXInstaller.App.Initialize()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.VisualStudio.Telemetry.WindowsErrorReporting.WatsonReport.GetClrWatsonExceptionInfo(Exception exceptionObject)
```

brianignacio5, 2022-05-11:
Well @Mair this extension is  for Visual Studio Code, not for Visual Studio.

We don't  have an official Visual Studio extension yet but is something we could implement in the future if many people need!

Mair, 2022-05-11:
@brianignacio5 I just launched it and that's what I get. I am using Vs code. So how do I target it to Vs code and not visual studio?

brianignacio5, 2022-05-12:
Follow the steps to install the extension from vsix [here](https://code.visualstudio.com/docs/editor/extension-marketplace#_install-from-a-vsix)

Mair, 2022-05-12:
@brianignacio5 thanks.
installed OK
[image]

restarted vscode
but error still remains
[image]

brianignacio5, 2022-05-13:
Could you share the extension log @Mair ?

Mair, 2022-05-14:
@brianignacio5 Sorry, How do I get that?

brianignacio5, 2022-05-14:
As shown in the [troubleshooting section](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting), the log file which can be obtained from:
* ` Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
*  `Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

laurento, 2022-07-17:
Just tried on my Debian, clean installation. Same error when I try to "Start QEMU"

Mair, 2022-07-18:
@brianignacio5 thanks for that however, you might consider that people are used to the vscode ecosystem and having to do other steps and side installs means you are going to get a lot of people frustrated with a bad experience.

I have tried it in the past on windows and it still did not work. I believe most people are finding alternative solutions like using the command line or just giving up. 

just a suggestion... create a VM with windows and ubuntu and try setting it up from scratch as a new user who was trying to get into learning the ESP for the first time. If you can do that and get QEMU, debugging, etc. to work, that would be brilliant!

brianignacio5, 2022-07-19:
@Mair Thanks for your feedback. The idea of sharing the vsix installer first is to see if the issue is fixed for the user who's facing the bug before we release to marketplace where more users could be affected.

Of course I've tested on Windows 11, Ubuntu 20.04 and MacOS before I even share the vsix installer. The thing is that I can't reproduce the issue locally (before I did reproduced, found the issue and fixed in GitHub, hence sharing this vsix installer). I spend most of the time trying to make it as smooth and remove as much configuration as possible.

Mair, 2022-07-19:
@brianignacio5 that's good to know.
As mentioned I tested it some time back and it did not work. I can test it again if that helps.
Do I need to uninstall the current ESP extension?

brianignacio5, 2022-07-19:
Yes need to uninstall thanks @Mair .

ingoha, 2022-07-19:
@brianignacio5 Regarding the obvious need to update the extension inside the container: Wouldn't it be easier for all of us if you just updated the Dockerfile to install the right version directly?
Just my two cents ;-)

ingoha, 2022-07-19:
Or perhaps use postCreateCommand in the devcontainer.json?

brianignacio5, 2022-07-19:
the thing is that the vsix installer we share is not a new version but a development release from current master (or sometimes a Pull Request). Of course we could just release a new version but that would be true for all users instead of a small testing group, which is what we need here to make sure things works before we release to everyone. 

We haven't implemented a workflow that gives us the master branch in the marketplace right away but hopefully with the preview release update we can work something up.

laurento, 2022-07-19:
> @laurento @moefear85 We have made some changes and use the latest esp-idf in the container docker file in this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/301447416.zip) so please try. We should have fixed the fsPath issue here.

@brianignacio5 Indeed, if I remove the esp-idf extension from the container and manually install the version you provided it seems to work as expected. Many thanks :)

Mair, 2022-07-20:
hi @brianignacio5 
I tried the following...
1. created a new project using the idf extension
2. launched the new project in the dev container
3. uninstalled the esp extension
4. downloaded and installed the new one
5. created a small project like so
```c
#include <stdio.h>

void app_main(void)
{
    for (size_t i = 0; i < 100; i++)
    {
        printf("count %d\n", count++);
    }
}
````
6. built the project using idf.py build.
7. clicked on the [ESP-IDF QEMU] link at the bottom
8. clicked on Start QEMU and got this
```bash
root@ea4e97f18984:/workspaces/test-qemu# qemu-system-xtensa -nographic -machine esp32 -drive file=build/merged_qemu.bin,if=mtd,format=raw -serial tcp::5555,server,nowait
QEMU 5.2.0 monitor - type 'help' for more information
(qemu) Adding SPI flash device
```
Question... How do I see the output?

9. pressed F1 an selected Launch QENU debug session
got this
[image]
my debug output as follows

```
Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
Starting. Cmd: /root/.vscode-server/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/debug_adapter_main.py -d 5 -e /workspaces/test-qemu/build/test-qemu.elf -l /workspaces/test-qemu/debug.log -p 43474 -dn esp32 -a 0x10000 -t xtensa-esp32-elf- -om without_oocd -x /workspaces/test-qemu/esp-idf-vscode-generated.gdb
Listening on port 43474
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
Got connection
Got json: {
    "arguments": {
        "adapterID": "espidf",
        "clientID": "vscode",
        "clientName": "Visual Studio Code",
        "columnsStartAt1": true,
        "linesStartAt1": true,
        "locale": "en-us",
        "pathFormat": "path",
        "supportsInvalidatedEvent": true,
        "supportsMemoryReferences": true,
        "supportsProgressReporting": true,
        "supportsRunInTerminalRequest": true,
        "supportsVariablePaging": true,
        "supportsVariableType": true
    },
    "command": "initialize",
    "seq": 1,
    "type": "request"
}
Created gdb object of type: GdbEsp32
Debug Adapter initialized
Processed command: initialize
Got json: {
    "arguments": {
        "__sessionId": "b494633f-b881-4d1e-9774-471a273828a7",
        "elfFilePath": "/workspaces/test-qemu/build/test-qemu.elf",
        "name": "GDB QEMU",
        "request": "launch",
        "sessionID": "qemu.debug.session",
        "type": "espidf"
    },
    "command": "launch",
    "seq": 2,
    "type": "request"
}
Failed to wait for completion of command "-interpreter-exec console "source /workspaces/test-qemu/esp-idf-vscode-generated.gdb"" / 5!
Failed to wait for completion of command "-interpreter-exec console "source /workspaces/test-qemu/esp-idf-vscode-generated.gdb"" / 5!

```

takaite, 2022-07-26:
Hello;

I'm facing this issues on release 1.4.0. There is any news about it? I searched for it, but couldn't find anything.

brianignacio5, 2022-07-26:
Have you tried the last vsix installer @takaite ?

mriksman, 2022-12-28:
Having same issue as @Mair.
I run the QEMU Server, and see this
```
root@d789c7701f57:/workspaces# qemu-system-xtensa -nographic -gdb tcp::3333 -S -machine esp32 -drive file=build/merged_qemu.bin,if=mtd,format=raw -serial tcp::5555,server,nowait
QEMU 7.1.0 monitor - type 'help' for more information
(qemu) Adding SPI flash device
```
Then when I run the command ` ESP-IDF: Monitor QEMU Device`, nothing at all happens.

Mair, 2022-12-29:
@mriksman It would be nice to have this working in VSCode but if you are really stuck have a look at https://www.youtube.com/watch?v=lZp9L7Ij4Yo&t=1s

mriksman, 2022-12-29:
@Mair Thanks - that's the video I followed to get to where I am :) In the (your) video, QEMU is run manually. It works fine like that. It's the ESP-IDF buttons that don't work. The QEMU Server starts OK, but the 'Monitor QEMU Device' does nothing.

brianignacio5, 2023-01-03:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/496218227.zip) from the PR #863 ? @Mair  @mriksman

mriksman, 2023-01-17:
@brianignacio5 No good. I installed it, reloaded VSCode, and it still hangs here
[image]
I'm running `espressif/idf:v4.4.2`. Could that cause an issue...?

mriksman, 2023-01-17:
Nice! ESP-IDF Extension 1.5.0 **does** work, 1.5.1 **does not** work, and the vsix installer you provided a link to **does** work.