espressif/vscode-esp-idf-extension #664: Unable to determine what CMake generator to use. Please install or configure a preferred generator, or update settings.json, your Kit configuration or PATH variable. Error: No usable generator found.

allrobot, 2022-03-09:
**ERROR:**
Vscode installed ESP plug-in to build the project, the output console reported an error:
```
[variant] Loaded new set of variants
[kit] Successfully loaded 13 kits from C:\Users\li1gu\AppData\Local\CMakeTools\cmake-tools-kits.json
[proc] Executing command: C:\Users\li1gu\ESP\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -v
[main] Unable to determine what CMake generator to use. Please install or configure a preferred generator, or update settings.json, your Kit configuration or PATH variable. Error: No usable generator found.
[main] Configuring folder: hello_world 
[proc] Executing command: C:\Users\li1gu\ESP\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -v
[main] Unable to determine what CMake generator to use. Please install or configure a preferred generator, or update settings.json, your Kit configuration or PATH variable. Error: No usable generator found.
```
I couldn't build the Hello project because of this error

**Repeat the process:**

These are repeat of `ESP-IDF: Doctor command`:

[report.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8213510/report.txt)
[report.json](https://github.com/allrobot/allrobot-blog/blob/main/report.json)

**Expected behavior:**
Because the online installation package provided on the ESP official website cannot be installed, the exit code is 1.

[Setup Log 2022-03-09 #003.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8213585/Setup.Log.2022-03-09.003.txt)

After installing esp-idf in VSCode and configuring 'configure esp-idf extension', I created Hello via 'esp-idf: Show Examples Projects'
'ESP-IDf: Select a Kit' without the CMake tool previously downloaded by VSCode, Select 'Scan for Kit' and Select this option a few minutes later, as shown in the picture:
[image]

And then I got an error....

Since it says I can't confirm what CMaker to use, I try adding a path to the user variable and restart VSCode
[image]

[image]

Them no had OpenOCD, Heap Trace, SDK.......

**Environment:**
- Windows 10 Professional [21H1][19043.1526]
- VSCode Version: [1.65.1]
- ESP-IDF Version: [4.4V]
- Python Version: [3.6.8][ESP python 3.8.7]

**Log:**
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/8213724/esp_idf_vsc_ext.log)

**Extension Configuration settings:**
```
{
    
    "editor.suggestSelection": "first",
    "vsintellicode.modify.editor.suggestSelection": "automaticallyOverrodeDefaultValue",
    "explorer.confirmDragAndDrop": false,
    "terminal.integrated.profiles.windows": {
        "PowerShell": {
        "source": "PowerShell",
        "overrideName": true,
        "args": ["-NoExit", "/c", "chcp 65001"],
        "icon": "terminal-powershell",
        "env": {
            "TEST_VAR": "value"
        }
        }
    },
    "terminal.integrated.defaultProfile.windows": "PowerShell",
    "hediet.vscode-drawio.local-storage": "eyIuZHJhd2lvLWNvbmZpZyI6IntcImxhbmd1YWdlXCI6XCJcIixcImN1c3RvbUZvbnRzXCI6W10sXCJsaWJyYXJpZXNcIjpcImdlbmVyYWxcIixcImN1c3RvbUxpYnJhcmllc1wiOltcIkwuc2NyYXRjaHBhZFwiXSxcInBsdWdpbnNcIjpbXSxcInJlY2VudENvbG9yc1wiOltdLFwiZm9ybWF0V2lkdGhcIjpcIjI0MFwiLFwiY3JlYXRlVGFyZ2V0XCI6ZmFsc2UsXCJwYWdlRm9ybWF0XCI6e1wieFwiOjAsXCJ5XCI6MCxcIndpZHRoXCI6ODI3LFwiaGVpZ2h0XCI6MTE2OX0sXCJzZWFyY2hcIjp0cnVlLFwic2hvd1N0YXJ0U2NyZWVuXCI6dHJ1ZSxcImdyaWRDb2xvclwiOlwiI2QwZDBkMFwiLFwiZGFya0dyaWRDb2xvclwiOlwiIzZlNmU2ZVwiLFwiYXV0b3NhdmVcIjp0cnVlLFwicmVzaXplSW1hZ2VzXCI6bnVsbCxcIm9wZW5Db3VudGVyXCI6MCxcInZlcnNpb25cIjoxOCxcInVuaXRcIjoxLFwiaXNSdWxlck9uXCI6ZmFsc2UsXCJ1aVwiOlwiXCJ9In0=",
    "workbench.colorTheme": "Visual Studio Light",
    "files.exclude": {
        "**/.classpath": true,
        "**/.project": true,
        "**/.settings": true,
        "**/.factorypath": true
    },
    "security.workspace.trust.untrustedFiles": "open",
    "idf.gitPath": "C:\\Users\\li1gu\\ESP\\Espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe",
    "idf.espIdfPathWin": "c:\\Users\\li1gu\\ESP\\esp-idf\\esp-idf",
    "idf.pythonBinPathWin": "C:\\Users\\li1gu\\ESP\\Espressif\\python_env\\idf4.4_py3.8_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "C:\\Users\\li1gu\\ESP\\Espressif",
    "idf.customExtraPaths": "C:\\Users\\li1gu\\ESP\\Espressif\\tools\\xtensa-esp32-elf\\esp-2021r2-patch2-8.4.0\\xtensa-esp32-elf\\bin;C:\\Users\\li1gu\\ESP\\Espressif\\tools\\xtensa-esp32s2-elf\\esp-2021r2-patch2-8.4.0\\xtensa-esp32s2-elf\\bin;C:\\Users\\li1gu\\ESP\\Espressif\\tools\\xtensa-esp32s3-elf\\esp-2021r2-patch2-8.4.0\\xtensa-esp32s3-elf\\bin;C:\\Users\\li1gu\\ESP\\Espressif\\tools\\riscv32-esp-elf\\esp-2021r2-patch2-8.4.0\\riscv32-esp-elf\\bin;C:\\Users\\li1gu\\ESP\\Espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Users\\li1gu\\ESP\\Espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Users\\li1gu\\ESP\\Espressif\\tools\\cmake\\3.20.3\\bin;C:\\Users\\li1gu\\ESP\\Espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20211220\\openocd-esp32\\bin;C:\\Users\\li1gu\\ESP\\Espressif\\tools\\ninja\\1.10.2;C:\\Users\\li1gu\\ESP\\Espressif\\tools\\idf-exe\\1.0.3;C:\\Users\\li1gu\\ESP\\Espressif\\tools\\ccache\\4.3\\ccache-4.3-windows-64;C:\\Users\\li1gu\\ESP\\Espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\li1gu\\\\ESP\\\\Espressif\\\\tools\\\\openocd-esp32\\\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "explorer.confirmDelete": false,
    "cmake.configureOnOpen": true,
    "editor.wordWrap": "on",
}
```

brianignacio5, 2022-03-09:
CMake generator is not used by this extension. This is a feature of the [Microsoft CMake Tools extension](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cmake-tools) and there is not such command `ESP-IDF: Select kit`

closed 2022-03-09 by brianignacio5

mo-han, 2024-08-12:
@brianignacio5 
So we just ignore the warning? Is there any method to auto suppress this warning when esp-idf extension is active?

brianignacio5, 2024-08-12:
We don't maintain the extension so we can't modify it. You can disable it when using our extension by making a user profile  or just disable for specific workspace.

The warning comes from that extension, so having it already configured will also remove the warning if you need that extension. To configure it you should refer to their repository.

citelao, 2025-01-07:
In my case, after disabling the `twxs.cmake` extension and reloading my VS Code window (and `Run idf.py reconfigure task`), I was able to build.

---

I was previously getting errors like:

```
CMake Error: Error: generator : Ninja
Does not match the generator used previously: Visual Studio 17 2022
```

I think the CMake extension was trying to auto-run some stuff & making my extension confused.