espressif/vscode-esp-idf-extension #667: [VS Code] Unable to build example project with latest v4.4 esp-idf

iamnagaky, 2022-03-12:
Unable to build example project with latest v4.4 esp-idf. 
In the configuration phase, the tools folder with all cmake, compilers were properly selected at
C:\User\***\.espressif\tools
But when trying to build using the Ctrl + E B build command, the xtensa compiler is not selected or not shown in the list.

brianignacio5, 2022-03-14:
Could you add the troubleshooting information as specified in the Issue template ?

brianignacio5, 2022-03-14:
**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration in your clipboard which you can copy here.

**Output**

In Visual Studio Code select menu `View` -> `Output` -> ESP-IDF, ESP-IDF Debug Adapter, Heap Trace, OpenOCD and SDK Configuration Editor.

> **NOTE:** Use `idf.openOcdDebugLevel` configuration setting to 3 or more to show debug logging in OpenOCD server output.

> **NOTE:** Use `logLevel` in your <project-directory>/.vscode/launch.json to 3 or more to show more debug adapter output.

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process output.

**Logs**
Please share the log file which can be obtained from:

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

brianignacio5, 2022-03-30:
Closing this issue for lack of response.

closed 2022-03-30 by brianignacio5