espressif/vscode-esp-idf-extension #669: [Bug Report]: Error Build process

Krypt0pr0xy, 2022-03-14:
Good day 

**Describe the bug**
I have a problem when I press Build Project in Visual Studio code. 
The Build process stops.

**To Reproduce**
I use a Windows 10 Computer. Version	10.0.19044 Build 19044
I use the blink example.

**Environment:**
- VSCode Version: 1.65.2
- ESP-IDF Version: v1.4.0 (from Install)
- Python Version: Python 3.9.10 (cmd python --version)

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration in your clipboard which you can copy here.

**Output**
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Measurement Computing\DAQ\;C:\Program Files (x86)\Razer\ChromaBroadcast\bin;C:\Program Files\Razer\ChromaBroadcast\bin;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Razer Chroma SDK\bin;C:\Program Files\Razer Chroma SDK\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\system32\config\systemprofile\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\PuTTY\;C:\Program Files\dotnet\;C:\Users\krypto\AppData\Local\Microsoft\WindowsApps;C:\Users\krypto\.dotnet\tools;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;%QUARTUS_ROOTDIR%\bin;C:\Program Files (x86)\dotnet\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Users\krypto\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Users\krypto\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\krypto\.dotnet\tools; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.65.2 
Visual Studio Code language de 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\krypto\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\krypto\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin;C:\Users\krypto\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\krypto\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\krypto\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch2-8.4.0\riscv32-esp-elf\bin;C:\Users\krypto\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\krypto\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\krypto\.espressif\tools\cmake\3.20.3\bin;C:\Users\krypto\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Users\krypto\.espressif\tools\ninja\1.10.2;C:\Users\krypto\.espressif\tools\idf-exe\1.0.3;C:\Users\krypto\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\krypto\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\krypto\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\krypto\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM11
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\krypto\.espressif
Git Path (idf.gitPath) C:\Users\krypto\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\krypto\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\krypto\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\krypto\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\Users\krypto\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch2-8.4.0\riscv32-esp-elf\bin: true
Access to C:\Users\krypto\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\krypto\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\krypto\.espressif\tools\cmake\3.20.3\bin: true
Access to C:\Users\krypto\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin: true
Access to C:\Users\krypto\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\krypto\.espressif\tools\idf-exe\1.0.3: true
Access to C:\Users\krypto\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:\Users\krypto\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.30.1.windows.1
ESP-IDF version 4.4
Python version 3.8.7
Python's pip version 20.3.3
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
[... 61 lines cut ...]
{"message":"WARNING: You are using pip version 20.3.3; however, version 22.0.4 is available.\r\nYou should consider upgrading via the 'C:\\Users\\krypto\\.espressif\\python_env\\idf4.4_py3.8_env\\Scripts\\python.exe -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 20.3.3; however, version 22.0.4 is available.\r\nYou should consider upgrading via the 'C:\\Users\\krypto\\.espressif\\python_env\\idf4.4_py3.8_env\\Scripts\\python.exe -m pip install --upgrade pip' command.\r\n\n\tat \\dist\\extension.js:1:7563\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2022-03-14T21:23:09.395Z"}
{"message":"WARNING: You are using pip version 20.3.3; however, version 22.0.4 is available.\r\nYou should consider upgrading via the 'C:\\Users\\krypto\\.espressif\\python_env\\idf4.4_py3.8_env\\Scripts\\python.exe -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 20.3.3; however, version 22.0.4 is available.\r\nYou should consider upgrading via the 'C:\\Users\\krypto\\.espressif\\python_env\\idf4.4_py3.8_env\\Scripts\\python.exe -m pip install --upgrade pip' command.\r\n\n\tat \\dist\\extension.js:1:7563\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2022-03-14T21:23:10.208Z"}
[... cut ...] 317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2022-03-14T23:04:03.869Z"}
{"user":true,"level":"warn","message":"Extension configuration report has been copied to clipboard with errors","timestamp":"2022-03-14T23:04:03.869Z"}

**Thanks for taking the time to look at this problem**

brianignacio5, 2022-03-15:
Seems the Python packages are not properly configured in the virtual environment.

You can try running the `Install ESP-IDF Python Packages` or `Configure ESP-IDF extension` command.

You could also share the build output: Click Menu View -> Terminal -> Choose the `ESP-IDF Build task` terminal from the list.

Krypt0pr0xy, 2022-03-15:
Thank you very much for the quick feedback.

I tried Install ESP-IDF Python Packages and Configure ESP-IDF extension

Output:
_____________________________________________________________________________________________________________________________________
ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.
WARNING: You are using pip version 20.3.3; however, version 22.0.4 is available.
You should consider upgrading via the 'C:\Users\krypto\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe -m pip install --upgrade pip' command.
Command failed: "C:\Users\krypto\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe" -m pip install --user virtualenv
ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.
WARNING: You are using pip version 20.3.3; however, version 22.0.4 is available.
You should consider upgrading via the 'C:\Users\krypto\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe -m pip install --upgrade pip' command.
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
Using existing C:\Users\krypto\.espressif\dist\idf-git-2.30.1-win64.zip
Installing C:\Users\krypto\.espressif\tools\idf-git\2.30.1 ...
Warning File C:\Users\krypto\.espressif\tools\idf-git\2.30.1\cmd\git.exe
                                        already exists and was not updated.
Warning File C:\Users\krypto\.espressif\tools\idf-git\2.30.1\dev\_._
                                        already exists and was not updated.
Warning File C:\Users\krypto\.espressif\tools\idf-git\2.30.1\etc\bash.bash_logout
                                        already exists and was not updated.
Warning File C:\Users\krypto\.espressif\tools\idf-git\2.30.1\etc\bash.bashrc
                                        already exists and was not updated.
Warning File C:\Users\krypto\.espressif\tools\idf-git\2.30.1\etc\fstab
                                        already exists and was not updated.
Warning File C:\Users\krypto\.espressif\tools\idf-git\2.30.1\etc\gitattributes
                                        already exists and was not updated.
Warning File C:\Users\krypto\.espressif\tools\idf-git\2.30.1\etc\msystem
                                        already exists and was not updated.
Warning File C:\Users\krypto\.espressif\tools\idf-git\2.30.1\etc\nsswitch.conf
                                        already exists and was not updated.
Warning File C:\Users\krypto\.espressif\tools\idf-git\2.30.1\etc\profile
                                        already exists and was not updated.
Extracted C:\Users\krypto\.espressif\tools\idf-git\2.30.1 ...
Using existing C:\Users\krypto\.espressif\tools\idf-python\3.8.7
C:\Users\krypto\esp\esp-idf already exists. Delete it or use another location. (ERROR_EXISTING_ESP_IDF)
C:\Users\krypto\esp\esp-idf already exists. Delete it or use another location. (ERROR_EXISTING_ESP_IDF)
Installing ESP-IDF extension python packages in C:\Users\krypto\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe ...

Requirement already satisfied: gcovr in c:\users\krypto\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from -r c:\Users\krypto\.vscode\extensions\espressif.esp-idf-extension-1.4.0\requirements.txt (line 1)) (5.0)
Requirement already satisfied: idf-component-manager in c:\users\krypto\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from -r c:\Users\krypto\.vscode\extensions\espressif.esp-idf-extension-1.4.0\requirements.txt (line 2)) (1.0.1)
Requirement already satisfied: websocket_client in c:\users\krypto\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from -r c:\Users\krypto\.vscode\extensions\espressif.esp-idf-extension-1.4.0\requirements.txt (line 3)) (1.3.1)
Requirement already satisfied: lxml in c:\users\krypto\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from gcovr->-r c:\Users\krypto\.vscode\extensions\espressif.esp-idf-extension-1.4.0\requirements.txt (line 1)) (4.8.0)
Requirement already satisfied: jinja2 in c:\users\krypto\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from gcovr->-r c:\Users\krypto\.vscode\extensions\espressif.esp-idf-extension-1.4.0\requirements.txt (line 1)) (3.0.3)
Requirement already satisfied: pygments in c:\users\krypto\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from gcovr->-r c:\Users\krypto\.vscode\extensions\espressif.esp-idf-extension-1.4.0\requirements.txt (line 1)) (2.11.2)
Requirement already satisfied: tqdm in c:\users\krypto\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager->-r c:\Users\krypto\.vscode\extensions\espressif.esp-idf-extension-1.4.0\requirements.txt (line 2)) (4.63.0)
Requirement already satisfied: requests-toolbelt in c:\users\krypto\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager->-r c:\Users\krypto\.vscode\extensions\espressif.esp-idf-extension-1.4.0\requirements.txt (line 2)) (0.9.1)
Requirement already satisfied: six in c:\users\krypto\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager->-r c:\Users\krypto\.vscode\extensions\espressif.esp-idf-extension-1.4.0\requirements.txt (line 2)) (1.16.0)
Requirement already satisfied: future in c:\users\krypto\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from idf-component-manager->-r c:\Users\krypto\.vscode\extensions\espressif.esp-idf-extension-1.4.0\requirements.txt (line 2)) (0.18.2)
[... cut ...]
Requirement already satisfied: colorama in c:\users\krypto\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages (from click->-r c:\Users\krypto\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 3)) (0.4.4)
WARNING: You are using pip version 20.3.3; however, version 22.0.4 is available.
You should consider upgrading via the 'C:\Users\krypto\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe -m pip install --upgrade pip' command.

_____________________________________________________________________________________________________________________________________

brianignacio5, 2022-03-15:
What is the doctor output after this ? Also what is the build output ?

Krypt0pr0xy, 2022-03-15:
> Executing task: ninja  <

ninja: error: loading 'build.ninja': Das System kann die angegebene Datei nicht finden.

Der Terminalprozess "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " wurde mit folgendem Exitcode beendet: 1.

brianignacio5, 2022-03-15:
Could you try deleting your build folder and try again building again ?

Krypt0pr0xy, 2022-03-15:
I have taken additional time to analyze the error in more detail. I have tried different settings and scenarios. It turned out that there was an error with the installation path. The path must not contain any capital letters, special characters or spaces. I made the installation on another hard disk and installed directly at the root directory. then everything worked so far : -) 

An old but known Bug which is hardly conceivable in this day and age. 

Thank you very much for your time.

closed 2022-03-15 by Krypt0pr0xy