espressif/vscode-esp-idf-extension #672: [Feature Request]: C/C++ friendly setting IntelliSense and code browsing

SinglWolf, 2022-03-16:
I'm determined to migrate my project from the PlatformIO environment to the ESP-IDF VS Code Extension environment.
But when viewing the source code of the text, I encountered an almost "hostile" behavior of C / C ++ IntelliSense and code review.
[image]

[image]
OK. Set `"C_Cpp.intelliSenseEngine": "Default"`
[image]
These red worms bloomed by the hundreds in almost every file of the working draft. I'm in a state of panic...

There are several ways.
The first is to ignore. The project is compiling, what else do you need?
I won't succeed after PlatformIO. There, a red squiggle indicates exactly that this is actually an ERROR. And where is the guarantee that the "fake" squiggle will not turn into a real threat to the project?
The second is to manually configure all(!) paths to header files for each one separately and enjoy life, recalling with horror the several days of editing paths to thousands of header files.
The third way is to make the C/C++ IntelliSense and code browsing friendly setup automatic. The PlatformIO team did the same.
Thanks for attention. I really hope that my emotional reasoning will be clear.

brianignacio5, 2022-03-16:
Did you tried following the [C/C++ Configuration documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) ?

SinglWolf, 2022-03-16:
> Did you tried following the [C/C++ Configuration documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) ?

PlatformIO has no requirement to study any documentation in this regard. Everything just works there. Is it possible to implement the same in ESP-IDF VS Code Extension? I'm sure it's possible.
I'm talking about the comfort of working in the ESP-IDF VS Code Extension, about user friendliness. After working with PlatformIO, working in ESP-IDF VS Code Extension causes a lot of negative emotions.
Or you can politely tell the user to study the manuals.
Where do you think the user will go? The question is rhetorical.

SinglWolf, 2022-04-01:
I'm still waiting for a response from the developers.

brianignacio5, 2022-04-08:
Setting `"compileCommands": "${workspaceFolder}/build/compile_commands.json"` in the `${workspaceFolder}/.vscode/c_cpp_properties.json` seems to provide a decent IntelliSense with the expected blackout. To resolve headers before build returns this file(compile_commands.json) is not a very trivial thing to do as you experience yourself. We hope to fix it soon.

I haven't found how PlatformIO solve headers for this but if someone know a bit more please point me into the code so i could use  it as reference and come with a solution too. As far as i understand they also depends in the Microsoft C/C++ extension too for this.

meltdown03, 2022-04-30:
I use this in the settings.json file:
`  "C_Cpp.intelliSenseEngine": "Default"`
and this in the c_cpp_properties.json file:
`"configurationProvider": "ms-vscode.cmake-tools"`
I get no squiggles and no extra suggestions. You do need the CMake Tools extension installed though.

mitchjs, 2022-06-18:
@meltdown03 ,

where exactly do i put "configurationProvider": "ms-vscode.cmake-tools" in the c_cpp_properies.jaon file
do i leave all that is normally there?
could you show example file?

thanks

meltdown03, 2022-06-18:
Yes, the `c_cpp_properties.json` file:

``` json
{
  "configurations": [
    {
      "name": "ESP-IDF",
      "compilerPath": "/home/neal/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc",
      "cStandard": "c11",
      "cppStandard": "c++17",
      "includePath": [
        "${config:idf.espIdfPath}/components/**",
        "${config:idf.espIdfPathWin}/components/**",
        "${config:idf.espAdfPath}/components/**",
        "${config:idf.espAdfPathWin}/components/**",
        "${workspaceFolder}/**"
      ],
      "browse": {
        "path": [
          "${config:idf.espIdfPath}/components",
          "${config:idf.espIdfPathWin}/components",
          "${config:idf.espAdfPath}/components/**",
          "${config:idf.espAdfPathWin}/components/**",
          "${workspaceFolder}"
        ],
        "limitSymbolsToIncludedHeaders": false
      },
      "configurationProvider": "ms-vscode.cmake-tools",
      "compileCommands": "${workspaceFolder}/build/compile_commands.json"
    }
  ],
  "version": 4
}
```
It seems that the compileCommands is required to show/hide the `Defines` that are set/unset.
Using the cmake-tools seems to help show only relevent symbols. Using the 2 together has worked best for me. And also setting the `"C_Cpp.intelliSenseEngine": "Default",` in the `settings.json` file may also limit the unwanted symbols/files from showing up.

mitchjs, 2022-06-18:
ok, i put that in, and enabled cmake... 
one thing... "compileCommands": "${workspaceFolder}/build/compile_commands.json"
says it cant find the file... i did build, its there... but its not happy got squigglies

do i have to do any settings for cmake or cpptools?

meltdown03, 2022-06-18:
If you have Windows, change the slashes (/) to (\\)

mitchjs, 2022-06-18:
i thought about that, but then i look at the paths in the include and browse, and they /
ill will try to make it happy

HiFiPhile, 2022-07-27:
While generating configuration folder, please add a configuration using `compileCommands`

For example I can switch to `ESP-IDF2` once project is compiled.

```
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "${config:idf.toolsPath}\\tools\\riscv32-esp-elf\\esp-2021r2-patch3-8.4.0\\riscv32-esp-elf\\bin\\riscv32-esp-elf-gcc.exe",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${workspaceFolder}/build/config",
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            },
            "defines": ["CONFIG_IDF_TARGET_ESP32C3",
                        "ESP_PLATFORM"]
        },
        {
            "name": "ESP-IDF2",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
}
```

gonzabrusco, 2023-04-24:
This feature would be nice (to work out of the box).

For now, I added compileCommands and changed "C_Cpp.intelliSenseEngine" to "default" and worked great.

gonzabrusco, 2023-05-02:
@italocjs try "default" (not Default). Also, you have to compile for this to work!

italocjs, 2023-05-02:
> @italocjs try "default" (not Default). Also, you have to compile for this to work!

Nailed it, just changed to "default" and it worked!  thanks!

reinderien, 2023-05-07:
For recent (5.0.1) ESP-IDF and vscode, I have a slightly different set of options that I find works well, or at least well enough:

**settings.json**

```json
    "C_Cpp.codeAnalysis.clangTidy.enabled": false,
    "C_Cpp.default.intelliSenseMode": "windows-gcc-arm",

    "cmake.cmakePath": "/ProgramData/esp-idf-tools/tools/cmake/3.24.0/bin/cmake.exe",
    // ...
```

**c_cpp_properties.json**
```json
{
    "configurations": [
        {
            "name": "riscv32",
            "includePath": [
                "${workspaceFolder}/**",
                "/ProgramData/esp-idf/components/**",
                "/ProgramData/esp-rainmaker/components/**",
                "/ProgramData/esp-rainmaker/components/esp_rainmaker/include/**"
            ],
            "defines": [
                "_GNU_SOURCE=1",
                "_POSIX_READER_WRITER_LOCKS=1",
                "configENABLE_FREERTOS_DEBUG_OCDAWARE=1",
                "ESP_PLATFORM=1",
                "IDF_VER=v5.0.1",
                "MBEDTLS_CONFIG_FILE=mbedtls/esp_config.h",
                "UNITY_INCLUDE_CONFIG_H=1"
            ],
            "compilerPath": "/ProgramData/esp-idf-tools/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe",
            "cStandard": "gnu17",
            "intelliSenseMode": "windows-gcc-arm",
            "compileCommands": "${workspaceFolder}/build/compile_commands.json",
            "configurationProvider": "ms-vscode.cmake-tools"
        }
    ],
    "version": 4
}
```

**extensions.json**
```json
{
    "recommendations": [
        "eamodio.gitlens",
        "ms-vscode.cpptools",
        "ms-vscode.cmake-tools",
        "twxs.cmake",
        "surajbarkale.ninja",
        "espressif.esp-idf-extension",
    ]
}
```

There's still some breakage - I'm fighting tooth and nail to try and tell my custom cmake kit where Ninja lives - but for now, the above is good enough. Note that if clang-tidy is enabled, it issues false positives for `stddef.h`.

emaayan, 2023-07-20:
> 

it would be great if you could add some comments as to what those settings do and what are the defines for..

patrickkeenan, 2023-10-07:
I had a bunch of errors and found out I needed to add "compileCommands": "${workspaceFolder}/build/compile_commands.json" to my ccp_properties file. It would be great if the esp extension did this by default.

PR #1103 (2024-01-25): remove ccpp extension dependency
## Description

Remove Microsoft C/C++ extension from vscode extension dependencies. This will simplify release flow for vsix for open-vsx   and vscode marketplace release. Allow user to define their own language extensions since we don't provide support for them.

Fix #672 

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Test `yarn package` in CLI to try to build this extension. Follow Build from source or CONTRIBUTING.md tutorial. Observe Microsoft C/C++ extension is not added automatically [... cut]
closed 2024-01-25 by brianignacio5