espressif/vscode-esp-idf-extension #674: Stuck on prerequisites for Configuring Extension

markandpathak, 2022-03-17:
The issue is:
I use vs code on Ubuntu 20.04 with ESP-IDF extension for development. I have already installed ESP-IDF v4.3 and now want to upgrade to v4.4.  The issue is in a new directory when I run command **ESP-IDF: Configure ESP-IDF extension**, it gets stuck on install prerequisites and run again screen for the ESP-IDF. 

On following the instructions given [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html) the `apt` shows all packages already installed. On running the **ESP-IDF: Configure ESP-IDF extension** it again shows screen to install prerequisites and can not move forward. Nothing works, I tried uninstalling the extension, deleting the existing ESP-IDF folders, deleting .espressif folder in home and even removing and installing vs code again.

The extension works in fresh install but not when we need to install additional version or upgrade existing version.

How to overcome this issue?

brianignacio5, 2022-03-17:
Could you share your troubleshooting information as shown in [here](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting)?

brianignacio5, 2022-03-30:
Closing this issue for lack of response.

closed 2022-03-30 by brianignacio5