espressif/vscode-esp-idf-extension #678: [Bug Report]: PROBLEMS tab empty when build succeeds

littleboot, 2022-03-23:
**Describe the bug**
A succesfull build clears all warnings from the "PROBLEMS" window, This makes finding and fixing the errors difficult when no error is present in the code.

**To Reproduce**
Open a project for example the "blinky" example.
Declare a function prototype and nothing else, this will introduce the [-Wunused-function] warning.
add `static void unusedFunc();` for example

Build the project: 
`>ESP-IDF: Build your project`

Open build output (should open automatically but someone already requested this feature) to validate the warning is present.
[image]

Check "Problems" tab, observe no warnings are listed.
[image]

Introduce an error that stops the build, (I added "error" to end of the file, this is invalid c).
Now the warning is listed together with the error inside the "PROBLEMS" tab.
[image]

**Expected behavior**
Always show the warnings even when the build succeeds.

**Environment:**
- OS: Windows 10 64-bit
- VSCode Version: 1.65.2
- ESP-IDF Version: 1.4.0

brianignacio5, 2022-04-08:
I think this happens because the `IDF Size task` is executed afterwards clearing out the warning from the previous task.

Will check if there is a way to persist the task problems

littleboot, 2022-04-12:
> I think this happens because the `IDF Size task` is executed afterwards clearing out the warning from the previous task.
> 
> Will check if there is a way to persist the task problems

Thank you for looking into this. I really like the VScode IDF extension sadly this bug made me switch to Eclipse for compiling and debugging, I hope I can switch back soon.

closed 2022-04-13 by brianignacio5

MarkusvonStaden, 2022-07-06:
I'm sorry to dig out this old topic, but I still have this problem, even though I am on the newest version of the extension and the issue is marked as completed. Am I missing something?

brianignacio5, 2022-07-06:
Are you using the vsix installer from master or the marketplace version ? We haven't released in marketplace these changes.

MarkusvonStaden, 2022-07-07:
I installed 1.4.0 from the Github releases Tab.

brianignacio5, 2022-07-11:
Yes the fix is not part of the 1.4.0 release. Will release a new version soon :'(

electronick-co, 2023-10-20:
Hello!
I'm having the same issue on version 1.6.5 install from the marketplace (that seems to be the same as the latest release available at the moment).

I'm using Ubuntu 22.04. and nothing shows in the problems except when i actually have an error (then all the warnings appear)

gonzabrusco, 2024-01-18:
Same issue. After a successful build, the PROBLEMS tab gets cleared. It's impossible to check for warnings.

littleboot, 2024-01-18:
Found a solution, have a look at the pull requests:
Someone fixed the issues so you can compile with idd v5
https://github.com/hiruna/lvgl_esp32_drivers/tree/develop/lvgl_8.3.7_idf_5.2

brianignacio5, 2024-01-19:
Have you tried disabling the idf size task after build task @gonzabrusco ?

gonzabrusco, 2024-01-19:
> Have you tried disabling the idf size task after build task @gonzabrusco ?

How do I do that?

brianignacio5, 2024-01-19:
In your project `.vscode/settings.json` add `idf.enableSizeTaskAfterBuildTask: false`

gonzabrusco, 2024-01-26:
Is it possible to work on a solution that runs idf size task but also keeps the compilation workings? Not sure why we have to choose between both options when they are displayed on different tabs (size info in TERMINAL and warnings in PROBLEMS).

mmrein, 2024-11-27:
I would suggest disabling the Size Task by default until/if this gets resolved. Showing build output with it's warnings is basic and expected functionality.

PR #692 (2022-04-13): clean terminal output before build task start
Fix #665 
Fix #678

* Clean terminal output before build task starts.
* Disable CCache if not enabled in the extension for setting IDF_TARGET task.
* idf_versions.txt URL refactor to config to unifiy hard coded strings url.
* Move presentation options as addTask argument, idf-size to use a dedicated terminal