espressif/vscode-esp-idf-extension #680: [Feature Request]:  build/flash/monitor/debug with espressif extension in the docker image container

xiaolongba, 2022-03-23:
Hello Espressif Experts,

as we all know, vscode + espressif extension + windows 10 can implement building/flashing/monitoring/debugging. so is it possible to do the same thing with the vscode + espressif extension + docker image container?

I tried to configure the espressif extension in the docker image container, but it seems that it cannot work. I still need to build the project with commands manually, so my purpose is to build/flash/monitor/debug with espressif extension directly, like windows 10.  is it possible?

thanks in advance.

xiaolongba, 2022-03-30:
Hello Espressif experts,

Any place can share with you my experience. the story is a bit long. I resolved this issue. if yes, pls told me, I will write a tutorial about it in the dedicated place you point out

brianignacio5, 2022-04-02:
Thanks @xiaolongba You could make a PR to this repository with a tutorial in `docs/tutorial/using-docker-container.md` for example.

xiaolongba, 2022-04-02:
@brianignacio5 
OK, I will. I am preparing the materials, I looked into the doc you mentioned above, it is a bit simple, what I write is to implement all features based on remote container extension esp-idf extension, such as flash, debug, monitor, etc.

as I know, the esp32 docker image ran into the issue which is to access to the serial port and it leads to impossible to flash or monitor, but now it seems that I found one solution to fix it.

pls help me double-check and optimize it if there are some unreasonable places. if it is possible

xiaolongba, 2022-04-11:
Hello @brianignacio5 Bro,

pls close this ticket, I made a PR #693  already

closed 2022-04-13 by brianignacio5

parksj10, 2023-01-30:
@brianignacio5 or @brianignacio5, is this possible on macOS? I'm trying to get the debugging device to pass through by adding `"--device=/dev/cu.usbserial-1101:/dev/cu.usbserial-1101",` (or the `tty` version for that matter) to my `runArgs` but the ESP Prog doesn't doesn't show up in the dev container

brianignacio5, 2023-01-30:
Take a look at the [docker container tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/using-docker-container.md) @parksj10 it should give you some additional indication.

parksj10, 2023-01-30:
Thanks! That's what I've been looking at (and why I added the comment to this issue), it however only shows information a Windows WSL install, it doesn't have any info about getting it up and running on a linux/unix system. Do I need to use `usbipd` on a *nix system as well?

xiaolongba, 2023-01-30:
Hello @parksj10 

sorry, I have no idea on a *nix system as the condition limitation. I was always using the win OS😂

brianignacio5, 2023-01-30:
The serial ports are obtained using this script:

```py
import serial.tools.list_ports
print([comport.device for comport in serial.tools.list_ports.comports()])
```
if you try running this in the container's terminal, what do you get ?

attaching the device to the docker arguments `--device` should work but maybe there is an issue with the dev container mechanism or docker arguments itself. When you run the container, there should be a log of the docker run args which can give more insight.

parksj10, 2023-02-04:
@brianignacio5, I get an empty array

```python
>>> import serial.tools.list_ports
>>> print([comport.device for comport in serial.tools.list_ports.comports()])
[]
```

brianignacio5, 2023-02-06:
seems that Python can't find the serial port itself. If you are sure that the device is accessible from the docker container. You can just manually define the port in your settings.json `idf.port` and try to flash with that.

PR #693 (2022-04-13): add a tutorial for using docker container (VSC-885)
this tutorial showed you how to configure/use the tools related. all features have been implemented when using ESP-IDF extension, including flashing, monitoring, debugging etc.

Fix #679 
Fix #680