espressif/vscode-esp-idf-extension #681: After using "Configure ESP-IDF Extension" with "release/v4.4", there are still many dirty git files

chipweinberger, 2022-03-24:
VSC Extension 1.4.0

Steps to repro:
1. delete esp-idf folder
2. run Configure ESP-IDF Extension with release/v4.4 (release branch)
3. cd into XXXXX/esp-idf
4. git status

**Expectation**, the branch should be checked out in esp-idf folder. No dirty files. 

How are we supposed to verify what commit and branch we are on if it checks out saying "no branch". Very confusing. 

[image]

[image]

chipweinberger, 2022-03-24:
Uh Oh! I just opened a random file to check if it really is checking out release/v4.4 and it is not even the right code!

components/heap/heap_caps.c is missing the latest changes!

Checking out a release branch seems to be broken!

chipweinberger, 2022-04-11:
any comments on this? I haven't tried again since

brianignacio5, 2022-04-11:
Sorry for late reply! 

This should have been fixed with #622 but will look at it and fix it soon!

PR #696 (2022-04-13): fix release branch validate response headers
Fix #681 

* Fix release branch option to use git clone instead of downloading a release zip file.
* Add validation in downloadManager.ts response headers `location` and `content-length`.
closed 2022-04-13 by brianignacio5