espressif/vscode-esp-idf-extension #682: idf.notificationSilentMode doesn't work

remyhx, 2022-03-25:
Tried to get rid of the annoying notifications (not errors) by checking idf.notificationSilentMode. It doesn't work, also not after restarting. 

Using the latest 1.4.0 version and the master branch of esp-idf (just updated by git pull). Windows 10.

Dokter report:

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Microchip\xc16\v1.70\bin;C:\Program Files\Microchip\xc32\v3.00\bin;C:\Program Files\Microchip\xc8\v2.31\bin;C:\Program Files\PuTTY\;C:\Users\localadmin\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\Scripts;C:\Program Files\Git\cmd;C:\Users\localadmin\.cargo\bin;C:\Users\localadmin\AppData\Local\Microsoft\WindowsApps;C:\Espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-pc-windows-msvc/bin/;C:\ESPRESSIF/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-pc-windows-msvc/bin/;C:\esp\tools/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-pc-windows-msvc/bin/;C:\Espressif\tools/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-pc-windows-msvc/bin/;C:\Users\localadmin\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.65.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/esp-idf/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2021r3-patch2-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\tools\xtensa-esp32s3-elf\esp-2021r3-patch2-8.4.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\tools\riscv32-esp-elf\esp-2021r3-patch3-8.4.0\riscv32-esp-elf\bin;C:\Espressif\tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Espressif\tools\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Espressif\tools\tools\cmake\3.20.3\bin;C:\Espressif\tools\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Espressif\tools\tools\ninja\1.10.2;C:\Espressif\tools\tools\idf-exe\1.0.3;C:\Espressif\tools\tools\ccache\4.3\ccache-4.3-windows-64;C:\Espressif\tools\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Espressif\\tools\\tools\\openocd-esp32\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:/Espressif/tools/python_env/idf5.0_py3.8_env/Scripts/python.exe
Serial port (idf.port) COM3
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif\tools
Git Path (idf.gitPath) C:/Espressif/tools/tools/idf-git/2.34.2/cmd/git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2021r3-patch2-8.4.0\xtensa-esp32s2-elf\bin: false
Access to C:\Espressif\tools\tools\xtensa-esp32s3-elf\esp-2021r3-patch2-8.4.0\xtensa-esp32s3-elf\bin: false
Access to C:\Espressif\tools\tools\riscv32-esp-elf\esp-2021r3-patch3-8.4.0\riscv32-esp-elf\bin: false
Access to C:\Espressif\tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Espressif\tools\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Espressif\tools\tools\cmake\3.20.3\bin: true
Access to C:\Espressif\tools\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin: true
Access to C:\Espressif\tools\tools\ninja\1.10.2: true
Access to C:\Espressif\tools\tools\idf-exe\1.0.3: true
Access to C:\Espressif\tools\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:\Espressif\tools\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.34.1.windows.1
ESP-IDF version 5.0
Python version 3.8.7
Python's pip version 20.3.3
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitstring version: 3.1.9
certifi version: 2021.10.8
cffi version: 1.15.0
charset-normalizer version: 2.0.11
click version: 8.0.4
colorama version: 0.4.4
construct version: 2.10.68
contextlib2 version: 21.6.0
cryptography version: 36.0.1
ecdsa version: 0.17.0
future version: 0.18.2
gcovr version: 5.0
idf-component-manager version: 1.0.1
idna version: 3.3
[... 72 lines cut ...]
    "version": 4
} 
--------------------------------------------------------------------------------------------------------------------------------------------

settings.json:
{
    "C_Cpp.intelliSenseEngine": "Default",
    "idf.adapterTargetName": "esp32",
    "idf.customExtraPaths": "C:\\Espressif\\tools\\tools\\xtensa-esp32-elf\\esp-2021r2-patch3-8.4.0\\xtensa-esp32-elf\\bin;C:\\Espressif\\tools\\tools\\xtensa-esp32s2-elf\\esp-2021r3-patch2-8.4.0\\xtensa-esp32s2-elf\\bin;C:\\Espressif\\tools\\tools\\xtensa-esp32s3-elf\\esp-2021r3-patch2-8.4.0\\xtensa-esp32s3-elf\\bin;C:\\Espressif\\tools\\tools\\riscv32-esp-elf\\esp-2021r3-patch3-8.4.0\\riscv32-esp-elf\\bin;C:\\Espressif\\tools\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Espressif\\tools\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Espressif\\tools\\tools\\cmake\\3.20.3\\bin;C:\\Espressif\\tools\\tools\\openocd-esp32\\v0.11.0-esp32-20211220\\openocd-esp32\\bin;C:\\Espressif\\tools\\tools\\ninja\\1.10.2;C:\\Espressif\\tools\\tools\\idf-exe\\1.0.3;C:\\Espressif\\tools\\tools\\ccache\\4.3\\ccache-4.3-windows-64;C:\\Espressif\\tools\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Espressif\\\\tools\\\\tools\\\\openocd-esp32\\\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.espIdfPathWin": "C:/Espressif/esp-idf/",
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "target/esp32.cfg"
    ],
    "idf.portWin": "COM3",
    "idf.pythonBinPathWin": "C:/Espressif/tools/python_env/idf5.0_py3.8_env/Scripts/python.exe",
    "idf.toolsPathWin": "C:\\Espressif\\tools",
    "files.associations": {
        "sensirion_gas_index_algorithm.h": "c"
    },
    "idf.flashType": "UART"
}

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`

brianignacio5, 2022-04-02:
Should be fixed with #662 

Can you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/201258128.zip) ?

brianignacio5, 2022-04-08:
In your settings.json i don't see you set `idf.notificationSilentMode: true`to disable those notifications.

Closing this issue for this and lack of response.

closed 2022-04-08 by brianignacio5

remyhx, 2022-04-08:
Well. Maybe that is the problem…  This is the updated version via your six file, as i mentioned before it didn’t work now. I was waiting for your response:))

[Image.jpeg]

Outlook voor iOS<https://aka.ms/o0ukef> downloaden

remyhx, 2022-10-11:
It doesn’t work yet for me. While building there is still the notice.

brianignacio5, 2022-10-26:
Could you share the required [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) and describe the current behavior ?

remyhx, 2022-10-26:
Strange: I replied (by email reply) the 3rd of April  and 'my reply' says I replied 15 days ago haha. Stuck somewhere I guess. It still doesn't work on my mac and windows computer, but I just learned to live with it ...

I usually put my output/terminals at the right, so it blocks the recent updates from the chip. But it's ok.