espressif/vscode-esp-idf-extension #687: [Bug Report]: In terminal idf.py menuconfig can't move up and down

QQ2439423861, 2022-03-31:
**Describe the bug**
In ESP-IDF terminal, `idf.py menuconfig` can't move up and down

**To Reproduce**
1. click the button `ESP-IDF: Open ESP-IDF Terminal`
2. `cd xxx` and then enter the project directory
3. input `idf.py menuconfig`, popup a menu, can't move up and down, but other command show in the bottom(such as `enter` to select the item) execute properly.

When I click the extension button `menuconfig`, it also work properly.

**Expected behavior**
use the key ↓ ↑ to move up or down

**Environment (please complete the following information):**

- OS Name & Version: [Win10]
- VSCode Version: [1.66.0]
- ESP-IDF Version: [4.4]
- Python Version: [3.x]

QQ2439423861, 2022-03-31:
[image]

brianignacio5, 2022-04-02:
This is not an extension issue but a vscode issue.

Either way, the scrolling should work. Make sure the ESP-IDF Terminal is selected. It seems in the picture that the pop up made focus out of the terminal which means up and down keys to now work. The ESP-IDF Terminal should be highlighted.

brianignacio5, 2022-04-13:
Closing for lack of response.

hazem3443, 2023-07-01:
I see this issue too and scrolling is not working for me and I am sure the terminal area is selected in VScode

hazem3443, 2023-07-01:
after trying different buttons I found that arrow buttons are mapped as follows

&uarr; - j 
&darr; - k
&larr; - h
&rarr; - L

AnibalHeredia, 2024-05-13:
> Después de probar diferentes botones, descubrirá que los botones de flecha están asignados de la siguiente manera.
> 
> ↑ - K ↓ - J ← - h → - L

eres un crack gracias