espressif/vscode-esp-idf-extension #688: Menuconfig Crash

baldhead69, 2022-03-31:
Hi, 

I'm trying to install a new version of esp-idf and all tools from scratch(i deleted all old installation). 
Every time i do this there are some bugs (always !!!!)

The menuconfig dont open.
In annex follow a picture.

esp-idf version: v5.0-dev-2326-g36b83fcb98
OS:  Windows  10 Pro
        Version	 21H2
        SO Compilation 19044.1586

[image]

baldhead69, 2022-04-01:
Some suggestion ?

baldhead69, 2022-04-01:
it looks like the path is wrong. 

My computer path: C:\.espressif\python_env\idf5.0_py3.8_env\Scripts

when i installed the tools i used "advanced mode" and specified another path to save the files.

it looks like the installation didn't adjust to the new path.

how can i change this path ?

brianignacio5, 2022-04-02:
Settings can be saved in different locations as specified in [the working with multiple projects documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/MULTI_PROJECTS.md). Make sure that you project's `.vscode/settings.json` is not overriding settings in User or Workspace settings.

The `ESP-IDF: Select where to save configuration settings` or the `idf.saveScope` setting specifies where the Setup Wizard settings are saved.

baldhead69, 2022-04-02:
@brianignacio5,

I installed the "ESP-IDF VS Code Extension" in another computer and same problem.
The problem is "ESP-IDF VS Code Extension" installation. 

Before putin git in environment variable:
[esp_idf_vsc_ext_before_put_git_in_environment_variable.log](https://github.com/espressif/vscode-esp-idf-extension/files/8403298/esp_idf_vsc_ext_before_put_git_in_environment_variable.log)

After puting git in environment variable:
[image]

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/8403312/esp_idf_vsc_ext.log)

baldhead69, 2022-04-04:
Some suggestion ? 

i'm stuck on it.

baldhead69, 2022-04-08:
I had to change the windows registry and still nothing.

At least i managed to change from "IDF_TOOLS_PATH=C:\ESP-IDF\TOOLS" to "IDF_TOOLS_PATH=c:\.espressif".

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/8447648/esp_idf_vsc_ext.log)

baldhead69, 2022-04-09:
Hi folks,

For those who have the same problem and are still waiting for an answer from espressif.

For windows users only(i installed in win10 in one computer and in win11 in another computer):

Freh installation:

Delete the "c:\\.espressif" folder.
Delete the "c:\\.esp-idf" folder.
Delete the visual studio code and all folder like that:
https://debug.to/1074/how-to-uninstall-vs-code-completely

In environment variables:

Set in "user variables": 
IDF_PATH c:\esp-idf
IDF_TOOLS_PATH c:\.espressif

Set in "system variables":
in "Path" put: 
C:\.espressif\tools\idf-git\2.30.1\cmd
C:\.espressif\python_env\idf5.0_py3.8_env\Scripts 

Install "visual studio code"
Install "ESP-IDF VS Code Extension" inside visual studio code
Run"ESP-IDF VS Code Extension" and install:
Install esp idf in c:\esp-idf
Install tools in c:\.espressif 

I think that's it folks.

Thanks for the help espressif team :(

baldhead69, 2022-04-12:
The program compile, but now...............

Out of nowhere the monitor stopped working:
```c
{"message":"non zero exit code 2\r\n\r\nERRO: padr�o inv�lido especificado em \"path:pattern\".\r\n","stack":"Error: non zero exit code 2\r\n\r\nERRO: padr�o inv�lido especificado em \"path:pattern\".\r\n\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:1:4499)\n\tat ChildProcess.emit (node:events:390:28)\n\tat Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)","level":"error","timestamp":"2022-04-12T20:35:38.381Z"}
{"message":"Cannot access filePath: c:\\.espressif\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe","stack":"Error: non zero exit code 2\r\n\r\nERRO: padr�o inv�lido especificado em \"path:pattern\".\r\n\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:1:4499)\n\tat ChildProcess.emit (node:events:390:28)\n\tat Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)","level":"error","timestamp":"2022-04-12T20:35:38.382Z"}
```

[image]

Out of nowhere the menuconfig stopped working too:
```c
{"message":"The following Python requirements are not satisfied:\r\r\njinja2<3.1\r\r\nTo install the missing packages, please run \"install.bat\"\r\r\nDiagnostic information:\r\r\n    IDF_PYTHON_ENV_PATH: c:\\.espressif\\python_env\\idf5.0_py3.8_env\r\r\n    Python interpreter used: c:\\.espressif\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe\r\r\nSkipping the download of C:\\Users\\02373893983\\.espressif\\espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.\r\r\nConstraint file: C:\\Users\\02373893983\\.espressif\\espidf.constraints.v5.0.txt\r\r\nRequirement files:\r\r\n - c:\\esp-idf\\tools\\requirements\\requirements.core.txt\r\r\nPython being checked: c:\\.espressif\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe\r\r\n\r\nESP-IDF v5.0-dev-2393-gfaed2a44aa\r\n","stack":"Error: The following Python requirements are not satisfied:\r\r\njinja2<3.1\r\r\nTo install the missing packages, please run \"install.bat\"\r\r\nDiagnostic information:\r\r\n    IDF_PYTHON_ENV_PATH: c:\\.espressif\\python_env\\idf5.0_py3.8_env\r\r\n    Python interpreter used: c:\\.espressif\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe\r\r\nSkipping the download of C:\\Users\\02373893983\\.espressif\\espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.\r\r\nConstraint file: C:\\Users\\02373893983\\.espressif\\espidf.constraints.v5.0.txt\r\r\nRequirement files:\r\r\n - c:\\esp-idf\\tools\\requirements\\requirements.core.txt\r\r\nPython being checked: c:\\.espressif\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe\r\r\n\r\nESP-IDF v5.0-dev-2393-gfaed2a44aa\r\n\n\tat m.printError (\\dist\\extension.js:1:30357)\n\tat Socket.<anonymous> (\\dist\\extension.js:1:29786)\n\tat Socket.emit (node:events:390:28)\n\tat addChunk (node:internal/streams/readable:315:12)\n\tat readableAddChunk (node:internal/streams/readable:289:9)\n\tat Socket.Readable.push (node:internal/streams/readable:228:10)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:199:23)","level":"error","timestamp":"2022-04-12T20:44:47.394Z"}
{"message":"SDK Configuration editor confserver process exited with code: 1","stack":"Error: SDK Configuration editor confserver process exited with code: 1\n\tat m.printError (\\dist\\extension.js:1:30357)\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:1:29963)\n\tat ChildProcess.emit (node:events:390:28)\n\tat Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)","level":"error","timestamp":"2022-04-12T20:44:47.404Z"}
```

I already run c:\esp-idf>install.bat and after this export.bat but the problem persists.

This is annoying.

baldhead69, 2022-04-12:
[espidf.constraints.v5.0.txt](https://github.com/espressif/vscode-esp-idf-extension/files/8477024/espidf.constraints.v5.0.txt)

baldhead69, 2022-04-15:
I didn't do anything to make this happen.

It looks like something is updating in the background.

Maybe windows or visual studio code or esp-idf. I dont know.

brianignacio5, 2022-04-15:
It might be an issue with your virtualenv python packages . Could you try deleting the virtual env folder (C:\.espressif\python_env\idf5.0_py3.8_env) and run the setup wizard again  ?

baldhead69, 2022-04-15:
Hi @brianignacio5,

I'm just a mere user of the tool, i don't understand anything about these phyton environments and tools.
This visual studio code application should be installed and ready to use, right ?!

I did a new install.

Delete the "c:\.espressif" folder.
Delete the "c:\esp-idf" folder.
Delete the visual studio code and all folders like that:
https://debug.to/1074/how-to-uninstall-vs-code-completely

Install "visual studio code"
Install "ESP-IDF VS Code Extension" inside visual studio code
Run"ESP-IDF VS Code Extension" and install:
Install esp idf in c:\esp-idf
Install tools in c:\.espressif

In environment variables:

Set in "user variables":
IDF_PATH c:\esp-idf
IDF_TOOLS_PATH c:\.espressif

Set in "system variables":
in "Path" put:
c:\.espressif\tools\idf-git\2.30.1\cmd
c:\.espressif\python_env\idf5.0_py3.8_env\Scripts

Starting menuconfig
ESP-IDF: Menuconfig: Configuring server
```c
{"message":"The following Python requirements are not satisfied:\r\r\nclick>=7.0,<8.1\r\r\r\njinja2<3.1\r\r\nTo install the missing packages, please run \"install.bat\"\r\r\nDiagnostic information:\r\r\n    IDF_PYTHON_ENV_PATH: c:\\.espressif\\python_env\\idf5.0_py3.8_env\r\r\n    Python interpreter used: c:\\.espressif\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe\r\r\nSkipping the download of c:\\.espressif\\espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.\r\r\nConstraint file: c:\\.espressif\\espidf.constraints.v5.0.txt\r\r\nRequirement files:\r\r\n - C:\\esp-idf\\tools\\requirements\\requirements.core.txt\r\r\nPython being checked: c:\\.espressif\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe\r\r\n\r\nESP-IDF v5.0-dev-2483-g3aeb80acb6\r\n","stack":"Error: The following Python requirements are not satisfied:\r\r\nclick>=7.0,<8.1\r\r\r\njinja2<3.1\r\r\nTo install the missing packages, please run \"install.bat\"\r\r\nDiagnostic information:\r\r\n    IDF_PYTHON_ENV_PATH: c:\\.espressif\\python_env\\idf5.0_py3.8_env\r\r\n    Python interpreter used: c:\\.espressif\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe\r\r\nSkipping the download of c:\\.espressif\\espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.\r\r\nConstraint file: c:\\.espressif\\espidf.constraints.v5.0.txt\r\r\nRequirement files:\r\r\n - C:\\esp-idf\\tools\\requirements\\requirements.core.txt\r\r\nPython being checked: c:\\.espressif\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe\r\r\n\r\nESP-IDF v5.0-dev-2483-g3aeb80acb6\r\n\n\tat m.printError (\\dist\\extension.js:1:30357)\n\tat Socket.<anonymous> (\\dist\\extension.js:1:29786)\n\tat Socket.emit (node:events:390:28)\n\tat addChunk (node:internal/streams/readable:315:12)\n\tat readableAddChunk (node:internal/streams/readable:289:9)\n\tat Socket.Readable.push (node:internal/streams/readable:228:10)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:199:23)","level":"error","timestamp":"2022-04-15T16:30:41.573Z"}
{"message":"SDK Configuration editor confserver process exited with code: 1","stack":"Error: SDK Configuration editor confserver process exited with code: 1\n\tat m.printError (\\dist\\extension.js:1:30357)\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:1:29963)\n\tat ChildProcess.emit (node:events:390:28)\n\tat Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)","level":"error","timestamp":"2022-04-15T16:30:41.583Z"}
```

Starting ESP-IDF Monitor
```c
{"message":"non zero exit code 2\r\n\r\nERRO: padr�o inv�lido especificado em \"path:pattern\".\r\n","stack":"Error: non zero exit code 2\r\n\r\nERRO: padr�o inv�lido especificado em \"path:pattern\".\r\n\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:1:4499)\n\tat ChildProcess.emit (node:events:390:28)\n\tat Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)","level":"error","timestamp":"2022-04-15T16:39:10.324Z"}
{"message":"Cannot access filePath: c:\\.espressif\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe","stack":"Error: non zero exit code 2\r\n\r\nERRO: padr�o inv�lido especificado em \"path:pattern\".\r\n\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:1:4499)\n\tat ChildProcess.emit (node:events:390:28)\n\tat Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)","level":"error","timestamp":"2022-04-15T16:39:10.324Z"}
```

baldhead69, 2022-04-15:
After run:
c:\esp-idf>install.bat

```c
c:\esp-idf>install.bat
Installing ESP-IDF tools
Selected targets are: esp32s2, esp32c2, esp32h2, esp32s3, esp32, esp32c3
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Skipping xtensa-esp32-elf@esp-2021r2-patch3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2021r2-patch3-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2021r2-patch3-8.4.0 (already installed)
Skipping riscv32-esp-elf@esp-2021r2-patch3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.20.3 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20211220 (already installed)
Skipping ninja@1.10.2 (already installed)
Skipping idf-exe@1.0.3 (already installed)
Skipping ccache@4.3 (already installed)
Skipping dfu-util@0.9 (already installed)
Setting up Python environment
Python 3.8.7
pip 20.3.3 from c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\pip (python 3.8)
Skipping the download of c:\.espressif\espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.
Installing Python packages
 Constraint file: c:\.espressif\espidf.constraints.v5.0.txt
 Requirement files:
  - c:\esp-idf\tools\requirements\requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Requirement already satisfied: setuptools in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 3)) (62.1.0)
Requirement already satisfied: pyserial in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 5)) (3.5)
Requirement already satisfied: future in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 6)) (0.18.2)
Requirement already satisfied: cryptography in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 7)) (36.0.2)
Requirement already satisfied: pyparsing in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 8)) (3.0.8)
Requirement already satisfied: pyelftools in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 9)) (0.27)
Requirement already satisfied: idf-component-manager in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.0.1)
Requirement already satisfied: esp-coredump in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 11)) (1.0)
Requirement already satisfied: reedsolo in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 14)) (1.5.4)
Requirement already satisfied: bitstring in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 15)) (3.1.9)
Requirement already satisfied: ecdsa in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 16)) (0.17.0)
Requirement already satisfied: kconfiglib in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 19)) (14.1.0)
Requirement already satisfied: windows-curses in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 20)) (2.3.0)
Collecting click
  Using cached https://dl.espressif.com/pypi/click/click-8.0.4-py3-none-any.whl (97 kB)
Requirement already satisfied: colorama in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from click->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 4)) (0.4.4)
Requirement already satisfied: cffi>=1.12 in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from cryptography->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 7)) (1.15.0)
Requirement already satisfied: pycparser in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from cffi>=1.12->cryptography->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 7)) (2.21)
Requirement already satisfied: six>=1.9.0 in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from ecdsa->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 16)) (1.16.0)
Requirement already satisfied: construct~=2.10 in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from esp-coredump->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 11)) (2.10.68)
Requirement already satisfied: pygdbmi<0.10.0.0 in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from esp-coredump->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 11)) (0.9.0.2)
[... cut ...] error","timestamp":"2022-04-15T16:46:31.125Z"}
{"message":"SDK Configuration editor confserver process exited with code: 1","stack":"Error: SDK Configuration editor confserver process exited with code: 1\n\tat m.printError (\\dist\\extension.js:1:30357)\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:1:29963)\n\tat ChildProcess.emit (node:events:390:28)\n\tat Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)","level":"error","timestamp":"2022-04-15T16:46:31.135Z"}
```

baldhead69, 2022-04-15:
```c
c:\esp-idf>install.bat
Installing ESP-IDF tools
Selected targets are: esp32s2, esp32c2, esp32h2, esp32s3, esp32, esp32c3
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Skipping xtensa-esp32-elf@esp-2021r2-patch3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2021r2-patch3-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2021r2-patch3-8.4.0 (already installed)
Skipping riscv32-esp-elf@esp-2021r2-patch3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.20.3 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20211220 (already installed)
Skipping ninja@1.10.2 (already installed)
Skipping idf-exe@1.0.3 (already installed)
Skipping ccache@4.3 (already installed)
Skipping dfu-util@0.9 (already installed)
Setting up Python environment
Python 3.8.7
pip 20.3.3 from c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\pip (python 3.8)
Skipping the download of c:\.espressif\espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.
Installing Python packages
 Constraint file: c:\.espressif\espidf.constraints.v5.0.txt
 Requirement files:
  - c:\esp-idf\tools\requirements\requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Requirement already satisfied: setuptools in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 3)) (62.1.0)
Requirement already satisfied: pyserial in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 5)) (3.5)
Requirement already satisfied: future in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 6)) (0.18.2)
Requirement already satisfied: cryptography in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 7)) (36.0.2)
Requirement already satisfied: pyparsing in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 8)) (3.0.8)
Requirement already satisfied: pyelftools in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 9)) (0.27)
Requirement already satisfied: idf-component-manager in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.0.1)
Requirement already satisfied: esp-coredump in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 11)) (1.0)
Requirement already satisfied: reedsolo in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 14)) (1.5.4)
Requirement already satisfied: bitstring in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 15)) (3.1.9)
Requirement already satisfied: ecdsa in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 16)) (0.17.0)
Requirement already satisfied: kconfiglib in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 19)) (14.1.0)
Requirement already satisfied: windows-curses in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\esp-idf\tools\requirements\requirements.core.txt (line 20)) (2.3.0)
Collecting click
  Using cached https://dl.espressif.com/pypi/click/click-8.0.4-py3-none-any.whl (97 kB)
Requirement already satisfied: colorama in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from click->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 4)) (0.4.4)
Requirement already satisfied: cffi>=1.12 in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from cryptography->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 7)) (1.15.0)
Requirement already satisfied: pycparser in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from cffi>=1.12->cryptography->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 7)) (2.21)
Requirement already satisfied: six>=1.9.0 in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from ecdsa->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 16)) (1.16.0)
Requirement already satisfied: construct~=2.10 in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from esp-coredump->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 11)) (2.10.68)
Requirement already satisfied: pygdbmi<0.10.0.0 in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from esp-coredump->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 11)) (0.9.0.2)
Requirement already satisfied: tqdm in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from idf-component-manager->-r c:\esp-idf\tools\requirements\requirements.core.txt (line 10)) (4.64.0)
[... cut ...]
Constraint file: c:\.espressif\espidf.constraints.v5.0.txt
Requirement files:
 - c:\esp-idf\tools\requirements\requirements.core.txt
Python being checked: c:\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
The following Python requirements are not satisfied:
jinja2<3.1
To install the missing packages, please run "install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: c:\.espressif\python_env\idf5.0_py3.8_env
    Python interpreter used: c:\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe

c:\esp-idf>
```

baldhead69, 2022-04-16:
```c
C:\.espressif\python_env\idf5.0_py3.8_env\Scripts>pip show jinja2
Name: Jinja2
Version: 3.1.1
Summary: A very fast and expressive template engine.
Home-page: https://palletsprojects.com/p/jinja/
Author: Armin Ronacher
Author-email: armin.ronacher@active-4.com
License: BSD-3-Clause
Location: c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages
Requires: MarkupSafe
Required-by: gcovr

C:\.espressif\python_env\idf5.0_py3.8_env\Scripts>
```

baldhead69, 2022-04-16:
Only downgrade jinja2 to "Jinja2==3.0.3" worked.

Not even the espressif people knew about it ?!

```c
C:\Users\Rangel>pip install Jinja2==3.0.3
Collecting Jinja2==3.0.3
  Using cached Jinja2-3.0.3-py3-none-any.whl (133 kB)
Requirement already satisfied: MarkupSafe>=2.0 in c:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from Jinja2==3.0.3) (2.1.1)
Installing collected packages: Jinja2
  Attempting uninstall: Jinja2
    Found existing installation: Jinja2 3.1.0
    Uninstalling Jinja2-3.1.0:
      Successfully uninstalled Jinja2-3.1.0
Successfully installed Jinja2-3.0.3
```

brianignacio5, 2022-04-19:
Sorry for late reply @baldhead69 I think ifound the issue.

The extension was not following the constraints given by ESP-IDF. This should be fixed in #700 and this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/216432431.zip)

baldhead69, 2022-04-23:
@brianignacio5 

idf.py size-components and idf.py size-files bug.

What is this problem ???

```
c:\esp-idf>export.bat
Checking Python compatibility
Setting IDF_PATH: c:\esp-idf

Adding ESP-IDF tools to PATH...
    c:\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin
    c:\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin
    c:\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin
    c:\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin
    c:\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    c:\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    c:\.espressif\tools\cmake\3.20.3\bin
    c:\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin
    c:\.espressif\tools\ninja\1.10.2\
    c:\.espressif\tools\idf-exe\1.0.3\
    c:\.espressif\tools\ccache\4.3\ccache-4.3-windows-64
    c:\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    c:\.espressif\python_env\idf5.0_py3.8_env\Scripts
    c:\esp-idf\tools

Checking if Python packages are up to date...
Skipping the download of c:\.espressif\espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.
Constraint file: c:\.espressif\espidf.constraints.v5.0.txt
Requirement files:
 - c:\esp-idf\tools\requirements\requirements.core.txt
Python being checked: c:\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Python requirements are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

c:\esp-idf>idf
'idf' não é reconhecido como um comando interno
ou externo, um programa operável ou um arquivo em lotes.

c:\esp-idf>idf.py
Usage: idf.py [OPTIONS] COMMAND1 [ARGS]... [COMMAND2 [ARGS]...]...

  ESP-IDF CLI build management tool. For commands that are not known to idf.py an attempt to execute it as a build
  system target will be made.

Options:
  --version                       Show IDF version and exit.
  --list-targets                  Print list of supported targets and exit.
  -C, --project-dir PATH          Project directory.
  -B, --build-dir PATH            Build directory.
  -w, --cmake-warn-uninitialized / -n, --no-warnings
                                  Enable CMake uninitialized variable warnings for CMake files inside the project
                                  directory. (--no-warnings is now the default, and doesn't need to be specified.) The
                                  default value can be set with the IDF_CMAKE_WARN_UNINITIALIZED environment variable.
  -v, --verbose                   Verbose build output.
  --preview                       Enable IDF features that are still in preview.
  --ccache / --no-ccache          Use ccache in build. Disabled by default. The default value can be set with the
                                  IDF_CCACHE_ENABLE environment variable.
  -G, --generator [Ninja]         CMake generator.
  -D, --define-cache-entry TEXT   Create a cmake cache entry. This option can be used at most once either globally, or
                                  for one subcommand.
  -b, --baud INTEGER              Baud rate for flashing. It can imply monitor baud rate as well if it hasn't been
                                  defined locally. The default value can be set with the ESPBAUD environment variable.
                                  This option can be used at most once either globally, or for one subcommand.
  -p, --port TEXT                 Serial port. The default value can be set with the ESPPORT environment variable.
[... 132 lines cut ...]

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_ASM_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "C:/esp-idf/build/CMakeFiles/CMakeOutput.log".
See also "C:/esp-idf/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
```

brianignacio5, 2022-04-24:
It seems you are trying to run these command from esp-idf folder instead of running them from a esp-idf project which is  the intended use.

I've tested on Windows 11 with the blink example and  `idf.py size-components` and `idf.py size-files`  seem to work.

baldhead69, 2022-04-24:
I tested within project too and not work in cmd of windows, even after running export.bat.

Inside visual studio code, after running idf-terminal, worked.

brianignacio5, 2022-04-25:
Well if it works with IDF Terminal, then the extension works :)

If you are using  any terminal (outside vscode) what is the output ? Maybe the export.bat is using  a different version of ESP IDF or ESP-IDF Tools from the project build folder cache and it fails.

ecochard, 2023-07-18:
for me (on windows) the menuconfig problem was solved by adding a path to the xtensa-esp32s3-elf-g++ compiler

PR #700 (2022-04-21): use constrainst in py reqs install
Use the constraints file in `ESP-IDF v5.0`  python packages install.

Fix #688
Fix #699