espressif/vscode-esp-idf-extension #689: [Bug Report]: `ESP-IDF setup` doesn't show setup wizard when vscode `Settings Sync is on` on another computer

sanwv, 2022-04-03:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
I have installed the extension on my home computer, vscode `Settings Sync is on`, whole vscode settings will sync across my all computers, include 

when I work on the computer in the office, the extension had installed successfully with vscode sync mechanism. but extension doesn't work properly, so i try reinstall the extension again and again,  `ESP-IDF setup` only show a notice let me to check linux environment. i cost many time to solve the problem, even del ~/esp and ~/.espressif

vscode sync in the background work silently, when i realized it was there and the problem also be easy to solve. only need del extension setting in vscode `settings.json`

* *To Reproduce**
had installed the extension, del ~/esp and .espressif  and reopen `ESP-IDF setup`

**Expected behavior**
ESP-IDF setup` show setup wizard  no matter the environment is configured or not, give detail reason for user

**Screenshots**
[image]

[image]

**Environment (please complete the following information):**

- OS Name & Version: ubuntu 20.04
- VSCode Version: latest
- ESP-IDF Version: 4.4
- Python Version: 3.7

brianignacio5, 2022-04-21:
Is there any reason to sync settings for this extension?

The extension settings use paths of tools that are in installed on each specific machine and only if your tools location is the same for both it would work. A change in ESP-IDF tools or python virtual environment in one computer would not be reflected in another computer.  For our purposes in think the sync settings feature should be disabled.

closed 2022-05-12 by github-actions[bot]

brianignacio5, 2022-06-09:
This should be fixed in master already and  avoided with the `idf.gitPathWin`