espressif/vscode-esp-idf-extension #690: [Bug Report]: Cannot read properties of undefined (reading 'uri')

aptly-io, 2022-04-03:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [*] Make sure you have searched for existing bugs and features request before you post an issue.
- [*] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
      idf.py command line works correctly (builds, flashes and starts monitor correctly)
- [*] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
On MacOS Monterery (12.3.1)
Trying to configure the serial port gives this error in Visual Studio Code:

Cannot read properties of undefined (reading 'uri')
Source: Espressif IDF (extension)

PS I can get around it by storing the name of the serial port manually in settings.json! Then building as described next succeeds!

The result is that I cannot build from within VSC. See how it tries to use the wrong port: /dev/ttyUSB1 below
<F1> -> Build, Flash and start a monitor on your device
```
> Executing task: /Users/franchan/.espressif/python_env/idf4.4_py3.9_env/bin/python /Users/franchan/dev/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB1 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x10000 hello_world.bin 0x0 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin <

esptool.py v3.3-dev
Serial port /dev/ttyUSB1
Traceback (most recent call last):
  File "/Users/franchan/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/franchan/dev/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 5347, in <module>
    _main()
  File "/Users/franchan/dev/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 5340, in _main
    main()
  File "/Users/franchan/dev/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 4647, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/Users/franchan/dev/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 114, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
  File "/Users/franchan/dev/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 320, in __init__
    self._port = serial.serial_for_url(port)
  File "/Users/franchan/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/Users/franchan/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
The terminal process "/Users/franchan/.espressif/python_env/idf4.4_py3.9_env/bin/python '/Users/franchan/dev/esp/esp-idf/components/esptool_py/esptool/esptool.py', '-p', '/dev/ttyUSB1', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32s3', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', 'detect', '0x10000', 'hello_world.bin', '0x0', 'bootloader/bootloader.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 1.```

**To Reproduce**
<F1> -> ESP-IDF: Device Configuration -> Device Port -> /dev/cu.usbserial-142 -> <ENTER>

**Expected behavior**
Configure the serial port through the VSC espressif extension on MacOS should just succeed.

**Screenshots**
All logs are included.

**Environment (please complete the following information):**

- OS Name & Version: MacOS X 12.3.1
- VSCode Version: 1.66.0
- ESP-IDF Version: v4.4-367-gc29343eb94
- Python Version: 3.9.10

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration in your clipboard which you can copy here.

**Output**

In Visual Studio Code select menu `View` -> `Output` -> ESP-IDF, ESP-IDF Debug Adapter, Heap Trace, OpenOCD and SDK Configuration Editor.

> **NOTE:** Use `idf.openOcdDebugLevel` configuration setting to 3 or more to show debug logging in OpenOCD server output.

> **NOTE:** Use `logLevel` in your <project-directory>/.vscode/launch.json to 3 or more to show more debug adapter output.

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process output.

**Logs**
Please share the log file which can be obtained from:

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1 or View Menu -> Command Palette) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.

brianignacio5, 2022-04-13:
How are you trying to set the serial port ? Are you using the `ESp-IDF: Select port to use` command ?

Could you share the extension log and the doctor command output as described in the issue template ?

aptly-io, 2022-04-17:
@brianignacio5 Sorry for the late reply. No I used
`Shift-Command-p -> ESP-IDF: Device Configuration -> Device Port`
This shows a small 1 line entry box to enter the port. After entering a device name it did not work as described.

As you suggest, I now tried:
`Shift-Command-p -> ESP-IDF: Select port to use`
This shows a list of ports to use. If the esp32-s3-wroom-1 is connected, I see and can select the option /dev/cu.usbserial-1420 from the list. It seems the local workspace settings are updated correctly (not the global settings) I did not know about this one. Thank you.

You can close the issue (though using `ESP-IDF: Device Configuration -> Device Port` does still not work)

I'm sorry for not including the doctor command output and log files. I feel uncomfortable doing so for privacy reasons.

brianignacio5, 2022-04-20:
Thanks for your feedback @aptly-io !

The `Device Port` command should be fixed with #701 and we will soon make a PR to add anonimity to the doctor command output.

PR #701 (2022-04-22): fix idf config  update conf parameter
* Fix idfConfiguration updateConfParameter to use the currently selected workspace folder to save manually define settings such as those from `ESP-IDF: Device Configuration` and `ESP-IDF: Configure Paths`.
* Add `-n` to openOCD rules copy command at the end of the `ESP-IDF: Configure ESP-IDF extension` setup wizard.

Fix #690
Fix #691
Fix #698
closed 2022-04-22 by brianignacio5